YosysHQ/nextpnr #254: ICE40UP5K-SG48I: SB_RGBA_DRV: LED pin mapping appears hard coded

ewenmcneill, 2019-03-23:
I'm not sure if this is a `nextpnr` specific issue (eg, bitstream generation), or a hardware limitation.  But since there *are* parameters which seem to be intended to map between ICE40 RGB driver channels and external chip pins, which don't seem to do anything in `nextpnr` I thought I'd ask here, as I can't find anything to confirm the mapping is supposed to be fixed.

In brief:

*   Is the mapping between the `SB_RGBA_DRV` `RGB0` / `RGB1` / `RGB2` output parameters and the `RGB0` / `RGB1` / `RGB2` output pins hard coded?  Or is it user controllable?

*   If it is user controllable, should swapping the `SB_RGBA_DRV` parameters to `pcf` pin around to different mappings alter which `SB_RGBA_DRV` channel controls which output pin?

*   If the `SB_RGBA_DRV` channel is not user controllable, why are the `RGB0` / `RGB1` / `RGB2` parameters required?

According to https://upverter.com/parts/Lattice/iCE40UP5K-SG48I/ `RGB0` / `RGB1` / `RGB2` *pins* are 39/40/41 respectively for this part.  Being high current LED driver pins, I assume the LEDs are constrained to connect to *one of* those pins to have the relevant function, but naively I assumed the *ordering* of those pins was controllable.

More detail below.

I found this while testing on a Fomu (Tomu FPGA) EVT3 board (https://github.com/im-tomu/fomu-hardware/), which has an ICE40UP5K-SG48I on it, with the[`fomu-tests` blink example(https://github.com/im-tomu/fomu-tests/tree/master/blink), which uses `nextpnr` to synthesize the ICE40UP5K bitstream, I've found that it appears like the mapping between `RGB0` / `RGB1` / `RGB2 on the `SB_RGBA_DRV` built in ICE40UP5K hard module and the external LED driver pins (39/40/41 AFAICT for this part) -- and thus the mapping to the external LEDs -- appears to be fixed.

The` RGB0` / `RGB1` / `RGB2` parameters need to be specified in order to instantiate the `SB_RGBA_DRV` module at all (if they're not specified the module gets removed from the synthesis, and the LEDs are just full on).  But the order in which the `led_r` / `led_b` / `led_g` pins (from `pcf` file -- in this case https://github.com/im-tomu/fomu-tests/blob/master/pcf/fomu-evt3.pcf) are mapped against `RGB0` / `RGB1` / `RGB2` appears to make no difference to the resulting physical output.  (The Fomu EVT3 has a tri-colour LED, so in this test this is obvious as the colour of the linking LED -- functionality connected to `RGB0` not changing with the rearrangement).

@xobs (Fomu hardware designer) [reported on IRC](https://logs.timvideos.us/%23tomu/%23tomu.2019-03-23.log.html) that he saw similar fixed ordering behaviour.

Relevant instantiation code (from https://github.com/im-tomu/fomu-tests/blob/26b9d86232ca8dc86ff810e1897b2d328df1424c/blink/blink.v#L90-L99):

```
    SB_RGBA_DRV RGBA_DRIVER (
        .CURREN(1'b1),
        .RGBLEDEN(1'b1),
        .RGB0PWM(~user_5_pulled),       // Blue
        .RGB1PWM(~user_6_pulled),       // Red
        .RGB2PWM(outcnt[4]),            // Green
        .RGB0(led_r),
        .RGB1(led_g),
        .RGB2(led_b)
    );
```

The comments against the `PWM` parameters reflect the actual *as observed in outside world* colour mapping of the channels, and thus the PCB layout (there's one pair of pins crossed between the EVT3 schematic and the PCB, which experimentally appears to be Red/Blue).

With both the above ordering of the `RGB0`/`RGB1`/`RGB2` parameters to the instantiation, or with this rearranged ordering:

```
fomu@fomu-dev:~/src/fomu-tests/blink $ grep -A 9 SB_RGBA_DRV blink.v 
    SB_RGBA_DRV RGBA_DRIVER (
        .CURREN(1'b1),
        .RGBLEDEN(1'b1),
        .RGB0PWM(~user_5_pulled),       // Blue
        .RGB1PWM(~user_6_pulled),       // Red
        .RGB2PWM(outcnt[4]),            // Green
        .RGB0(led_g),
        .RGB1(led_b),
        .RGB2(led_r)
    );
fomu@fomu-dev:~/src/fomu-tests/blink $ 
```

the LED colours activated by the SB_RGBA_DRV channels are the same each time, even over full "reset FPGA" / "make clean" / "make" / `fomu-flash` cycles.

Test environment is Raspberry Pi running 1.2.6 (pre-release) [Fomu Development Environment Boot Image](https://github.com/im-tomu/fomu-pi-gen/), which has:

```
fomu@fomu-dev:~/src/fomu-tests/blink $ nextpnr-ice40 --version
nextpnr-ice40 -- Next Generation Place and Route (git sha1 cadbf42)
fomu@fomu-dev:~/src/fomu-tests/blink $ 
```

(built from `nextpnr` git, a little while ago; package built details are in the README at https://github.com/im-tomu/fomu-pi-gen/)

Ewen

ewenmcneill, 2019-03-23:
TL;DR: `arachne-pnr` refuses to place this design, with a "fatal error: bad constraint", implying the mapping from `SB_RGBA_DRV` `RGBn` to physical pins is hard coded inside the chip.  In which case the main `nextpnr` issue might be that it's missing this constraint check, as `nextpnr` will complete PNR ignoring this apparent error.  More detail below.

I've managed to build `arachne-pnr` on the Raspberry Pi (Fomu development platform), and figure out suitable `yosys` / `arachne-pnr` commands to synthesise / PNR this same example.  `arachne-pnr` fails the PNR with:

```
place_constraints...
fatal error: bad constraint on `led_b': pin 40 does not correspond to RGB driver output `RGB2'
Makefile:63: recipe for target '.build/blink.asc' failed
```

with the first example above, ie:

```
fomu@fomu-dev:~/src/fomu-tests-arachne-pnr/blink $ grep -A 9 SB_RGBA_DRV blink.v | grep 'RGB[0-9]('
        .RGB0(led_r),
        .RGB1(led_g),
        .RGB2(led_b)
fomu@fomu-dev:~/src/fomu-tests-arachne-pnr/blink $ grep led_ ../pcf/fomu-evt3.pcf 
set_io led_r 39
set_io led_b 40
set_io led_g 41
fomu@fomu-dev:~/src/fomu-tests-arachne-pnr/blink $ 
```

As tested with https://github.com/im-tomu/fomu-tests/blob/master/blink/blink.v, https://github.com/im-tomu/fomu-tests/blob/master/blink/blink.v and:

```
yosys -p "synth_ice40 -top blink -blif .build/blink.blif" blink.v
arachne-pnr -d 5k -P sg48 .build/blink.blif -p ../pcf/fomu-evt3.pcf -o .build/blink.asc
```

which suggests that there *is* a hard coded mapping required.

If I swap the parameter order to:

```
fomu@fomu-dev:~/src/fomu-tests-arachne-pnr/blink $ grep -A 9 SB_RGBA_DRV blink.v | grep 'RGB[0-9]('
        .RGB0(led_r),
        .RGB1(led_b),
        .RGB2(led_g)
fomu@fomu-dev:~/src/fomu-tests-arachne-pnr/blink $ grep led_ ../pcf/fomu-evt3.pcf 
set_io led_r 39
set_io led_b 40
set_io led_g 41
fomu@fomu-dev:~/src/fomu-tests-arachne-pnr/blink $ 
```

then `yosys` / `arachne-pnr` do complete successfully, and the resulting bitstream works successfully.  (The `RGB0` / `led_r` and `RGB1` / `led_b` are swapped with the reality of the actual LED colours, but this is consistent with my understanding of the layout issue when the PCB was laid out, suggesting the `pcf` file needs updating to match the PCB layout rather than the original schematic design.)

Ewen

ewenmcneill, 2019-03-23:
FTR, it looks like `nextpnr` doesn't even issue a warning that I can see:

```
fomu@fomu-dev:~ $ egrep '^yosys|^nextpnr|led_|RGB' /tmp/build-with-nextpnr 
yosys -p 'synth_ice40 -top blink -json .build/blink.json' blink.v
Generating RTLIL representation for module `\SB_RGBA_DRV'.
     SB_RGBA_DRV                     1
nextpnr-ice40 --up5k --package sg48 --json .build/blink.json --pcf ../pcf/fomu-evt3.pcf --asc .build/blink.asc
Info: constrained 'led_r' to bel 'X4/Y31/io0'
Info: constrained 'led_b' to bel 'X5/Y31/io0'
Info: constrained 'led_g' to bel 'X6/Y31/io0'
Info: led_r use by SB_RGBA_DRV RGBA_DRIVER, not creating SB_IO
Info: led_g use by SB_RGBA_DRV RGBA_DRIVER, not creating SB_IO
Info: led_b use by SB_RGBA_DRV RGBA_DRIVER, not creating SB_IO
Info:   constrained SB_RGBA_DRV 'RGBA_DRIVER' to X0/Y30/rgba_drv_0
Info: 	         SB_RGBA_DRV:     1/    1   100%
fomu@fomu-dev:~ $ 
```

instead it appears it's just silently ignoring the wrong mapping of `SB_RGB_DRV` output pins.

(It's also not clear to me why the `SB_RGB_DRV` module needs these pin mappings listed if they're hard coded, but I guess the *number* of pins available from the driver varies between FPGAs and listing them is one way to ensure the correct *number* of mapped pins for the targeted FPGAs.  It also seems to be what the Lattice tools do.)

Ewen

PS: The only warnings generated seem to relate to pins from the `pcf` file which are unused in the design:

```
fomu@fomu-dev:~ $ egrep -i '^yosys|^nextpnr|led_|RGB|warn|err' /tmp/build-with-nextpnr 
yosys -p 'synth_ice40 -top blink -json .build/blink.json' blink.v
blink.v:30: Warning: Identifier `\clk' is implicitly declared.
Generating RTLIL representation for module `\SB_RGBA_DRV'.
ABC: Warning: The network is combinational (run "fraig" or "fraig_sweep").
     SB_RGBA_DRV                     1
Warnings: 1 unique messages, 1 total
Yosys 0.8+284 (git sha1 90bce041, gcc 6.3.0-18+rpi1+deb9u1 -fPIC -Os)
nextpnr-ice40 --up5k --package sg48 --json .build/blink.json --pcf ../pcf/fomu-evt3.pcf --asc .build/blink.asc
Info: constrained 'led_r' to bel 'X4/Y31/io0'
Info: constrained 'led_b' to bel 'X5/Y31/io0'
Info: constrained 'led_g' to bel 'X6/Y31/io0'
Warning: unmatched constraint 'pmoda_1' (on line 8)
Warning: unmatched constraint 'pmoda_2' (on line 9)
Warning: unmatched constraint 'pmoda_3' (on line 10)
Warning: unmatched constraint 'pmoda_4' (on line 11)
Warning: unmatched constraint 'clki_alt' (on line 12)
Warning: unmatched constraint 'user_1' (on line 14)
Warning: unmatched constraint 'user_2' (on line 15)
Warning: unmatched constraint 'user_3' (on line 16)
Warning: unmatched constraint 'user_4' (on line 17)
Warning: unmatched constraint 'pmodb_1' (on line 20)
Warning: unmatched constraint 'pmodb_2' (on line 21)
Warning: unmatched constraint 'pmodb_3' (on line 22)
Warning: unmatched constraint 'pmodb_4' (on line 23)
Warning: unmatched constraint 'spi_mosi' (on line 24)
Warning: unmatched constraint 'spi_miso' (on line 25)
Warning: unmatched constraint 'spi_clk' (on line 26)
Warning: unmatched constraint 'spi_io2' (on line 27)
Warning: unmatched constraint 'spi_io3' (on line 28)
Warning: unmatched constraint 'spi_cs' (on line 29)
Warning: unmatched constraint 'uart_tx' (on line 30)
Warning: unmatched constraint 'uart_rx' (on line 31)
Warning: unmatched constraint 'usb_dn' (on line 32)
Warning: unmatched constraint 'usb_dp' (on line 33)
Warning: unmatched constraint 'usb_dp_pu' (on line 34)
Warning: unmatched constraint 'usb_dn_pu' (on line 35)
Warning: unmatched constraint 'dbg_1' (on line 36)
Warning: unmatched constraint 'dbg_2' (on line 37)
Warning: unmatched constraint 'dbg_3' (on line 38)
Warning: unmatched constraint 'dbg_4' (on line 39)
Warning: unmatched constraint 'dbg_5' (on line 40)
Warning: unmatched constraint 'dbg_6' (on line 41)
Info: led_r use by SB_RGBA_DRV RGBA_DRIVER, not creating SB_IO
Info: led_g use by SB_RGBA_DRV RGBA_DRIVER, not creating SB_IO
Info: led_b use by SB_RGBA_DRV RGBA_DRIVER, not creating SB_IO
Info:   constrained SB_RGBA_DRV 'RGBA_DRIVER' to X0/Y30/rgba_drv_0
Info: 	         SB_RGBA_DRV:     1/    1   100%
31 warnings, 0 errors
fomu@fomu-dev:~ $ 
```

ewenmcneill, 2019-03-23:
FTR, `tnt` answered my question whether it was hard coded on [`#fomu` on FreeNode IRC today](https://logs.timvideos.us/%23tomu/%23tomu.2019-03-23.log.html):

```
19:02 < tnt> ewen: The RGB0/1/2 connections in HDL are purely cosmetic. They 
             are hardwired in the the silicon to matching pads and can't be 
             changed.
[...]
19:05 < tnt> it's for compatibility with lattice docs, we need to accept those 
             connections, but the doc from lattic is quite clear that they are 
             hw connections and map to the RGB0/1/2 pins of the fpga.
```

So erroring out (like `arachne-pnr`) or warning of the miss match is probably the best that `nextpnr` can do here.

Ewen

julian1, 2023-02-09:
Is it known if the RGB pins (39,40,41) of ICE40UP5K-SG48I, be used as cmos/push-pull outputs(ie no PWM) with nextpnr? 
Or, is the hardware dedicated as open-drain outputs?

ewenmcneill, 2024-09-17:
Closing issue since (a) after 5 years I think it's unlikely I'm going to work on this, and (b) it's pretty obvious no one else is going to work on it either :-)

As noted above https://github.com/YosysHQ/nextpnr/issues/254#issuecomment-475843155 the "hard coding" is embedded in the silicon for the LED driver in the iCE40UP5K, so at best we could generate a warning that the input parameters to the module.  But it seems few if any other people have run into this "lack of warning" surprise.

Ewen

closed 2024-09-17 by ewenmcneill