YosysHQ/nextpnr #256: Generic Architecture Documentation

AbigailCliche, 2019-03-25:
Are there plans to add to the documentation for implementing a generic architecture any time soon?
Any guidance that could be given for using this feature would be appreciated, especially relating to getting started with implementing the API.

eddiehung, 2019-03-26:
Hi @AbigailCliche! Thanks for raising this issue. The generic architecture is definitely an area we have not been focussing on, but we do have plans to allow to reinforce the Python API so that one can create new architectures on the fly -- this would also enable some other things we are interested in. 

It's great to hear that others could also be interested! However, the best I can promise right now is that will happen soon (TM). Perhaps you can share a little more about what brought you here, and what you're interested in using the generic architecture for?

AbigailCliche, 2019-03-27:
Thank you for a quick response. I am a part-time research assistant working to build a hardware security research tool. I am trying to use nextpnr with a custom virtual architecture. We are currently using Yosys for synthesis.

mithro, 2019-03-27:
If you haven't already, you might want to look at [Verilog to Routing](https://verilogtorouting.org/). The [SymbiFlow Architecture Definition repository](https://github.com/SymbiFlow/symbiflow-arch-defs) has some pretty complete examples of using Yosys + vpr for targeting real world architectures (examples are iCE40 and Xilinx Series 7 devices). Verilog to Routing is very much designed for researching a large number of different architecture design choices.

FYI We eventually hope that the SymiFlow Architecture Definitions will mutate into a device database generation system which supports both vpr and nextpnr in the future.

eddiehung, 2019-04-25:
HI @AbigailCliche, in case you missed it @daveshah1's excellent work was recently merged in, which might be useful to you: https://github.com/YosysHQ/nextpnr/pull/261

AbigailCliche, 2019-05-08:
Thank you, this is exactly what I was hoping for.

closed 2019-05-18 by daveshah1