YosysHQ/nextpnr #265: functionally identical netlists produce different results

jlwehle, 2019-04-13:
This may just be an example of nextpnr (by the nature of the algorithms)
being sensitive to initial conditions, however just in case something
else is going on here's the deal:

The attached design has a bit of logic which can be written as:

  winner = awin || ((ewin ^ lwin) & ~iwin);

or:

  winner = awin
           || ((ewin && !lwin) && !iwin)
           || ((!ewin && lwin) && !iwin);

Synthesizing each approach like so:

  yosys -q -p 'synth_ice40 -top rx_cdr_decimate -json rx_cdr_decimate.json' rx_cdr_decimate.v

generates functionally equivalent netlists ... checked by using netlistsvg
to generate an image of each netlist and then comparing the images.

Running try-seeds on the netlist for the bitwise xor approach results
in icetime saying the fastest bitstream has total path delay: 3.82 ns.
Commands are:

  nextpnr-ice40 --hx8k --package tq144:4k --asc rx_cdr_decimate-xor.asc --json rx_cdr_decimate-xor.json --freq 260 --seed 35
  icetime -d hx8k -t rx_cdr_decimate-xor.asc

Running try-seeds on the netlist for the logical or approach results
in icetime saying the fastest bitstream has total path delay: 4.08 ns.
Commands are:

  nextpnr-ice40 --hx8k --package tq144:4k --asc rx_cdr_decimate-or.asc --json rx_cdr_decimate-or.json --freq 255 --seed 23
  icetime -d hx8k -t rx_cdr_decimate-or.asc

Also running:

  nextpnr-ice40 --hx8k --package tq144:4k --asc rx_cdr_decimate-or.asc --json rx_cdr_decimate-or.json --freq 260 --seed 35

results in nextpnr-ice40 failing to meet the goal of 3.84 ns (260 MHz).

Given that the netlists are functionally identical it seems desirable
for nextpnr-ice40 to be able to produce solutions for both netlists
which are the same speed.  Though perhaps I just have not tried enough
seeds.

[files.zip](https://github.com/YosysHQ/nextpnr/files/3075744/files.zip)

daveshah1, 2019-04-13:
As soon as you change any logic, naming and ordering inside the netlist will change, which will inevitably affect all of nextpnr's algorithms - reliance on this is something of an inevitability with standard FPGA place and route algorithms.

Matthew-Bradley, 2019-10-26:
Since the different approaches produce outcomes with different performance, is there a set of best practices for nextpnr to produce the best results?

closed 2024-09-30 by gatecat