YosysHQ/nextpnr #271: nextpnr-ice40 --gui not working

jmhmot, 2019-04-23:
I have compiled nexpnr and all seems to be working fine in non-gui mode, but when I run:
nextpnr-ice40 --gui at the terminal nothing happens, no CPU activity, no output, terminal held waiting....
I'm using Ubuntu 16.04.
gdb attach to process states:
...Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
pthread_cond_timedwait@@GLIBC_2.3.2 ()
    at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
225	../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S: No such file or directory.
Sorry I'm not really experienced to know if this is a clue to the problem or not.
Is the gui having some thread-locking issues?

I built nextpnr from the instructions given at:
http://www.clifford.at/icestorm/

jmhmot, 2019-04-25:
Please, does anyone have any suggestions?

ZipCPU, 2019-04-25:
@jmhmot,
I just ran `nextpnr-ice40 --gui` at my terminal on an Ubuntu 16.04 platform with no problems.
Could this problem be due to a missing dependency?
Dan

jmhmot, 2019-04-26:
@ZipCPU Thanks for your reply. How would I go about finding which missing dependency? Could it be a graphics driver issue? I'm using 2 graphics card (for 4 monitors). Output from lshw -c video is:
 *-display               
       description: VGA compatible controller
       product: RV730 XT [Radeon HD 4670]
       vendor: Advanced Micro Devices, Inc. [AMD/ATI]
       physical id: 0
       bus info: pci@0000:01:00.0
       version: 00
       width: 64 bits
       clock: 33MHz
       capabilities: vga_controller bus_master cap_list rom
       configuration: driver=radeon latency=0
       resources: irq:27 memory:e0000000-efffffff memory:f7c20000-f7c2ffff ioport:e000(size=256) memory:c0000-dffff
  *-display
       description: Display controller
       product: Xeon E3-1200 v2/3rd Gen Core processor Graphics Controller
       vendor: Intel Corporation
       physical id: 2
       bus info: pci@0000:00:02.0
       version: 09
       width: 64 bits
       clock: 33MHz
       capabilities: bus_master cap_list
       configuration: driver=i915 latency=0
       resources: irq:26 memory:f7800000-f7bfffff memory:d0000000-dfffffff ioport:f000(size=64)

ZipCPU, 2019-05-04:
Can you run `glxinfo | grep "OpenGL version` and let us know what version of OpenGL you are using?

jmhmot, 2019-05-07:
Hi ZipCPI
$glxinfo | grep "OpenGL version" 
gives:
$OpenGL version string: 3.0 Mesa 18.0.5

mmicko, 2019-05-07:
Ok, so probably issue is hardware/OpenGL supported version. Minimal version nexpnr is supporting and enforcing is 3.2, we are letting users run anyway since 3.1 could even work on some, but it is unsupported.
So bad news is that you hardware just can not be supported due to opengl calls use by shaders inside nextpnr :(

ric96, 2019-05-22:
same as #240

ric96, 2019-05-28:
@jmhmot can you check now with --gui-no-aa ? Its solved for me

jmhmot, 2019-06-03:
@ric96 I just cloned the latest code and rebuilt it. Then ran :
$nextpnr-ice40 --gui --gui-no-aa
and.... no change, command line hangs doing nothing as before.
BTW I do have it all working on another PC also with Ubuntu 16.04 (but different hardware, an X220 laptop). Anyone any idea why it would hang doing nothing?

davidthings, 2019-06-29:
This happened to me a while ago.  The problem was that I had upgraded some of the underlying OS packages.  The fix, I believe, was to rebuild nextpnr.