YosysHQ/nextpnr #287: Assertion failure: c == '0' || c == '1' || c == 'X' || c == 'x'

sbourdeauducq, 2019-06-07:
[repro.zip](https://github.com/YosysHQ/nextpnr/files/3266773/repro.zip)

Run:
```
nextpnr-ecp5 --json result/top.json
```
Result:
```
...
Info: [ 79048,  79640) |*************************+
Info: [ 79640,  80232) |***********************************+
Info: [ 80232,  80824) |*************************************+
Info: [ 80824,  81416) |***************+
Info: [ 81416,  82008) |*****+
terminate called after throwing an instance of 'nextpnr_ecp5::assertion_failure'
  what():  Assertion failure: c == '0' || c == '1' || c == 'X' || c == 'x' (/build/source/ecp5/bitstream.cc:164)
Aborted
```

Using nextpnr 206761b41589f218e7c3f1ebc360183f3d8c1c79 and prjtrellis 9fb38f849e8fa2adb77c1795b77ee1f899d31e4a (bug also present on older versions).

daveshah1, 2019-06-08:
Hi! Thanks for the bug and the MCVE. Can you provide the RTLIL (or Verilog if it isn't nMigen) that this JSON came from, so I can understand exactly where the `-`s in the RAM init values are coming from and how to deal with them (I'm pretty sure they can be treated just like `x` but I want to be certain)?

sbourdeauducq, 2019-06-08:
After looking more into this, this seems to appear when a RAM init value is negative in nMigen, which had happened by accident in my code.
Probably nMigen needs fixing, and the error message in nextpnr improved.

daveshah1, 2019-06-08:
Thanks for the explanation! Better error handling added in 187db92b

closed 2019-06-08 by daveshah1