YosysHQ/nextpnr #290: nextpnr fails on pin in package for ECP5 85k

bmentink, 2019-06-17:
I  get the following error when nextpnr (latest git, also trellis latest git ) runs ..

```
ERROR: IO pin 'PIN34$tr_io' constrained to pin 'R16', which does not exist for package 'CSFBGA285'. 
```
The relevant line in the Makefile is:

`nextpnr-ecp5  --json $< --textcfg $@ --85k --package CSFBGA285 --lpf ${PROJ}.lpf
`
I had a look in the .json file for that package, and there is an entry for pin R16, so not sure why the error ..

daveshah1, 2019-06-17:
The reason for the error is that this pin is a special pin in its own tile type (SPICB0) that isn't being imported as an IO tile. I need to look into the correct behaviour as to how to handle this tile type (it's also missing bit data in Trellis at the moment).

bmentink, 2019-06-17:
Ok thanks for the quick response ... :)

bmentink, 2019-06-19:
Can you point me to where I can get the correct *.lpf constraint files for the ECP5 devices? Lattice site is not helpful as they only point to GUI development, and I don't want to install all the Diamond tools ..

daveshah1, 2019-06-19:
There are no lpf files for ECP5 chips, they are specific to a board (or even a design in many cases).

If you are looking for the pinout, Lattice have CSV pinouts here:  https://www.latticesemi.com/Products/FPGAandCPLD/ECP5 (in the Downloads at the bottom)

bmentink, 2019-06-19:
Thanks,
I am trying to find the specs for each pin, as I am getting this error on some pins:

`ERROR: TERMINATION is not supported with Vcc = 3V3 (on PIO PIN37$tr_io)`

daveshah1, 2019-06-19:
You might find some information at: https://www.latticesemi.com/view_document?document_id=50464 - otherwise I am not aware of any complete IO mode support matrix.

Indeed TERMINATION is only supported for SSTL type standards, at 1.35V/1.5V/1.8V and on left/right IO pins only.

Note that LVDS differential termination uses the DIFFRESISTOR rather than TERMINATION attribute.

bmentink, 2019-06-19:
Cool, that was what I was looking for ... I now have a .bit file :)

xobs, 2019-09-30:
I'm running into this error as well, trying to use the SPI pins as part of a verilog-defined quad-spi implementation:

```sh
$ nextpnr-ecp5 --json top.json --lpf top.lpf --textcfg top.config --45k --package CABGA381 --freq 48 --speed 8
Info: Importing module top
Info: Rule checker, verifying imported design
Info: Checksum: 0x4e3dbd15

Info: constraining clock net 'clk8' to 8.00 MHz
Info: constraining clock net 'sys_clk' to 48.00 MHz
Info: constraining clock net 'clk12_clk' to 12.00 MHz
Info: constraining clock net 'clk48_clk' to 48.00 MHz

Info: Packing IOs..
Info: pin 'usb_pullup$tr_io' constrained to Bel 'X0/Y17/PIOA'.
Info: pin 'serial_rx$tr_io' constrained to Bel 'X13/Y71/PIOB'.
Info: pin 'serial_tx$tr_io' constrained to Bel 'X6/Y71/PIOA'.
Info: pin 'clk8$tr_io' constrained to Bel 'X90/Y68/PIOA'.
Info: usb_d_p feeds TRELLIS_IO TRELLIS_IO, removing $nextpnr_ibuf usb_d_p.
Info: pin 'TRELLIS_IO' constrained to Bel 'X0/Y32/PIOB'.
Info: usb_d_n feeds TRELLIS_IO TRELLIS_IO_1, removing $nextpnr_ibuf usb_d_n.
Info: pin 'TRELLIS_IO_1' constrained to Bel 'X0/Y32/PIOA'.
Info: pin 'usb_vbusdet$tr_io' constrained to Bel 'X0/Y20/PIOA'.
Info: pin 'led3$tr_io' constrained to Bel 'X0/Y17/PIOC'.
Info: pin 'led4$tr_io' constrained to Bel 'X0/Y14/PIOA'.
Info: pin 'spiflash4x_cs_n$tr_io' constrained to Bel 'X15/Y71/PIOA'.
ERROR: IO pin 'spiflash4x_clk$tr_io' constrained to pin 'U3', which does not exist for package 'CABGA381'.
ERROR: Packing design failed.
0 warnings, 2 errors
$
```

What is the recommended workaround?  Pin `U3` does exist on this package...

xobs, 2019-09-30:
To anyone else who finds this: Apparently the workaround is to use a `USRMCLK` block, which takes care of interacting with that pin for you.