YosysHQ/nextpnr #291: Design blows out LUT's on ECP5-85k comparted to Spartan-6-9k

bmentink, 2019-06-17:
Hi,

I am trying to port across a project that is currently running on a 9k spartan-6 device. ON that device the device utilisation is:

```
Device Utilization Summary:

Slice Logic Utilization:
  Number of Slice Registers:                 3,718 out of  11,440   32%
    Number used as Flip Flops:               3,713
    Number used as Latches:                      0
    Number used as Latch-thrus:                  0
    Number used as AND/OR logics:                5
  Number of Slice LUTs:                      5,459 out of   5,720   95%
    Number used as logic:                    5,361 out of   5,720   93%
      Number using O6 output only:           4,108
      Number using O5 output only:             227
      Number using O5 and O6:                1,026
      Number used as ROM:                        0
    Number used as Memory:                      45 out of   1,440    3%
      Number used as Dual Port RAM:              0
      Number used as Single Port RAM:            0
      Number used as Shift Register:            45
        Number using O6 output only:             5
        Number using O5 output only:             0
        Number using O5 and O6:                 40
    Number used exclusively as route-thrus:     53
      Number with same-slice register load:     12
      Number with same-slice carry load:        41
      Number with other load:                    0

```

When I use yosys/nextpnr-ecp5, I get this utilization:

```
Info: Device utilisation:
Info:          TRELLIS_SLICE: 38529/41820    92%
Info:             TRELLIS_IO:    49/  364    13%
Info:                   DCCA:     3/   56     5%
Info:                 DP16KD:     0/  208     0%
Info:             MULT18X18D:     0/  156     0%
Info:                 ALU54B:     0/   78     0%
Info:                EHXPLLL:     1/    4    25%
Info:                EXTREFB:     0/    2     0%
Info:                   DCUA:     0/    2     0%
Info:              PCSCLKDIV:     0/    2     0%
Info:                IOLOGIC:     0/  224     0%
Info:               SIOLOGIC:     0/  140     0%
Info:                    GSR:     0/    1     0%
Info:                  JTAGG:     0/    1     0%
Info:                   OSCG:     0/    1     0%
Info:                  SEDGA:     0/    1     0%
Info:                    DTR:     0/    1     0%
Info:                USRMCLK:     0/    1     0%
Info:                CLKDIVF:     0/    4     0%
Info:              ECLKSYNCB:     0/    8     0%
Info:                DLLDELD:     0/    8     0%
Info:                 DDRDLL:     0/    4     0%
Info:                DQSBUFM:     0/   14     0%
Info:        TRELLIS_ECLKBUF:     0/    8     0%

```

I have no idea why so many LUT's are used with nextpnr ... 
I have 16k of ram in the project (inferred), maybe that is an issue?
Also nextpnr is still trying to route the above (taken 40 minutes so far ) ...

Any idea's ..

daveshah1, 2019-06-17:
The lack of DP16KD mean that RAM inference is failing for some reason (which is a Yosys issue).

There is also a known issue with delay estimates for the slower speed grades (I've only tuned the delay estimates for the 5G grade so far, need to make sure the delay estimates are correct for the other grades). Combined with the somewhat nasty structure to route that is usually formed when RAM isn't inferred (a lot of high fanout nets and congestion), I'm less surprised by the route time.

bmentink, 2019-06-17:
Is there any way I can force it to use the DP16KD, rather than inferring it? I will also update yosys to latest git to see if anything has changed  ..

daveshah1, 2019-06-17:
Yes, you can instantiate it like https://github.com/skristiansson/wb_sdram_ctrl/blob/master/rtl/verilog/dpram_ecp5.v

Please do provide a test case as a Yosys GitHub issue if this isn't resolved with latest Yosys, in case this is an issue that isn't known about.

bmentink, 2019-06-17:
Thanks, will do ..

bmentink, 2019-06-19:
I have placed a Yosys Github issue ... they have mapped as a "Feature"
I instantiated the dpram ... now I get something sensible ..

```
Info: Device utilisation:
Info:          TRELLIS_SLICE:  4254/41820    10%
Info:             TRELLIS_IO:    49/  364    13%
Info:                   DCCA:     3/   56     5%
Info:                 DP16KD:     2/  208     0%
Info:             MULT18X18D:     0/  156     0%
Info:                 ALU54B:     0/   78     0%
Info:                EHXPLLL:     1/    4    25%
Info:                EXTREFB:     0/    2     0%
Info:                   DCUA:     0/    2     0%
Info:              PCSCLKDIV:     0/    2     0%
Info:                IOLOGIC:     0/  224     0%
Info:               SIOLOGIC:     0/  140     0%
Info:                    GSR:     0/    1     0%
Info:                  JTAGG:     0/    1     0%
Info:                   OSCG:     0/    1     0%
Info:                  SEDGA:     0/    1     0%
Info:                    DTR:     0/    1     0%
Info:                USRMCLK:     0/    1     0%
Info:                CLKDIVF:     0/    4     0%
Info:              ECLKSYNCB:     0/    8     0%
Info:                DLLDELD:     0/    8     0%
Info:                 DDRDLL:     0/    4     0%
Info:                DQSBUFM:     0/   14     0%
Info:        TRELLIS_ECLKBUF:     0/    8     0%

```
Thanks for the link

closed 2019-06-19 by daveshah1