YosysHQ/nextpnr #292: qt5.6 version not found error

noah95, 2019-06-19:
I followed the build instructions in README.md to build nextpnr but now the command

```bash
nextpnr-ice40 --up5k --package uwg30 --json .build/blink.json --pcf ../pcf/fomu-hacker.pcf --asc .build/blink.asc
```
fails with
```
nextpnr-ice40: /usr/lib/x86_64-linux-gnu/libQt5Core.so.5: version `Qt_5.6' not found (required by nextpnr-ice40)
nextpnr-ice40: /usr/lib/x86_64-linux-gnu/libQt5Core.so.5: version `Qt_5' not found (required by nextpnr-ice40)
nextpnr-ice40: /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5: version `Qt_5' not found (required by nextpnr-ice40)
nextpnr-ice40: /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5: version `Qt_5' not found (required by nextpnr-ice40)
```

I am on Ubuntu 16.04 and have `qt5-default` installed. Do I have to install more Qt stuff?

daveshah1, 2019-06-19:
`qt5-default` should be all that you need. Is it possible that you have a, possibly incomplete, manual install of Qt 5.6 that CMake is picking up instead of the OS-provided 5.5?

noah95, 2019-06-19:
Oh, I recently built cmake from source to get a most recent version. Just purged it and reinstalled from apt, doesn't make a difference (using 3.5.1 now).

My `LD_LIBRARY_PATH` is on `/usr/local/cuda-8.0/lib64`. I installed Qt 5.13.0 using the unified installer GUI, pointed LD path to the Qt installation and now it works. But not the way I like with the installer...

daveshah1, 2019-06-19:
It is most likely there is a `Qt5Config.cmake` from version 5.6 somewhere that is causing this problem - note that this is provided by Qt rather than CMake

daveshah1, 2019-07-05:
Following another report of an issue, the problem in that case seemed to be a conda provided cmake taking precedence over the distro-provided one.

noah95, 2019-07-05:
Okay, anaconda was providing some `qmake`. After removing anaconda, qmake is used from `/usr/bin/qmake` but `version `Qt_5.6' not found` is still occuring.

I just purged all packages containint `*qt*`, reinstalled `qt5-default` but still get the same error

daveshah1, 2019-07-05:
Have you tried removing CMakeLists.txt and CMakeFiles then rebuilding?

`grep -i Qt CMakeCache.txt` should tell you where it is getting the Qt cmake config from.

noah95, 2019-07-05:
Thank god, it's working - finally!

I did not rebuild after removing anaconda. Thank you very much for all the help!

closed 2019-07-05 by noah95