YosysHQ/nextpnr #295: ODDRX1F is broken

whitequark, 2019-06-25:
MCVE:  [build.zip](https://github.com/YosysHQ/nextpnr/files/3326098/build.zip)

Crash:
```
Info: Importing module top
Info: Rule checker, verifying imported design
Info: Checksum: 0xa87f1bc6

Info: constraining clock net 'clk100_0__p' to 100.00 MHz

Info: Packing IOs..
Info: test_0__io feeds TRELLIS_IO $3, removing $nextpnr_obuf test_0__io.
Info: pin '$3' constrained to Bel 'X44/Y0/PIOA'.
Info: clk100_0__p feeds TRELLIS_IO $1, removing $nextpnr_ibuf clk100_0__p.
Info: pin '$1' constrained to Bel 'X0/Y68/PIOC'.
Info: IOLOGIC component $2 connected to PIO Bel X44/Y0/PIOA
terminate called after throwing an instance of 'nextpnr_ecp5::assertion_failure'
  what():  Assertion failure: tgt != BelId() (/home/whitequark/Projects/nextpnr/ecp5/cells.cc:54)
Aborted
```

daveshah1, 2019-06-25:
I can't reproduce this. Looking at where the assertion failure is coming from, it looks like the database nextpnr is using might be from before IOLOGICs were added.

Can you make sure that prjtrellis/libtrellis is up to date, and force a database rebuild by touching ecp5/trellis_import.py before rebuilding nextpnr?

whitequark, 2019-06-25:
@daveshah1 Thanks for looking. Indeed my database was outdated.

closed 2019-06-25 by whitequark