YosysHQ/nextpnr #296: bba generation must be explicitly serialized

whitequark, 2019-06-25:
Generating `chipdb-85k.bba` results in a Python process that allocates almost 4 GB of RAM. If for some reason, nextpnr-ecp5 is built with `-j2` or greater (which is desirable as a default set in `MAKEFLAGS`), even a powerful system (my laptop has 16 GB of RAM, much of which is used for things other than building nextpnr) can end up in a state where it needs a hard reset.

Although, arguably, this is Linux's failure to properly prioritize processes in its scheduler, the practical reality is that building nextpnr causes crashes, and nextpnr shouldn't do this.

eddiehung, 2019-06-26:
I don't disagree. But I think we need to make it optional (or auto detect?) for those with bigger or cloud machines....

cliffordwolf, 2019-06-26:
I agree with both of you. I think serialized is a good default. Unfortunately I have no idea how to do this in cmake.

whitequark, 2019-06-26:
I think I know how to do it, I'll do it in a moment.

PR #297 (2019-06-27): Serialize chipdb generation by default
Fixes #296.
commit 1b3c8ea9 (2019-06-27): CMake: serialize chipdb generation by default.

Fixes #296.
closed 2019-06-27 by daveshah1