YosysHQ/nextpnr #299: nextpnr-ice40 unable to process MAC and IO primitives

BPJWES, 2019-07-03:
It seems that nextpnr is unable to place 2 IO primititives an internal oscillator primitive and more three MAC primitives for the up5k. (Or at least that is the minimum amount of MAC primitives for which i have seen it happen, the axample attached uses five MAC primitives)
Nextpnr seems to complete the placement and routing, but in the final stages errors out with the following message:

'ERROR: clearing already set config bit IpConfig.CBIT_5'

which seems to suggest that nextpnr wants to somehow overwrite a previously defined primitive.

I am runnning the following versions:

Yosys 0.8+531 (git sha1 d4f77d40, clang 6.0.0-1ubuntu2 -fPIC -Os)
nextpnr-ice40 -- Next Generation Place and Route (git sha1 29adacf)

Attached are the files for a (somewhat) minimized example showing the problem.

[TEST.pcf.txt](https://github.com/YosysHQ/nextpnr/files/3353842/TEST.pcf.txt)
[CommandLineDump.txt](https://github.com/YosysHQ/nextpnr/files/3353820/CommandLineDump.txt)
[synth.ys.txt](https://github.com/YosysHQ/nextpnr/files/3353852/synth.ys.txt)
[TOP_LEVEL_NEXTPNR_BUG_MINIMIZED.v.txt](https://github.com/YosysHQ/nextpnr/files/3353855/TOP_LEVEL_NEXTPNR_BUG_MINIMIZED.v.txt)
[yosys_nextpnr_icepack.sh.txt](https://github.com/YosysHQ/nextpnr/files/3353863/yosys_nextpnr_icepack.sh.txt)

daveshah1, 2019-07-03:
Thanks for the report and testcase! This came from the odd swapping of config bits between one of the DSPs and the HFOSC not being fully defined in icestorm. It should be fixed by cliffordwolf/icestorm#224

BPJWES, 2019-07-03:
Thanks for the fast response.

I have tried the fix, after updating both nextpnr and icestorm to the most recent commit.

The fix works for both the example I provided, and the actual project I am working on.

closed 2019-07-03 by cliffordwolf