YosysHQ/nextpnr #303: Allow multiple PCF files

olofk, 2019-07-18:
For larger designs it can be nice to split up the pin constraints into multiple files so that they can be swapped in or out together with the associated RTL. For that reason it would be nice to be able to specify multiple PCF files on the command-line

Aclassifier, 2019-12-27:
Maybe that could open for refining as well. Like some definition of a pin at the lowest level (and a lowest level pindef pcf file for the actual board) and functionality next level (like in a usagedef pcf file). Like _io_icore_p54_ at the lowest level and then when used as _external_i2c_sda_ at the next. I learned in [https://forum.mystorm.uk/t/hard-to-find-p16-p17-p20-p21-in-blackice-mx-pcf/771/5](url) that this is not possible with arachne-pnr at least. So I want multiple pcf files plus this hierarchical type system. I am surprised that Verlig does not need this info and surprised that it's not already defined? Or is it?

nmz787, 2020-08-02:
Is it possible to include/import/inherit PCF files from inside another?

umarcor, 2020-10-28:
> For larger designs it can be nice to split up the pin constraints into multiple files so that they can be swapped in or out together with the associated RTL.

> Like some definition of a pin at the lowest level (and a lowest level pindef pcf file for the actual board) and functionality next level (like in a usagedef pcf file)

Precisely... https://github.com/hdl/constraints/#short-term-goal