YosysHQ/nextpnr #304: Default value for --package is error-prone

whitequark, 2019-07-19:
I've seen beginners (both in FPGAs and wrt the FOSS toolchain) have an issue where their design is mysteriosly not working because they forgot `--package`, and I made that mistake myself too. I think it should not have a default value.

daveshah1, 2019-08-08:
#311 is the first part of a solution to this, lets give people a month or two to fix their flows (if they pay attention to warnings) before removing this altogether.

eddiehung, 2019-08-12:
Low priority so I'll put it here:
`ice40/picorv32.sh` gives:

```
+ ../nextpnr-ice40 --hx8k --asc picorv32.asc --json picorv32.json
Warning: Use of default value for --package is deprecated. Please add '--package ct256' to arguments.
Info: Importing module top
Info: Rule checker, verifying imported design
Info: Checksum: 0x51279ff9
```

so perhaps our examples should contain packages too