YosysHQ/nextpnr #307: Absolute placement constraints are ignored for ECP5

DurandA, 2019-08-05:
[Absolute placement constraints](https://github.com/YosysHQ/nextpnr/blob/master/docs/constraints.md#absolute-placement-constraints) using the `(* BEL="X2/Y5/lc0" *)` notation are ignored with ECP5 while it works fine with iCE40.

I initially thought this was due to the _HeAP_ placer but constraints are also ignored by the _sa_ placer.

DurandA, 2019-08-06:
I am trying to make a hand-placed ring oscillator (similar to [https://github.com/YosysHQ/nextpnr/blob/master/ice40/examples/floorplan/floorplan.v](floorplan.v) from ice40/examples) based on chained LUT_4 on ECP5.

Since two LUTs are packed into a slice, I am not sure if the same approach can be used.

daveshah1, 2019-08-08:
As a workaround, a `TRELLIS_SLICE` could be used to bypass the packer.

e.g. (not tested)

` (* BEL="X2/Y5/SLICEA" *) TRELLIS_SLICE #(.LUT0_INITVAL(1)) slice_i (.A0(a), .B0(1'b0), .C0(1'b0), .D0(1'b0), .F0(out));

DurandA, 2019-08-08:
It doesn't seem to work either (even with long chain of inverters of odd length). I used this [sim implementation](https://github.com/SymbiFlow/symbiflow-arch-defs/blob/d1c040a261e97a4542e57d2bf17d70ef081f4a74/ecp5/cells/slice/TRELLIS_SLICE.sim.v#L58-L73) as documentation. According to [FPGA Libraries Reference Guide](https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=1&ved=2ahUKEwiJpuPznvTjAhWCposKHbV0DOMQFjAAegQIAhAC&url=http%3A%2F%2Fwww.latticesemi.com%2F-%2Fmedia%2FLatticeSemi%2FDocuments%2FUserManuals%2FEI%2FFPGALibrariesReferenceGuide33.ashx%3Fdocument_id%3D50790&usg=AOvVaw21rgR6Qg6bsp4-Eos6Zq_3) , _LUT0/1_INITVAL_ seems to be have the same format that LUT_INIT from iCE40's SB_LUT4 so I suppose that .LUT0_INITVAL(16'h0001) is really an inverter. I am missing something?

DurandA, 2019-08-21:
While placement constraints are not taken into account for LUT4-8, placement constraints work for `TRELLIS_SLICE` so I close this issue.

> It doesn't seem to work either (even with long chain of inverters of odd length). I used this [sim [...]

see https://github.com/DurandA/verilog-buildingblocks/issues/1#issuecomment-523573379

closed 2019-08-21 by DurandA