YosysHQ/nextpnr #310: nextpnr-ecp5 throws std::out_of_range when using TRELLIS_SLICE

DurandA, 2019-08-08:
_nextpnr-ecp5_ thrown the following error:
```
terminate called after throwing an instance of 'std::out_of_range'
  what():  _Map_base::at
```

The error was generated by nextpnr e0a114fcb3389108bd5821e442b8deb78b532da7 using the output from [_yosys_@ac2fc3a](ac2fc3a144fe1094bedcc6b3fda8a498ad43ae76).

```
module top(input clk, input btn, output [7:0] led);
	wire [100:0] buffers_in, buffers_out, interlut;
	assign buffers_in = {buffers_out[99:0], buffers_out[100]};

	(* keep *)
	TRELLIS_SLICE #(
		.LUT0_INITVAL(16'd1),
		.LUT1_INITVAL(16'd2)
	) buffers [100:0] (	
		.A0(buffers_in),
		.B0(1'b0),
		.C0(1'b0),
		.D0(1'b0),
		.F0(buffers_out)
	);

	reg [28:0] counter = 0;
	always @(posedge buffers_out[100]) begin
		counter <= counter + 1;
	end

	assign led[7] = counter[28];
	assign led[6] = counter[27];
	assign led[5] = counter[26];

endmodule
```

This can be reproduced using the following output using `nextpnr-ecp5 --json ringosc.json --textcfg ringosc_out.config --um5g-85k --package CABGA381 --lpf ecp5evn.lpf --ignore-loops`: [ringosc.json.gz](https://github.com/YosysHQ/nextpnr/files/3483495/ringosc.json.gz)

closed 2019-08-08 by daveshah1

DurandA, 2019-08-08:
Thanks! Sorry for spamming issues, I hope this will be useful to other people.

daveshah1, 2019-08-08:
No problem! Bug reports with these kind of easily reproducible examples are always welcome!

commit f0abbc71 (2019-08-08): ecp5: Fix handling of missing ports in LUT permutation

Fixes #310