YosysHQ/nextpnr #318: Instalation Problem

andres091096, 2019-08-23:
Hi, i hace yhis issue when i try to install nextpnr. 

andres@andres-HP-Notebook:~/nextpnr$ make -j$(nproc)
[  1%] Automatic moc for target gui_ice40
[  2%] Automatic moc for target QtPropertyBrowser
[  4%] Built target bbasm
make[2]: *** No hay ninguna regla para construir el objetivo '/usr/local/share/icebox/chipdb-u4k.txt', necesario para 'ice40/chipdbs/chipdb-u4k.bba'.  Alto.
make[2]: *** Se espera a que terminen otras tareas....
[  5%] Generating ice40/chipdbs/chipdb-384.bba
[  5%] Built target gui_ice40_automoc
[  5%] Built target QtPropertyBrowser_automoc
[ 18%] Built target QtPropertyBrowser
[ 53%] Built target gui_ice40
Traceback (most recent call last):
  File "/home/andres/nextpnr/ice40/chipdb.py", line 254, in <module>
    slow_timings = read_timings(args.slow)
  File "/home/andres/nextpnr/ice40/chipdb.py", line 233, in read_timings
    with open(filename) as f:
FileNotFoundError: [Errno 2] No such file or directory: '/usr/local/share/icebox/timings_lp384.txt'
CMakeFiles/ice40_chipdb.dir/build.make:94: fallo en las instrucciones para el objetivo 'ice40/chipdbs/chipdb-384.bba'
make[2]: *** [ice40/chipdbs/chipdb-384.bba] Error 1
CMakeFiles/Makefile2:143: fallo en las instrucciones para el objetivo 'CMakeFiles/ice40_chipdb.dir/all'
make[1]: *** [CMakeFiles/ice40_chipdb.dir/all] Error 2
Makefile:127: fallo en las instrucciones para el objetivo 'all'
make: *** [all] Error 2
``

andres091096, 2019-08-23:
In the  /usr/local/share/icebox folder I have this files:

chipdb-1k.txt  chipdb-384.txt  chipdb-5k.txt  chipdb-8k.txt

daveshah1, 2019-08-23:
You need to install a newer (latest git master preferably) version of icestorm.

andres091096, 2019-08-23:
> You need to install a newer (latest git master preferably) version of icestorm.

But, i need unistall the previous version?

daveshah1, 2019-08-23:
No, `sudo make install` should overwrite the old files.

andres091096, 2019-08-23:
Thanks, it work :)

closed 2019-08-23 by andres091096