YosysHQ/nextpnr #322: macOS: No version of Boost::Python 3.x could be found

nalzok, 2019-09-07:
I got the following error when trying to install **nextpnr-ice40**

```
$ cmake -DARCH=ice40 .
-- The C compiler identification is AppleClang 10.0.1.10010046
-- The CXX compiler identification is AppleClang 10.0.1.10010046
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PythonInterp: /usr/local/bin/python3 (found suitable version "3.7.4", minimum required is "3.5")
-- Found PythonLibs: /usr/local/Frameworks/Python.framework/Versions/3.7/lib/libpython3.7m.dylib (found suitable version "3.7.4", minimum required is "3.5")
CMake Warning at /usr/local/lib/cmake/boost_filesystem-1.71.0/libboost_filesystem-variant-shared.cmake:59 (message):
  Target Boost::filesystem already has an imported location
  '/usr/local/lib/libboost_filesystem-mt.dylib', which will be overwritten
  with '/usr/local/lib/libboost_filesystem.dylib'
Call Stack (most recent call first):
  /usr/local/lib/cmake/boost_filesystem-1.71.0/boost_filesystem-config.cmake:43 (include)
  /usr/local/lib/cmake/Boost-1.71.0/BoostConfig.cmake:117 (find_package)
  /usr/local/lib/cmake/Boost-1.71.0/BoostConfig.cmake:182 (boost_find_component)
  /usr/local/Cellar/cmake/3.15.2/share/cmake/Modules/FindBoost.cmake:422 (find_package)
  CMakeLists.txt:106 (find_package)

CMake Warning at /usr/local/lib/cmake/boost_program_options-1.71.0/libboost_program_options-variant-shared.cmake:59 (message):
  Target Boost::program_options already has an imported location
  '/usr/local/lib/libboost_program_options-mt.dylib', which will be
  overwritten with '/usr/local/lib/libboost_program_options.dylib'
Call Stack (most recent call first):
  /usr/local/lib/cmake/boost_program_options-1.71.0/boost_program_options-config.cmake:43 (include)
  /usr/local/lib/cmake/Boost-1.71.0/BoostConfig.cmake:117 (find_package)
  /usr/local/lib/cmake/Boost-1.71.0/BoostConfig.cmake:182 (boost_find_component)
  /usr/local/Cellar/cmake/3.15.2/share/cmake/Modules/FindBoost.cmake:422 (find_package)
  CMakeLists.txt:106 (find_package)

CMake Warning at /usr/local/lib/cmake/boost_iostreams-1.71.0/libboost_iostreams-variant-shared.cmake:59 (message):
  Target Boost::iostreams already has an imported location
  '/usr/local/lib/libboost_iostreams-mt.dylib', which will be overwritten
  with '/usr/local/lib/libboost_iostreams.dylib'
Call Stack (most recent call first):
  /usr/local/lib/cmake/boost_iostreams-1.71.0/boost_iostreams-config.cmake:43 (include)
  /usr/local/lib/cmake/Boost-1.71.0/BoostConfig.cmake:117 (find_package)
  /usr/local/lib/cmake/Boost-1.71.0/BoostConfig.cmake:182 (boost_find_component)
  /usr/local/Cellar/cmake/3.15.2/share/cmake/Modules/FindBoost.cmake:422 (find_package)
  CMakeLists.txt:106 (find_package)

-- Found Boost: /usr/local/lib/cmake/Boost-1.71.0/BoostConfig.cmake (found version "1.71.0") found components:  filesystem thread program_options iostreams
-- Found OpenGL: /System/Library/Frameworks/OpenGL.framework
CMake Error at CMakeLists.txt:182 (message):
  No version of Boost::Python 3.x could be found.

-- Configuring incomplete, errors occurred!
See also "/Users/nalzok/Developer/nextpnr/CMakeFiles/CMakeOutput.log".
```

In case you need CMakeOutput.log: [CMakeOutput.log](https://github.com/YosysHQ/nextpnr/files/3586161/CMakeOutput.log)

Both `boost` and `boost-python3` has been installed via Homebrew

```
$ brew info boost boost-python3
boost: stable 1.71.0 (bottled), HEAD
Collection of portable C++ source libraries
https://www.boost.org/
/usr/local/Cellar/boost/1.71.0 (14,377 files, 522.8MB) *
  Poured from bottle on 2019-09-07 at 10:17:36
From: https://github.com/Homebrew/homebrew-core/blob/master/Formula/boost.rb
==> Dependencies
Required: icu4c ✔
==> Options
--HEAD
	Install HEAD version

boost-python3: stable 1.71.0 (bottled), HEAD
C++ library for C++/Python3 interoperability
https://www.boost.org/
/usr/local/Cellar/boost-python3/1.71.0 (460 files, 17.9MB) *
  Poured from bottle on 2019-09-07 at 10:17:57
From: https://github.com/Homebrew/homebrew-core/blob/master/Formula/boost-python3.rb
==> Dependencies
Required: boost ✔, python ✔
==> Options
--HEAD
	Install HEAD version
==> Analytics
install: 1,517 (30 days), 5,566 (90 days), 19,197 (365 days)
install_on_request: 775 (30 days), 2,897 (90 days), 11,172 (365 days)
build_error: 0 (30 days)
```

fred260571, 2019-09-07:
I used these lines and it worked like a charm on MacOS 10.11.6
brew install cmake python boost boost-python3 qt5 eigen

cmake -DARCH=ice40 -DICEBOX_ROOT=/usr/local/share/icebox  -DCMAKE_INSTALL_PREFIX=/usr/local . -DBUILD_PYTHON=OFF -DBUILD_GUI=OFF

nalzok, 2019-09-07:
@fred260571 Thanks, it works! However, I guess I won't be able to use the GUI with these parameters? In that case, I would consider it a _workaround_ instead of a _solution_.

Also, some nasty warnings persist. While they are not harmful, something must be wrong.

```
$ cmake -DARCH=ice40 -DICEBOX_ROOT=/usr/local/share/icebox -DCMAKE_INSTALL_PREFIX=/usr/local . -DBUILD_PYTHON=OFF -DBUILD_GUI=OFF
CMake Warning at /usr/local/lib/cmake/boost_filesystem-1.71.0/libboost_filesystem-variant-shared.cmake:59 (message):
  Target Boost::filesystem already has an imported location
  '/usr/local/lib/libboost_filesystem-mt.dylib', which will be overwritten
  with '/usr/local/lib/libboost_filesystem.dylib'
Call Stack (most recent call first):
  /usr/local/lib/cmake/boost_filesystem-1.71.0/boost_filesystem-config.cmake:43 (include)
  /usr/local/lib/cmake/Boost-1.71.0/BoostConfig.cmake:117 (find_package)
  /usr/local/lib/cmake/Boost-1.71.0/BoostConfig.cmake:182 (boost_find_component)
  /usr/local/Cellar/cmake/3.15.2/share/cmake/Modules/FindBoost.cmake:422 (find_package)
  CMakeLists.txt:106 (find_package)

CMake Warning at /usr/local/lib/cmake/boost_program_options-1.71.0/libboost_program_options-variant-shared.cmake:59 (message):
  Target Boost::program_options already has an imported location
  '/usr/local/lib/libboost_program_options-mt.dylib', which will be
  overwritten with '/usr/local/lib/libboost_program_options.dylib'
Call Stack (most recent call first):
  /usr/local/lib/cmake/boost_program_options-1.71.0/boost_program_options-config.cmake:43 (include)
  /usr/local/lib/cmake/Boost-1.71.0/BoostConfig.cmake:117 (find_package)
  /usr/local/lib/cmake/Boost-1.71.0/BoostConfig.cmake:182 (boost_find_component)
  /usr/local/Cellar/cmake/3.15.2/share/cmake/Modules/FindBoost.cmake:422 (find_package)
  CMakeLists.txt:106 (find_package)

CMake Warning at /usr/local/lib/cmake/boost_iostreams-1.71.0/libboost_iostreams-variant-shared.cmake:59 (message):
  Target Boost::iostreams already has an imported location
  '/usr/local/lib/libboost_iostreams-mt.dylib', which will be overwritten
  with '/usr/local/lib/libboost_iostreams.dylib'
Call Stack (most recent call first):
  /usr/local/lib/cmake/boost_iostreams-1.71.0/boost_iostreams-config.cmake:43 (include)
  /usr/local/lib/cmake/Boost-1.71.0/BoostConfig.cmake:117 (find_package)
  /usr/local/lib/cmake/Boost-1.71.0/BoostConfig.cmake:182 (boost_find_component)
  /usr/local/Cellar/cmake/3.15.2/share/cmake/Modules/FindBoost.cmake:422 (find_package)
  CMakeLists.txt:106 (find_package)

-- Configuring architecture : ice40
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/nalzok/Developer/nextpnr
```

fred260571, 2019-09-07:
maybe it could help
[https://github.com/YosysHQ/nextpnr/issues/198](url)

nalzok, 2019-09-07:
Can you check if the link is correct? It points me to `https://github.com/YosysHQ/nextpnr/issues/url`

fred260571, 2019-09-07:
it should have linked to issue #198
But you have already found it :)

daveshah1, 2019-09-11:
Can you have a look at `/usr/local/lib/libboost*` and see what the Boost::Python 3 library for OS X is called? There are several different naming conventions in force.

nalzok, 2019-09-11:
@daveshah1 I'm not sure which one are you referring to, but it should be `libboost_python37-mt.a`, or `libboost_python37-mt.dylib`, or `libboost_python37.a`, or `libboost_python37.dylib`. See below for a comprehensive output.

```
$ ls /usr/local/lib/libboost*
/usr/local/lib/libboost_atomic-mt.a
/usr/local/lib/libboost_atomic-mt.dylib
/usr/local/lib/libboost_chrono-mt.a
/usr/local/lib/libboost_chrono-mt.dylib
/usr/local/lib/libboost_chrono.a
/usr/local/lib/libboost_chrono.dylib
/usr/local/lib/libboost_container-mt.a
/usr/local/lib/libboost_container-mt.dylib
/usr/local/lib/libboost_container.a
/usr/local/lib/libboost_container.dylib
/usr/local/lib/libboost_context-mt.a
/usr/local/lib/libboost_context-mt.dylib
/usr/local/lib/libboost_contract-mt.a
/usr/local/lib/libboost_contract-mt.dylib
/usr/local/lib/libboost_contract.a
/usr/local/lib/libboost_contract.dylib
/usr/local/lib/libboost_coroutine-mt.a
/usr/local/lib/libboost_coroutine-mt.dylib
/usr/local/lib/libboost_coroutine.a
/usr/local/lib/libboost_coroutine.dylib
/usr/local/lib/libboost_date_time-mt.a
/usr/local/lib/libboost_date_time-mt.dylib
/usr/local/lib/libboost_date_time.a
/usr/local/lib/libboost_date_time.dylib
/usr/local/lib/libboost_exception-mt.a
/usr/local/lib/libboost_exception.a
/usr/local/lib/libboost_fiber-mt.a
/usr/local/lib/libboost_fiber-mt.dylib
/usr/local/lib/libboost_filesystem-mt.a
/usr/local/lib/libboost_filesystem-mt.dylib
/usr/local/lib/libboost_filesystem.a
/usr/local/lib/libboost_filesystem.dylib
/usr/local/lib/libboost_graph-mt.a
/usr/local/lib/libboost_graph-mt.dylib
/usr/local/lib/libboost_graph.a
/usr/local/lib/libboost_graph.dylib
/usr/local/lib/libboost_iostreams-mt.a
/usr/local/lib/libboost_iostreams-mt.dylib
/usr/local/lib/libboost_iostreams.a
/usr/local/lib/libboost_iostreams.dylib
/usr/local/lib/libboost_locale-mt.a
/usr/local/lib/libboost_locale-mt.dylib
/usr/local/lib/libboost_log-mt.a
/usr/local/lib/libboost_log-mt.dylib
/usr/local/lib/libboost_log.a
/usr/local/lib/libboost_log.dylib
/usr/local/lib/libboost_log_setup-mt.a
/usr/local/lib/libboost_log_setup-mt.dylib
/usr/local/lib/libboost_log_setup.a
/usr/local/lib/libboost_log_setup.dylib
/usr/local/lib/libboost_math_c99-mt.a
/usr/local/lib/libboost_math_c99-mt.dylib
/usr/local/lib/libboost_math_c99.a
/usr/local/lib/libboost_math_c99.dylib
/usr/local/lib/libboost_math_c99f-mt.a
/usr/local/lib/libboost_math_c99f-mt.dylib
/usr/local/lib/libboost_math_c99f.a
/usr/local/lib/libboost_math_c99f.dylib
/usr/local/lib/libboost_math_c99l-mt.a
/usr/local/lib/libboost_math_c99l-mt.dylib
/usr/local/lib/libboost_math_c99l.a
/usr/local/lib/libboost_math_c99l.dylib
/usr/local/lib/libboost_math_tr1-mt.a
/usr/local/lib/libboost_math_tr1-mt.dylib
/usr/local/lib/libboost_math_tr1.a
/usr/local/lib/libboost_math_tr1.dylib
/usr/local/lib/libboost_math_tr1f-mt.a
/usr/local/lib/libboost_math_tr1f-mt.dylib
/usr/local/lib/libboost_math_tr1f.a
/usr/local/lib/libboost_math_tr1f.dylib
/usr/local/lib/libboost_math_tr1l-mt.a
/usr/local/lib/libboost_math_tr1l-mt.dylib
/usr/local/lib/libboost_math_tr1l.a
/usr/local/lib/libboost_math_tr1l.dylib
/usr/local/lib/libboost_numpy37-mt.a
/usr/local/lib/libboost_numpy37-mt.dylib
/usr/local/lib/libboost_numpy37.a
/usr/local/lib/libboost_numpy37.dylib
/usr/local/lib/libboost_prg_exec_monitor-mt.a
/usr/local/lib/libboost_prg_exec_monitor-mt.dylib
/usr/local/lib/libboost_prg_exec_monitor.a
/usr/local/lib/libboost_prg_exec_monitor.dylib
/usr/local/lib/libboost_program_options-mt.a
/usr/local/lib/libboost_program_options-mt.dylib
/usr/local/lib/libboost_program_options.a
/usr/local/lib/libboost_program_options.dylib
/usr/local/lib/libboost_python37-mt.a
/usr/local/lib/libboost_python37-mt.dylib
/usr/local/lib/libboost_python37.a
/usr/local/lib/libboost_python37.dylib
/usr/local/lib/libboost_random-mt.a
/usr/local/lib/libboost_random-mt.dylib
/usr/local/lib/libboost_random.a
/usr/local/lib/libboost_random.dylib
/usr/local/lib/libboost_regex-mt.a
/usr/local/lib/libboost_regex-mt.dylib
/usr/local/lib/libboost_regex.a
/usr/local/lib/libboost_regex.dylib
/usr/local/lib/libboost_serialization-mt.a
/usr/local/lib/libboost_serialization-mt.dylib
/usr/local/lib/libboost_serialization.a
/usr/local/lib/libboost_serialization.dylib
/usr/local/lib/libboost_stacktrace_addr2line-mt.a
/usr/local/lib/libboost_stacktrace_addr2line-mt.dylib
/usr/local/lib/libboost_stacktrace_addr2line.a
/usr/local/lib/libboost_stacktrace_addr2line.dylib
/usr/local/lib/libboost_stacktrace_basic-mt.a
/usr/local/lib/libboost_stacktrace_basic-mt.dylib
/usr/local/lib/libboost_stacktrace_basic.a
/usr/local/lib/libboost_stacktrace_basic.dylib
/usr/local/lib/libboost_stacktrace_noop-mt.a
/usr/local/lib/libboost_stacktrace_noop-mt.dylib
/usr/local/lib/libboost_stacktrace_noop.a
/usr/local/lib/libboost_stacktrace_noop.dylib
/usr/local/lib/libboost_system-mt.a
/usr/local/lib/libboost_system-mt.dylib
/usr/local/lib/libboost_system.a
/usr/local/lib/libboost_system.dylib
/usr/local/lib/libboost_test_exec_monitor-mt.a
/usr/local/lib/libboost_test_exec_monitor.a
/usr/local/lib/libboost_thread-mt.a
/usr/local/lib/libboost_thread-mt.dylib
/usr/local/lib/libboost_timer-mt.a
/usr/local/lib/libboost_timer-mt.dylib
/usr/local/lib/libboost_timer.a
/usr/local/lib/libboost_timer.dylib
/usr/local/lib/libboost_type_erasure-mt.a
/usr/local/lib/libboost_type_erasure-mt.dylib
/usr/local/lib/libboost_type_erasure.a
/usr/local/lib/libboost_type_erasure.dylib
/usr/local/lib/libboost_unit_test_framework-mt.a
/usr/local/lib/libboost_unit_test_framework-mt.dylib
/usr/local/lib/libboost_unit_test_framework.a
/usr/local/lib/libboost_unit_test_framework.dylib
/usr/local/lib/libboost_wave-mt.a
/usr/local/lib/libboost_wave-mt.dylib
/usr/local/lib/libboost_wave.a
/usr/local/lib/libboost_wave.dylib
/usr/local/lib/libboost_wserialization-mt.a
/usr/local/lib/libboost_wserialization-mt.dylib
/usr/local/lib/libboost_wserialization.a
/usr/local/lib/libboost_wserialization.dylib
```

hanhha, 2019-09-14:
I also got this issue when I built on Mac OS.
I have no idea about the root cause and have no knowledge about CMake, so I tried this dirty workaround (please see my + lines):
1 - Fix Boost_PYTHON_FOUND to TRUE to overcome the check.
2 - Add library to link_libraries to make it link-able.

```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 67b0c18..db3515d 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -183,7 +183,8 @@ if (BUILD_PYTHON)
             set(Boost_PYTHON_FOUND TRUE)
         endif ()
     endif ()
-
+
+               set(Boost_PYTHON_FOUND TRUE) # Temporarily ignore
     if (NOT Boost_PYTHON_FOUND )
         message( FATAL_ERROR "No version of Boost::Python 3.x could be found.")
     endif ()
@@ -273,6 +274,8 @@ foreach (family ${ARCH})
         endif()
         if (BUILD_PYTHON)
             target_link_libraries(${target} LINK_PUBLIC ${PYTHON_LIBRARIES})
+            target_link_libraries(${target} LINK_PUBLIC -lboost_python37)
+            target_link_libraries(${target} LINK_PUBLIC -L/usr/local/lib)
             if (STATIC_BUILD)
                 target_link_libraries(${target} LINK_PUBLIC ${CMAKE_THREAD_LIBS_INIT} ${CMAKE_DL_LIBS} ${ZLIB_LIBRARIES} ${EXPAT_LIBRARIES} -lutil)
                 if(${CMAKE_SYSTEM_NAME} MATCHES "Darwin")
```
It worked for me. Hope it would help :D

nalzok, 2019-09-15:
@hanhha Thanks, it also works on my machine (MacBook Pro 2017, macOS Mojave)!

satnam6502, 2019-09-19:
I had the same issue. I did:
```
cmake -DARCH=ice40 -DICEBOX_ROOT=/usr/local/share/icebox -DCMAKE_INSTALL_PREFIX=/usr/local . -DBUILD_PYTHON=OFF -DBUILD_GUI=OFF
```
which unblocked me but it would be nice if this worked out of the box?

mmicko, 2019-09-19:
Sorry all, in process of getting Mac Mini so we can do better support for macOS

satnam6502, 2019-09-19:
No need to apologize! I'm very appreciative of the work that has gone into making this work on OS X! Thank you! It's very cool.

daveshah1, 2019-09-19:
Another option is to use the static binaries built by @xobs at https://github.com/xobs/toolchain-nextpnr-ice40/releases (these have Python but not the GUI, afaik, as they are intended for integration deeper into other flows)

satnam6502, 2019-09-20:
I'm in no hurry, I'm unblocked enough to get my silly wee Coq -> Bluespec -> Lattice FPGA example working with only open source tools: https://github.com/project-oak/oak-hardware/tree/master/kami/icestick-counter

closed 2019-09-28 by mmicko

mmicko, 2019-09-28:
For those that are interested, this issue was due to cmake files distributed with boost itself. Placing this option (Boost_NO_BOOST_CMAKE) makes sure those files are not used, but default ones from cmake that are valid.

nalzok, 2019-12-08:
Hi there, I wonder why did @daveshah1 revert the change in https://github.com/YosysHQ/nextpnr/commit/e9cced57bf210faa4674a78358cb1a94765b34db?

daveshah1, 2019-12-08:
I didn't revert it, I applied it to all arches as it was causing problems on Arch Linux too

nalzok, 2019-12-08:
Ah sorry @daveshah1, I was confused and thought you enabled boost on all platforms instead of disabling it!

commit 7cd1e049 (2019-09-28): Fix issue with latest boost on macOS, fixes #322