YosysHQ/nextpnr #323: Document how to use `EXTERNAL_CHIPDB`

xobs, 2019-09-07:
The `nextpnr` CMakeLists.txt has a parameter `EXTERNAL_CHIPDB`.  This appears to take a path to existing chipdbs, ostensibly for use without statically linking them into the resulting binary.

Compiling the `.bba` files takes a very long time, so it would be nice to be able to do this once and then never have to do it again.  However, `EXTERNAL_CHIPDB` appears to expect `.bin` files.

Please document how to use `EXTERNAL_CHIPDB` and `EXTERNAL_CHIPDB_ROOT` in order to speed up compiles / cross-compile nextpnr.

0xDRRB, 2019-11-13:
After a first build, you can create *.bin files from ice40/chipdbs/*.cc with a simple C code writing arrays to files, on a first host (amd64 machine for example)
Then use -DEXTERNAL_CHIPDB=1 and -DEXTERNAL_CHIPDB_ROOT=/some/path to build on another host (Raspberry Pi in my case).
nextpnr-ice40 will read bin files from /some/path/ice40 and do his job.

whitequark, 2020-06-29:
I've documented this: https://github.com/YosysHQ/nextpnr#pre-generating-chip-databases

whitequark, 2020-06-29:
Note that I changed the mechanism to work slightly differently than the one people refer to in this issue. Specifically, *pregenerated chipdb* (offloading the time-consuming step of generating bba files) and *external chipdb* (avoiding the embedding of chipdb into nextpnr) are now separate steps.

closed 2020-06-29 by whitequark