YosysHQ/nextpnr #324: clock constraints not propagating across TRELLIS_IO or SB_GB_IO

dlharmon, 2019-09-10:
When a clock signal passes through a TRELLIS_IO or SB_GB_IO, it appears to not propagate the clock constraint. 

Replacing the TRELLIS_IO with a verilog `assign` statement results in correct clock constraint propagation.

[nMigen blinky](https://github.com/m-labs/nmigen/blob/master/examples/board/blinky.py) example on iCE40, built with output:
[blinky_ice40.zip](https://github.com/YosysHQ/nextpnr/files/3597104/blinky_ice40.zip)
same example, modified to use Versa ECP5 board:
[blinky_ecp5.zip](https://github.com/YosysHQ/nextpnr/files/3597138/blinky_ecp5.zip)

There is a shell script in both to rerun. The nextpnr output is in top.tim in both.

dlharmon, 2019-09-10:
A few things I forgot:

nextpnr built yesterday `git sha1 c0f02561`

Adding a clock constraint to the output of the input buffers resulted in correct clock propagation through an EHXPLLL and CLKDIVF in an ECP5 design with fairly complex clocking.

daveshah1, 2019-09-13:
I think #328 should fix both cases

PR #328 (2019-09-13): Move clock constraints across buffers in more cases
Fixes #324