YosysHQ/nextpnr #333: JSON Frontend/backend Improvements Tracking

daveshah1, 2019-09-22:
In order to support more advanced flows such as partial reconfiguration and incremental builds, the JSON frontend needs to be improved to support things like:
 - hierarchy (currently a flatttened netlist only is allowed)
 - improved handling of user-specified blackboxes
 - improved handling of attributes in some places

As the current frontend has had significant modifications since its original development anyway, it is likely time for a rewrite. At the same time it might be worth considering if other approaches than JSON (e.g. protobufs) are better suited to fast design checkpoint load/save type situations.

mithro, 2019-09-23:
@daveshah1 - It would be good if we could come to some idea around common file standards between nextpnr and Verilog to Routing. 

@duck2 has been working on improving the Verilog to Routing code to be [auto-generated from an XML schemas](https://github.com/duck2/uxsdcxx). This has also enabled the usage of both XML and [Cap’n Proto](https://capnproto.org/). @duck2 [published a report about this work here](https://docs.google.com/document/d/1SZm44g-9Bo-xD2lDfXcxb8bpzYjewGyK7bVyOYj5Gqs/edit).

@duck2 will be at OrConf and giving a short talk about his work, so would be good to chat with him there.

mithro, 2019-09-23:
It should be possible to also generate [a JSON schema](https://json-schema.org/) from the XML Schema Definition and a compatible C++ JSON parser + generator?

mithro, 2019-09-23:
@daveshah1 - What do you use for json parser? It seems to be hand written? As you are already usinf boost, have you consider using PropertyTrees?
 * https://www.boost.org/doc/libs/1_65_1/doc/html/property_tree.html
 * http://techgate.fr/boost-property-tree/
 * https://theboostcpplibraries.com/boost.propertytree

mithro, 2019-09-23:
@duck2 - I added https://github.com/duck2/uxsdcxx/issues/25 to consider adding a PropertyTree generator similar to the pugixml and Cap'n Proto generators.

duck2, 2019-09-23:
I like the idea of common file formats!

About property trees, I think they don't require a generator? Looks like their purpose is to enable a memory structure with many possible I/O formats.

gatecat, 2022-08-18:
the frontend rewrite covering these points (most importantly hierarchy) was done long ago in #353

closed 2022-08-18 by gatecat