YosysHQ/nextpnr #336: SB_PLL DELAY_ADJUSTMENT_MODE_FEEDBACK=DYNAMIC ignored

jordens, 2019-09-23:
The ice40 pll parameter `DELAY_ADJUSTMENT_MODE_FEEDBACK` is ignored by nextpnr. It is always set to "FIXED" in the resulting `.asc`.

# MVCE:

top.v:

```verilog
module top(input clock,
           output led);

wire clk1;
reg [7:0] delay = 8'd32;

SB_PLL40_2F_CORE #(
        .DELAY_ADJUSTMENT_MODE_FEEDBACK("DYNAMIC"),
        .DELAY_ADJUSTMENT_MODE_RELATIVE("DYNAMIC"),
        .DIVF(1'd0),
        .DIVQ(2'd2),
        .DIVR(1'd0),
        .ENABLE_ICEGATE_PORTA(1'd0),
        .ENABLE_ICEGATE_PORTB(1'd0),
        .FDA_FEEDBACK(1'd0),
        .FDA_RELATIVE(4'd15),
        .FEEDBACK_PATH("PHASE_AND_DELAY"),
        .FILTER_RANGE(2'd3),
        .PLLOUT_SELECT_PORTA("SHIFTREG_0deg"),
        .PLLOUT_SELECT_PORTB("GENCLK"),
        .SHIFTREG_DIV_MODE(1'd1)
) SB_PLL40_2F_CORE (
        .BYPASS(1'd0),
        .DYNAMICDELAY(delay),
        .LATCHINPUTVALUE(1'd0),
        .REFERENCECLK(clock),
        .RESETB(1'd1),
        // .LOCK(),
        .PLLOUTGLOBALA(clk1),
        // .PLLOUTGLOBALB(sr_clk)
);

assign led = clk1;

endmodule
```

top.pcf:

```
set_io led H3
set_io clock G1
```

Makefile:

```makefile
SOURCES = top.v

all: top.asc

top.json: $(SOURCES)
        yosys -q -p 'synth_ice40 -top top -json top.json' $(SOURCES)

top.asc: top.json top.pcf
        nextpnr-ice40 --hx8k --package ct256 --json top.json --pcf top.pcf --asc top.asc
```

And comparing `top.asc` for `.DELAY_ADJUSTMENT_MODE_FEEDBACK("DYNAMIC"),` and `.DELAY_ADJUSTMENT_MODE_FEEDBACK("FIXED"),`.

nextpnr: 395db49

daveshah1, 2019-09-23:
Thanks for reporting! Should be working now.

closed 2019-09-23 by daveshah1

jordens, 2019-09-24:
Works now. Thanks!

commit 30e3c846 (2019-09-23): ice40: Add support for PLL DELAY_ADJUSTMENT_MODE

Fixes #336