YosysHQ/nextpnr #347: Segfault when loading JSON netlist

Bugbrother, 2019-10-25:
When migrating from arachne-pnr to nextpnr, loading a netlist that was converted to JSON, a segfault is triggered at line 624 of `jsonparse.cc` :

```C++
if (GetSize(pdir_node->data_dict_keys) != GetSize(connections->data_dict_keys))
    log_error("JSON number of connections doesnt "
                  "match number of ports in node \'%s\' "
                  "of module \'%s\'\n",
                  cell->name.c_str(ctx), modname.c_str());
```

the `pdir_node` pointer is initialized to NULL at line 598 and and its value modified using:
```C++
if (cell_node->data_dict.count("port_directions") > 0) 
{
  pdir_node = cell_node->data_dict.at("port_directions");
  ...
}
else if(cell_node->data_dict.count("ports") > 0)
{
  pdir_node = cell_node->data_dict.at("ports");
  ...
}
```

The two conditions not being mutually exclusive, the pointer can remain set to NULL and when checked at line 624, trigger a segfault.

The netlist has been converted from BLIF to JSON using yosys (`read_blif netlist.blif; write_json netlist.json`) and was properly processed by arachne before. 

Not knowing the internals of the JSON parser, I'm not able to fix the issue myself...

daveshah1, 2019-10-27:
I'll look into a fix (a rewrite of the JSON parser is on my TODO list too). In the mean time a workaround should be `read_verilog -lib +/ice40/cells_sim.v; read_blif netlist.blif; write_json netlist.json` to make sure it has the iCE40 blackboxes available.

daveshah1, 2019-12-27:
Although nextpnr still cannot import a netlist without port directions (which are lost in the BLIF translation), I don't see this being a particular requirement and the above workaround should solve it anyway. Closing as the new frontend gives an appropriate error rather than a segfault.

closed 2019-12-27 by daveshah1