YosysHQ/nextpnr #348: ERROR: No wire found for port PADDT on destination cell USRMCLK.

xobs, 2019-10-25:
`nextpnr-ecp5` is now giving me the error `ERROR: No wire found for port PADDT on destination cell USRMCLK.`.  This is a new error, and is happening as of 371d33146fcfda9a1c568f46fd1f5775cd4fb1be.

I am using `nextpnr-ecp5` built on Windows from 371d33146fcfda9a1c568f46fd1f5775cd4fb1be and yosys 84982b308343315c889d3d00116db820a51cad78 built for Windows on an Ubuntu 18.04 machine.

The test Verilog file was generated by litex.  Here is the resulting `gateware` directory.  The top.v file is "only" 566 lines because I've stripped out everything but the CSR bus arbiter, scratch register, and memory-mapped SPI.

On ECP5, the SPI CLK line can be reused by a bit-banged implementation by utilizing the `USRMCLK` block, which gets instantiated on line 474.  This does not have a `PADDT` port.

[gateware.zip](https://github.com/YosysHQ/nextpnr/files/3772465/gateware.zip)

xobs, 2019-10-25:
Note that after updating both yosys and nextpnr, I've started to get `ERROR: timing analysis failed due to presence of combinatorial loops, incomplete specification of timing ports, etc.` messages.  This happened on my own design, which I "fixed" by removing functionality in https://github.com/xobs/haddecks/commit/802b3e3eb3339eb4375fdb83fe4da51cf7adcac0 -- the complaint was on the `d.o` line, which doesn't have any loops as far as I can see.

I'm seeing similar regressions in another ecp5 project,

It is unclear whether that problem lies in yosys or in nextpnr-ecp5.

daveshah1, 2019-10-25:
Can you confirm the bba files are as new as the nextpnr build? There was a missing dependency recently, now fixed, that meant a constid update wasn't triggering a bba rebuild. Stale bba files could easily cause this behaviour.

xobs, 2019-10-25:
That could be it.  Let me try regenerating them.  I will post results once that's done.

xobs, 2019-10-26:
Yes, regenerating the bba files solves the issue.  Nice to know what that particular failure looks like, and now I know when I need to regenerate them!

closed 2019-10-26 by xobs

daveshah1, 2019-10-26:
475fcd44 adds a much clearer error for this case in the future.

xobs, 2019-10-27:
Thanks! This'll ensure things fail in testing before we ship out packaged versions.