YosysHQ/nextpnr #349: NextPNR fails with 'std::out_of_range'

Spritetm, 2019-10-29:
This happened after I added input/output registers (OFS1P3DX, IFS1P3DX) to the IO pins of an interface, [here](https://github.com/Spritetm/hadbadge2019_fpgasoc/blob/nextpnr_fail_ioff/soc/top_fpga.v#L278). Pre-synthesized config file is [here](https://github.com/YosysHQ/nextpnr/files/3781203/soc_out_synth.config.gz).

```
$ecppack --svf-rowsize 100000 --svf soc.svf --input soc_out_synth.config
terminate called after throwing an instance of 'std::out_of_range'
  what():  map::at
Aborted
```

I may or may not have made a mistake in that code (still need to double-check it), however I don't think this is the failure mode anyone'd want in that situation.

Spritetm, 2019-10-29:
Derp, seems this got fixed in prjtrellis very recently, pull/rebuild fixed it.

closed 2019-10-29 by Spritetm

daveshah1, 2019-10-29:
Latest Trellis should have a slightly better message if this error is hit by something else again, too