YosysHQ/nextpnr #354: how to read the path report.

janrinze, 2019-11-18:
```
Info: Critical path report for clock 'doubleclk' (posedge -> posedge): 
Info: curr total
Info:  0.3  0.3  Source clk_TRELLIS_FF_Q_DI_LUT4_Z_SLICE.Q1
Info:  1.0  1.3    Net clk budget 4.480000 ns (67,2) -> (67,0)
Info:                Sink $gbuf$clk.CLKI
Info:  0.0  1.3  Source $gbuf$clk.CLKO
Info:  0.0  1.3    Net $glbnet$clk budget 4.480000 ns (67,0) -> (67,2)
Info:                Sink clk_TRELLIS_FF_Q_DI_LUT4_Z_SLICE.D1
Info:  0.1  1.4  Source clk_TRELLIS_FF_Q_DI_LUT4_Z_SLICE.F1
Info:  0.1  1.5    Net clk_TRELLIS_FF_Q_DI budget 4.479000 ns (67,2) -> (67,2)
Info:                Sink clk_TRELLIS_FF_Q_DI_LUT4_Z_SLICE.DI1
Info:  0.0  1.5  Setup clk_TRELLIS_FF_Q_DI_LUT4_Z_SLICE.DI1
Info: 0.4 ns logic, 1.1 ns routing

Info: Critical path report for clock '$glbnet$clk' (posedge -> posedge):
Info: curr total
Info:  0.3  0.3  Source mycpu.din_LUT4_Z_B_LUT4_B_Z_LUT4_B_Z_LUT4_Z_3_D_LUT4_C_SLICE.Q0
Info:  2.0  2.3    Net cpu_address[13] budget 0.821000 ns (26,69) -> (16,44)
Info:                Sink memory.1.3.0_DOB0_LUT4_A_SLICE.D1
Info:  0.1  2.5  Source memory.1.3.0_DOB0_LUT4_A_SLICE.F1
Info:  0.3  2.8    Net memory.8.3.0_DOB0_LUT4_A_C budget 0.821000 ns (16,44) -> (16,44)
Info:                Sink memory.8.3.0_DOB0_LUT4_A_C_LUT4_D_1_SLICE.D0
Info:  0.1  2.9  Source memory.8.3.0_DOB0_LUT4_A_C_LUT4_D_1_SLICE.F0
Info:  0.7  3.6    Net memory.8.3.0_DOB0_LUT4_A_C_LUT4_D_1_Z budget 0.905000 ns (16,44) -> (16,47)
Info:                Sink mycpu.din_LUT4_Z_2_D_LUT4_D_SLICE.C1
Info:  0.1  3.7  Source mycpu.din_LUT4_Z_2_D_LUT4_D_SLICE.F1
Info:  0.6  4.3    Net mycpu.din_LUT4_Z_C budget 1.259000 ns (16,47) -> (16,45)
Info:                Sink mycpu.din_PFUMX_Z_SLICE.D1
Info:  0.2  4.6  Source mycpu.din_PFUMX_Z_SLICE.OFX0
Info:  0.1  4.7    Net cpu_din[15] budget 1.525000 ns (16,45) -> (16,45)
Info:                Sink mycpu.din_PFUMX_Z_5_C0_PFUMX_C0_Z_LUT4_B_D_LUT4_C_Z_PFUMX_Z_BLUT_LUT4_Z_D_PFUMX_Z_SLICE.D1
Info:  0.2  4.9  Source mycpu.din_PFUMX_Z_5_C0_PFUMX_C0_Z_LUT4_B_D_LUT4_C_Z_PFUMX_Z_BLUT_LUT4_Z_D_PFUMX_Z_SLICE.OFX0
Info:  1.4  6.3    Net mycpu.din_PFUMX_Z_5_C0_PFUMX_C0_Z_LUT4_B_D_LUT4_C_Z_PFUMX_Z_BLUT_LUT4_Z_D budget 1.271000 ns (16,45) -> (22,59)
Info:                Sink mycpu.din_PFUMX_Z_5_C0_PFUMX_C0_Z_LUT4_B_D_LUT4_C_Z_PFUMX_Z_SLICE.D1
Info:  0.2  6.6  Source mycpu.din_PFUMX_Z_5_C0_PFUMX_C0_Z_LUT4_B_D_LUT4_C_Z_PFUMX_Z_SLICE.OFX0
Info:  1.2  7.7    Net mycpu.din_PFUMX_Z_5_C0_PFUMX_C0_Z_LUT4_B_D_LUT4_C_Z[15] budget 0.988000 ns (22,59) -> (24,69)
Info:                Sink memory.0.2.0_CEB_LUT4_Z_SLICE.D0
Info:  0.1  7.9  Source memory.0.2.0_CEB_LUT4_Z_SLICE.F0
Info:  2.0  9.9    Net memory.0.2.0_CEB budget 0.988000 ns (24,69) -> (49,46)
Info:                Sink memory.7.2.0.CEB
Info:  0.1  9.9  Setup memory.7.2.0.CEB
Info: 1.7 ns logic, 8.3 ns routing

Info: Critical path report for cross-domain path 'posedge $glbnet$clk' -> '<async>':
Info: curr total
Info:  0.3  0.3  Source tvs_TRELLIS_FF_Q_SLICE.Q0
Info:  2.2  2.5    Net VGAVS$TRELLIS_IO_OUT budget 13.579000 ns (3,48) -> (0,83)
Info:                Sink VGAVS$tr_io.I
Info: 0.3 ns logic, 2.2 ns routing

Info: Critical path report for cross-domain path 'posedge doubleclk' -> '<async>':
Info: curr total
Info:  0.3  0.3  Source VGACLK_TRELLIS_FF_Q_SLICE.Q0
Info:  0.6  0.9    Net VGACLK$TRELLIS_IO_OUT budget 13.579000 ns (2,89) -> (0,89)
Info:                Sink VGACLK$tr_io.I
Info: 0.3 ns logic, 0.6 ns routing

Info: Max frequency for clock   'doubleclk': 646.41 MHz (PASS at 72.00 MHz)
ERROR: Max frequency for clock '$glbnet$clk': 70.61 MHz (FAIL at 72.00 MHz)

Info: Max delay posedge $glbnet$clk -> <async>: 2.55 ns
Info: Max delay posedge doubleclk   -> <async>: 0.93 ns

Info: Slack histogram:
Info:  legend: * represents 18 endpoint(s)
Info:          + represents [1,18) endpoint(s)
Info: [  -275,    398) |*+
Info: [   398,   1071) |******+
Info: [  1071,   1744) |**************+
Info: [  1744,   2417) |******************+
Info: [  2417,   3090) |*******************+
Info: [  3090,   3763) |********************+
Info: [  3763,   4436) |**+
Info: [  4436,   5109) |*+
Info: [  5109,   5782) |**+
Info: [  5782,   6455) |****+
Info: [  6455,   7128) |*********+
Info: [  7128,   7801) |*****************+
Info: [  7801,   8474) |***********************************+
Info: [  8474,   9147) |******************************************************+
Info: [  9147,   9820) |************************************************************ 
Info: [  9820,  10493) |****************************************+
Info: [ 10493,  11166) |****************************************************+
Info: [ 11166,  11839) |********************************************+
Info: [ 11839,  12512) |******************************+
Info: [ 12512,  13185) |***+
0 warnings, 1 error
```

To me it looks like the critical path is 9.9 ns but the max frequency is 70.61 MHz.
I would expect 14.16 ns.

daveshah1, 2019-11-18:
This is a timing analysis bug (ime it's the path generation that usually goes wrong, so assume 70.61MHz to be the correct Fmax and 9.9ns to be wrong for now)

daveshah1, 2019-11-18:
Can you provide the JSON associated with this to help with tracking the bug down?

janrinze, 2019-11-18:
[design.json.gz](https://github.com/YosysHQ/nextpnr/files/3860672/design.json.gz)

hope it helps.

daveshah1, 2019-11-18:
This is hopefully fixed by f28186b

closed 2019-11-18 by daveshah1