YosysHQ/nextpnr #360: ECLKBRIDGECS primitive fails to place.

gregdavill, 2019-11-29:
I'm attempting to synthesis a LiteX SoC with DDR3 memory targeting the OrangeCrab ECP5 board.

Due to the small package I'm using I've had to route the DDR3 signals across two sides of the chip, BANKS 2,7,6.

In order to get Diamond to compile this design I've had to use the ECLKBRIDGECS block.
When compiling the same design with trellis I get an assertion failure originating when it tries to place/wire the ECLKBRIDGE.

```
terminate called after throwing an instance of 'nextpnr_ecp5::assertion_failure'
  what():  Assertion failure: closest != BelId() (/home/greg/projects/nextpnr/ecp5/pack.cc:2431)
build_top.sh: line 4: 32358 Aborted                 (core dumped) nextpnr-ecp5 --json top.json --lpf top.lpf --textcfg top.config --45k --package CSFBGA285 --freq 0.0
```

The LiteX design I'm trying to synth is here: https://github.com/gregdavill/OrangeCrab/tree/master/gateware/litex_ddr3_test

When attempting to synth without the bridge I get an error that the routing fails.

daveshah1, 2019-11-29:
I can't reproduce the ECLKBRIDGECS failure (running `python OrangeCrab-bitstream.py base trellis`). It is possible that your trellis is out of date - please make sure you have latest libtrellis built and run `touch ecp5/trellis_import.py` in nextpnr to make sure that it updates the database too.

The routing failure is a separate issue that I am looking into.

daveshah1, 2019-11-29:
ff30bc87 and 1c1c0968 should now fix all the build issues. It would be great if you could test and see if it works on hardware.

I note that the nextpnr command line doesn't include a speed grade (I think this is a LiteX issue). If as the BoM suggests you are using a `-8` grade part, adding `--speed 8` should give better timing figures and slightly faster routing as it conservatively chooses `-6` by default.

gregdavill, 2019-11-29:
Thanks, I'll ensure all my tools are updated and test it again.

gregdavill, 2019-11-29:
Turns out I had an old/duplicate libtrellis.so/pytrellis.so in my library path. 
I THINK this resulted in not providing nextpnr with the newest database.
 
It now completes, and LiteX BIOS runs and reports the memtest is OK! Thanks!

Thanks for the tip, I've got some -8 and some -6 parts depending on the densities, as I've just been buying stock that is available from distributors.

closed 2019-11-29 by gregdavill

daveshah1, 2019-11-29:
Thanks for testing!