YosysHQ/nextpnr #366: basename not declared in bba/main.c

derekmulcahy, 2019-12-05:
When I try to build for ice40 I get the following error
```
bba/main.cc:450:49: error: use of undeclared identifier
      'basename'
        fprintf(fileOut, "#embed_str \"%s\"\n", basename(bin_basename_buf));

```
I "fixed" the issue by including libgen.h in bba/main.c

daveshah1, 2019-12-06:
Cc @cliffordwolf

twam, 2019-12-06:
Same issues for me when building with clang on Mac:

```
[1/85] Building CXX object CMakeFiles/bbasm.dir/bba/main.cc.o
FAILED: CMakeFiles/bbasm.dir/bba/main.cc.o 
/usr/local/Homebrew/Library/Homebrew/shims/mac/super/clang++  -DGIT_COMMIT_HASH="" -DNO_GUI -DNO_PYTHON -DWITH_HEAP -Icommon -Ijson -I/usr/local/include/eigen3 -Wall -fPIC -O3 -g -pipe -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.15.sdk   -std=gnu++11 -MD -MT CMakeFiles/bbasm.dir/bba/main.cc.o -MF CMakeFiles/bbasm.dir/bba/main.cc.o.d -o CMakeFiles/bbasm.dir/bba/main.cc.o -c bba/main.cc
bba/main.cc:450:49: error: use of undeclared identifier 'basename'
        fprintf(fileOut, "#embed_str \"%s\"\n", basename(bin_basename_buf));
                                                ^
1 error generated.
ninja: build stopped: subcommand failed.
```

derekmulcahy, 2019-12-06:
My issue was on a Mac as well, Mac running Mojave. Maybe on other systems libgen.h is a freebie but needs to be explicitly included on the Mac.

daveshah1, 2019-12-06:
Fix for all arches pending in  #368

PR #368 (2019-12-07): Fix for bba to make it portable
We anyway use filesystem and system library from Boost and this makes it portable on macOS and Windows, fixes #366
closed 2019-12-07 by mmicko