YosysHQ/nextpnr #371: ECP5 issues to pack FF in MGIOL (syn_useioff)

Dolu1990, 2019-12-23:
Hi,

Was trying to get all the FF of a SDRAM interface to be placed in the MGIOL, but it seem there is an issue with the FF driving the output buffer enable (tristate command).

As far my understanding goes, it seem that the synthesis detected duplications in the tristates command, and consequantly merged tristate FF, which endup in a non-MGIOL friendly netlist.

There is a sample to reproduce the issue. It work fine on diamond :

   Number of registers:     66 out of 44583 (0%)
      PFU registers:           17 out of 43848 (0%)
      PIO registers:           49 out of   735 (7%)
   Number of SLICEs:         9 out of 21924 (0%)

But endup with the following with nextpnr
ERROR: Failed to pack flipflop 'dq_t_reg_TRELLIS_FF_Q' with 'syn_useioff' set into IOLOGIC.

```
module Toplevel (
            input clk,
            input  T, //Enable DQ output buffer
            inout [15:0] DQ);

      	(* syn_useioff *) reg  t_reg; //T IO FF
      	(* syn_keep *) reg t_buffer;  //FPGA fabric FF
      	(* syn_keep *) wire [15:0] t_comb; // Signal with 16 times t_buffer
      	(* syn_keep *) reg [15:0] dq_buffer; // DQ read FF in FPGA fabric
      	(* syn_useioff *) reg [15:0] dq_t_reg; //DQ tristate IO FF
      	(* syn_useioff *) reg [15:0] dq_o_reg; //DQ output IO FF
      	(* syn_useioff *) reg [15:0] dq_i_reg; //DQ input IO FF

        always @ (posedge clk) begin
      		t_reg <= T;
      		t_buffer <= t_reg;
      		dq_buffer <= dq_i_reg;
      		dq_o_reg <= dq_buffer;
      		dq_t_reg <= t_comb;
      		dq_i_reg <= DQ;
      	end

      	assign t_comb = t_buffer ? 16'hFFFF : 16'h0000;

      	assign DQ[0] = (dq_t_reg[0]) ? dq_o_reg[0] : 1'bz;
      	assign DQ[1] = (dq_t_reg[1]) ? dq_o_reg[1] : 1'bz;
      	assign DQ[2] = (dq_t_reg[2]) ? dq_o_reg[2] : 1'bz;
      	assign DQ[3] = (dq_t_reg[3]) ? dq_o_reg[3] : 1'bz;
      	assign DQ[4] = (dq_t_reg[4]) ? dq_o_reg[4] : 1'bz;
      	assign DQ[5] = (dq_t_reg[5]) ? dq_o_reg[5] : 1'bz;
      	assign DQ[6] = (dq_t_reg[6]) ? dq_o_reg[6] : 1'bz;
      	assign DQ[7] = (dq_t_reg[7]) ? dq_o_reg[7] : 1'bz;
      	assign DQ[8] = (dq_t_reg[8]) ? dq_o_reg[8] : 1'bz;
      	assign DQ[9] = (dq_t_reg[9]) ? dq_o_reg[9] : 1'bz;
      	assign DQ[10] = (dq_t_reg[10]) ? dq_o_reg[10] : 1'bz;
      	assign DQ[11] = (dq_t_reg[11]) ? dq_o_reg[11] : 1'bz;
      	assign DQ[12] = (dq_t_reg[12]) ? dq_o_reg[12] : 1'bz;
      	assign DQ[13] = (dq_t_reg[13]) ? dq_o_reg[13] : 1'bz;
      	assign DQ[14] = (dq_t_reg[14]) ? dq_o_reg[14] : 1'bz;
      	assign DQ[15] = (dq_t_reg[15]) ? dq_o_reg[15] : 1'bz;
endmodule
```

daveshah1, 2019-12-27:
I think the best option here is to re-duplicate the FF in nextpnr rather than making significant modifications to Yosys to avoid the merge in the first place. I will look into this.

Dolu1990, 2019-12-27:
I made some more experiments with diamond, and so far it seem tha that the tristate FF in MGIOL is quite hard to get right. + it seem that the logical level to enable the output is '0'. I will give some more try with Yosys to diagnostic things better.

> I think the best option here is to re-duplicate the FF in nextpnr rather than making significant modifications to Yosys to avoid the merge in the first place. I will look into this.

So it would require a specific netlist for the yosys synthesis, without the IO FF, right ?

daveshah1, 2019-12-27:
> So it would require a specific netlist for the yosys synthesis, without the IO FF, right ?

I don't think there would be any Yosys changes at all - just that nextpnr would duplicate the flipflop automatically

>  it seem that the logical level to enable the output is '0'

This is correct at the hardware level, I think there might be a Diamond bug related to this in certain cases though