YosysHQ/nextpnr #374: ECP5 pll clock -> ODDRX1F.clk done via fabric routing

Dolu1990, 2019-12-27:
Hi,

The routing between the clock output of the PLL and the ODDRX1F.CLK is very likely done via the fabric routing, which is quite problematic, as is make the clock phase unpredictable.

 -> Is there any way to force / infer the use of the clock tree instead ? I checked the ECP5 lattice doc, but it seem that there is no way to force it explicitly.

To explaine the case more precisely, i have one PLL which generate two clock
- clk0 => internal logic + SDR outputs
- clk1 => sdram chip clock generation (via ODDRX1F)

And basicaly, if i set the phase between clk0 and clk1 to zero in the PLL config : 
```
	ecppll \
                -i 25 \
                -o 10 \
		--clkout1 20 \
		--phase1 0 \
                -n  pll_linux\
                -f pll_linux.v
```

I can see on the logic analyser that clk1 is about 5 ns later than clk0 with nextpnr, while with diamond that's fine.

my hypotesis is : clk0 drive a lot of flops so it goes to the clock tree, while clk1 is only driving one ODDRX1F.clk =>  no clock tree used by yosys1

Else if i manualy compensate the clk1 delay, the whole design work and the ULX3S is booting linux just fine, with 50 Mhz core clock, 100 Mhz SDRAM :)

daveshah1, 2019-12-27:
> my hypotesis is : clk0 drive a lot of flops so it goes to the clock tree, while clk1 is only driving one ODDRX1F.clk => no clock tree used by yosys1

Correct, this is how nextpnr determines clocks. I will add a heuristic for IOLOGIC clocks, I have one to always promote certain other clock types already.

closed 2019-12-27 by daveshah1

Dolu1990, 2019-12-27:
Thanks, everything good now :)

``` 
Info:     routing clock net $glbnet$pll_clkout0 using global 0
Info:     routing clock net $glbnet$system_cpu_tck$TRELLIS_IO_IN using global 1
Info:     routing clock net $glbnet$pll_clkout1 using global 2
```

commit 0d43aff2 (2019-12-27): ecp5: Always promote IOLOGIC SCLK to global

Fixes #374