YosysHQ/nextpnr #375: libQt5Core.so.5: cannot open shared object file

RobertBaruch, 2019-12-31:
I've cloned and built nextpnr just now (`cmake -DARCH=ice40 .`) , and I'm running into the dreaded libQt5Core.so.5 not found. I have already installed qt-default:

```
lrwxrwxrwx 1 root root      20 Oct 30 04:47 /usr/lib/x86_64-linux-gnu/libQt5Core.so -> libQt5Core.so.5.11.3             
lrwxrwxrwx 1 root root      20 Oct 30 04:47 /usr/lib/x86_64-linux-gnu/libQt5Core.so.5 -> libQt5Core.so.5.11.3           
lrwxrwxrwx 1 root root      20 Oct 30 04:47 /usr/lib/x86_64-linux-gnu/libQt5Core.so.5.11 -> libQt5Core.so.5.11.3        
-rw-r--r-- 1 root root 5200168 Oct 30 04:47 /usr/lib/x86_64-linux-gnu/libQt5Core.so.5.11.3
```

`ldd` on `nextpnr-ice40` indeed shows things like:

```
libicuuc.so.63 => /usr/lib/x86_64-linux-gnu/libicuuc.so.63 (0x00007f4bfa340000)
libQt5Core.so.5 => not found
```

I am not using Anaconda, so I'm pretty sure https://github.com/YosysHQ/nextpnr/issues/292 is not applicable.

```
$ grep -i Qt CMakeCache.txt

//The directory containing a CMake configuration file for Qt5Core.
Qt5Core_DIR:PATH=/usr/lib/x86_64-linux-gnu/cmake/Qt5Core
//The directory containing a CMake configuration file for Qt5Gui.
Qt5Gui_DIR:PATH=/usr/lib/x86_64-linux-gnu/cmake/Qt5Gui
//The directory containing a CMake configuration file for Qt5OpenGL.
Qt5OpenGL_DIR:PATH=/usr/lib/x86_64-linux-gnu/cmake/Qt5OpenGL
//The directory containing a CMake configuration file for Qt5Widgets.
Qt5Widgets_DIR:PATH=/usr/lib/x86_64-linux-gnu/cmake/Qt5Widgets
//The directory containing a CMake configuration file for Qt5.
Qt5_DIR:PATH=/usr/lib/x86_64-linux-gnu/cmake/Qt5
QtPropertyBrowser_BINARY_DIR:STATIC=/home/robertbaruch/repos/nextpnr/generated/3rdparty/QtPropertyBrowser
QtPropertyBrowser_LIB_DEPENDS:STATIC=general;Qt5::Widgets;
QtPropertyBrowser_SOURCE_DIR:STATIC=/home/robertbaruch/repos/nextpnr/3rdparty/QtPropertyBrowser
gui_ice40_LIB_DEPENDS:STATIC=general;Qt5::Widgets;
```

Any ideas?

daveshah1, 2019-12-31:
Just to check:

What is the full error when running nextpnr?
What is the full output of `ldd -v nextpnr-ice40`?
What distro is this?

RobertBaruch, 2019-12-31:
It seems to have been a Windows Subsystem for Linux issue, when using Debian. Apparently this is a [known issue](https://superuser.com/questions/1347723/arch-on-wsl-libqt5core-so-5-not-found-despite-being-installed).

Also see the [bug in the WSL repo](https://github.com/Microsoft/WSL/issues/3023).

Long story short, you need to strip out the ELF section called `.note.ABI-tag` from libQt5Core.so.5:

`sudo strip --remove-section=.note.ABI-tag /usr/lib/x86_64-linux-gnu/libQt5Core.so.5`

closed 2019-12-31 by RobertBaruch

yousefcodes, 2022-11-09:
This saved me!

DeflateAwning, 2023-11-20:
Note that the `strip` command can be installed on Ubuntu/Debian with `sudo apt install binutils`