YosysHQ/nextpnr #378: GUI no highlithing, runtime warnings

Riconec, 2020-01-09:
I tried to open existing json file generated for picosoc, bels, nets etc seems to load from json, but nothing is highlighted in gui (clicking on fpga structure selects appropriate bel for example, but "highlight group, clicking nets gives no effect on fpga structure).

There are a lot of errors\warning, there are some of them:
1)
```
qt.glx: qglx_findConfig: Failed to finding matching FBConfig (8 8 8 0)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig (1 8 8 0)
```

2)
```
frozen_importlib:219: RuntimeWarning: to-Python converter for nextpnr_ice40::GraphicElement::type_t already registered; second conversion method ignored.
_frozen_importlib:219: RuntimeWarning: to-Python converter for nextpnr_ice40::GraphicElement::style_t already registered; second conversion method ignored.
_frozen_importlib:219: RuntimeWarning: to-Python converter for nextpnr_ice40::GraphicElement already registered; second conversion method ignored.
_frozen_importlib:219: RuntimeWarning: to-Python converter for nextpnr_ice40::PortType already registered; second conversion method ignored.
_frozen_importlib:219: RuntimeWarning: to-Python converter for nextpnr_ice40::BaseCtx* already registered; second conversion method ignored.
```

3)
```
QObject::installEventFilter(): Cannot filter events for objects in a different thread.
QObject::installEventFilter(): Cannot filter events for objects in a different thread.
QObject: Cannot create children for a parent that is in a different thread.
(Parent is Breeze::WidgetStateEngine(0x5609724b6c70), parent's thread is QThread(0x56097241a240), current thread is QThread(0x5609724e5c60)
QObject: Cannot create children for a parent that is in a different thread.
(Parent is Breeze::WidgetStateEngine(0x5609724b6c70), parent's thread is QThread(0x56097241a240), current thread is QThread(0x5609724e5c60)
```

P.S. buit from latest branch, ice40 arch.

Riconec, 2020-01-09:
It also fatal error could happen, I just clicked file->new->up5k->sg48 and when I loaded json it shows up "fatal error" message window.

in general window looks like this no matter what I am doing
[image]

mmicko, 2020-01-10:
Thanks for report. Seams to be a problem when structures are reinitialized on reload
Can you please give more info about your system (Linux distribution version you are using) ?
Also if you can try doing nextpnr-ice40 --gui --json yourfile.json --pcf your.pcf and then try highlight to see if then it is working fine with you.
Thanks.