YosysHQ/nextpnr #383: Optimal procedure to build three versions: nextpnr-ice40, nextpnr-ecp5 and nextpnr

eine, 2020-01-16:
I am currently building nextpnr three times independently:

```Dockerfile
FROM nextpnr-build-base AS nextpnr-ice40-build
COPY --from=icestorm-build /opt/icestorm/usr/local/share/icebox /usr/local/share/icebox

RUN mkdir -p /tmp/nextpnr/build \
 && cd /tmp/nextpnr \
 && curl -fsSL https://codeload.github.com/YosysHQ/nextpnr/tar.gz/master | tar xzf - --strip-components=1 \
 && cd build \
 && cmake .. \
   -DARCH=ice40 \
   -DBUILD_GUI=OFF \
   -DBUILD_PYTHON=ON \
   -DUSE_OPENMP=ON \
 && make -j $(nproc) \
 && make DESTDIR=/opt/nextpnr install

FROM nextpnr-base AS nextpnr-ice40
COPY --from=nextpnr-ice40-build /opt/nextpnr /
```

```Dockerfile
FROM nextpnr-build-base AS nextpnr-ecp5-build
COPY --from=trellis-build /opt/trellis /

RUN mkdir -p /tmp/nextpnr/build \
 && cd /tmp/nextpnr \
 && curl -fsSL https://codeload.github.com/YosysHQ/nextpnr/tar.gz/master | tar xzf - --strip-components=1 \
 && cd build \
 && cmake .. \
   -DARCH=ecp5 \
   -DBUILD_GUI=OFF \
   -DBUILD_PYTHON=ON \
   -DUSE_OPENMP=ON \
 && make -j $(nproc) \
 && make DESTDIR=/opt/nextpnr install

FROM nextpnr-base AS nextpnr-ecp5
COPY --from=nextpnr-ecp5-build /opt/nextpnr /
```

```Dockerfile
FROM nextpnr-build-base AS nextpnr-build
COPY --from=icestorm-build /opt/icestorm/usr/local/share/icebox /usr/local/share/icebox
COPY --from=trellis-build /opt/trellis /

RUN mkdir -p /tmp/nextpnr/build \
 && cd /tmp/nextpnr \
 && curl -fsSL https://codeload.github.com/YosysHQ/nextpnr/tar.gz/master | tar xzf - --strip-components=1 \
 && cd build \
 && cmake .. \
   -DARCH=all \
   -DBUILD_GUI=OFF \
   -DBUILD_PYTHON=ON \
   -DUSE_OPENMP=ON \
 && make -j $(nproc) \
 && make DESTDIR=/opt/nextpnr install

FROM nextpnr-base AS nextpnr
COPY --from=nextpnr-build /opt/nextpnr /
```

I'd like to know whether any clean procedure exists to:

- Build the files that are common to the three variants just one.
- Reuse the ICE40 and ECP5 files from `nextpnr-ice40-build` and `nextpnr-ecp5-build`, respectively, to avoid building them again in `nextpnr-build`.

daveshah1, 2020-01-16:
The architecture of nextpnr means that this is not possible (deliberately as it allows a lot more inlining in the core algorithms).

closed 2020-01-16 by daveshah1

eine, 2020-01-16:
Thanks for the quick response. Is it possible to build for ICE40 and ECP5 but not generic? Something similar to `-DARCH=ice40,ecp5`?

daveshah1, 2020-01-16:
Yes, almost that but separated by semicolons rather than commas