YosysHQ/nextpnr #387: Can't manually place cells in carry chains

newhouseb, 2020-01-26:
First, thanks for making nextpnr! It's lovely to do digital design outside of the kludge that is often vendor-specific stacks.

I've been working on a Time-to-Digital converter for a high-speed ADC (as in [this paper](https://cas.tudelft.nl/pubs/Homulle15fpga.pdf)). This requires manually placed carry chains. After reading a ton of nextpnr code (and adding extra logging) I've come to understand the following:

- After cells are packed, if carry is enabled on a cell, it's placed into a carry chain, which is capped with an ICESTORM_LC cell on either end of the chain.
- These additional ICESTORM_LC cells don't inherit any attributes from the origin cells in the chain.
- Cells in a chain are placed relative to the "parent" at the beginning of the chain and as such the placement algorithm (at least for the default HeAP, haven't looked at SA) only solves for the "parent." This means that any placement attributes within the chain are ignored.

I prototyped a patch on top of nextpnr that moves BEL attributes at the top of a chain to the extra ICESTORM_LC cells and then places and propagates these constraints at the beginning of placement here: https://github.com/newhouseb/nextpnr/commit/39f15c4d8ce7ec0832ad2aeb1433cabd0033a323. I'm totally new to this code base and to any FPGA P&R so I present this humbly with the understanding that I have no idea what I'm doing.

It seems to me there are a couple of enhancements that seem reasonable here:

- An algorithm to deduce when things can be placed without the end-cap ICESTORM_LC blocks (iCECube2 happily places such cells without the extra PLB usage).
- An addition to constraint mapping that fixes the placement of carry chains when the user has specified a physical location of the top cell (assuming it's legal).

---

(For what it's worth, I've prototyped what I need using iCECube2 and actually found that the carry chains are actually _too_ fast within a PLB because of (what I presume is) fast carry look-ahead, so I think I'll be using a delay chain with a different component.)

daveshah1, 2020-01-27:
Hi @newhouseb !

This change looks very reasonable to me. Please see the comments I have left on the commit and create a pull request.

I would need to look a bit more at the carry chain legalisation issue, in terms of the end-cap blocks.

newhouseb, 2020-01-27:
Oh that's great to hear, I'll clean up the code when I get a chance and look at adding tests and then submit a pull request.

aWZHY0yQH81uOYvH, 2025-02-09:
Bump, I've been having the same issue