YosysHQ/nextpnr #395: Build on Mac with standard CMake 3.16 failing

BracketMaster, 2020-02-13:
The following lines cause building NextPNR on Mac to fail. For some reason, Cmake 3.16 thinks Mac is windows...

Is there a way to get around this? Do we need to support CygWin builds, nextpnr works pretty well on windows subsystem for linux Ununtu, Microsofts officially support version of Linux for Windows...

https://github.com/YosysHQ/nextpnr/blob/e0b4f0ee63072f0e5aa86b4fa9ca1f488ae57790/ecp5/family.cmake#L29-L33

daveshah1, 2020-02-14:
@mmicko can you have a look at this?

mmicko, 2020-02-15:
@BracketMaster Just checked with latest cmake brew package and all works just fine. 

Have built and installed prjtrellis (using just `cmake .`  without enforcing it to write to /usr, but letting it use default /usr/local instead) and then for nextpnr run `cmake -DARCH=ecp5 -DTRELLIS_INSTALL_PREFIX=/usr/local . `

BracketMaster, 2020-02-15:
Hmm OK. Maybe it. was something on my end.
I have a fresh Mac lying around with a clean install of Catalina lying around, curious to see if I get the same results.

Should probably close this for now.

closed 2020-02-15 by BracketMaster