YosysHQ/nextpnr #407: Crashes with "std::out_of_range: vector"

porglezomp, 2020-03-05:
I was working with an un-flattened synthesis for synthesis speed, and ran into an uncaught exception.

Error:
```
```
Automatically reduced JSON case:
```json
{
  "modules": {
    "example": {
      "ports": {
        "A": {"direction": "input", "bits": [1]},
        "B": {"direction": "output", "bits": [1]}
      }
    },
    "": {
      "cells": {
        "": {
          "type": "example", "connections": {
            "A": [1],
            "B": [1]
          }
        }
      }
    }
  }
}
```
Command:
```
nextpnr-ecp5 --json reduce.json
```

porglezomp, 2020-03-05:
I see another issue mentions "currently a flattened netlist only is allowed" which I thought might be the cause of this, but it also still seems to work with only one of the ports included.

Still, if this is both the cause and the currentexpected behavior, would it be easy to add an error message explaining that only flat netlists are supported—at least before those improvements might be implemented?

daveshah1, 2020-03-05:
There is now experimental support for non-flattened netlists. I am not sure where this JSON came from however, in particular a missing `port_directions` on the cell is likely to cause problems. Improved error handling in the JSON frontend is still on the TODO list.

Incidentally did you also mention some timing analysis errors on IRC, or was that someone else? If that is related then that is likely to be a genuine issue that needs looking into.

porglezomp, 2020-03-05:
This was reduced by C-Reduce from a yosys synthesis output, trying to keep that bounds check error, which would be why it doesn't properly match the expected format. Versions involved in the original failure case:

```
Yosys 0.9+1706 (git sha1 0ec97144, clang 11.0.0 -fPIC -Os)
nextpnr-ecp5 -- Next Generation Place and Route (Version 5fa10e1)
```

porglezomp, 2020-03-05:
I'll see about getting a more well-behaved failing case for the related issues as well.

daveshah1, 2020-03-06:
Can you try af7be21 - it fixes a bound checks issue on an unflattened netlist that I tried?

porglezomp, 2020-03-06:
That fixes the issue for me, I can make progress on this design again. Thanks a bunch!

closed 2020-03-06 by porglezomp