YosysHQ/nextpnr #408: Error evaluating generator expression: $<TARGET_OBJECTS:ice40_chipdb>

pcotret, 2020-03-05:
Error similar to [#173](https://github.com/YosysHQ/nextpnr/issues/173). I've tried to update my `cmake` as @uXeBoy suggested. Still doesn't work...

```
CMake Error at ice40/family.cmake:119 (target_sources):
  Error evaluating generator expression:

    $<TARGET_OBJECTS:ice40_chipdb>

  The evaluation of the TARGET_OBJECTS generator expression is only suitable
  for consumption by CMake.  It is not suitable for writing out elsewhere.
Call Stack (most recent call first):
  CMakeLists.txt:266 (include)

CMake Error at ice40/family.cmake:119 (target_sources):
  Error evaluating generator expression:

    $<TARGET_OBJECTS:ice40_chipdb>

  The evaluation of the TARGET_OBJECTS generator expression is only suitable
  for consumption by CMake.  It is not suitable for writing out elsewhere.
Call Stack (most recent call first):
  CMakeLists.txt:266 (include)

CMake Error at ice40/family.cmake:119 (target_sources):
  Error evaluating generator expression:

    $<TARGET_OBJECTS:ice40_chipdb>

  The evaluation of the TARGET_OBJECTS generator expression is only suitable
  for consumption by CMake.  It is not suitable for writing out elsewhere.
Call Stack (most recent call first):
  CMakeLists.txt:266 (include)

CMake Error at ice40/family.cmake:119 (target_sources):
  Error evaluating generator expression:

    $<TARGET_OBJECTS:ice40_chipdb>

  The evaluation of the TARGET_OBJECTS generator expression is only suitable
  for consumption by CMake.  It is not suitable for writing out elsewhere.
Call Stack (most recent call first):
  CMakeLists.txt:266 (include)

CMake Error at ice40/family.cmake:119 (target_sources):
  Error evaluating generator expression:

    $<TARGET_OBJECTS:ice40_chipdb>

  The evaluation of the TARGET_OBJECTS generator expression is only suitable
  for consumption by CMake.  It is not suitable for writing out elsewhere.
Call Stack (most recent call first):
  CMakeLists.txt:266 (include)
```
## My configuration
- Ubuntu 18.04 LTS
- CMake 3.3.2

smunaut, 2020-03-05:
Yeah your CMake is too old to build this.

daveshah1, 2020-03-05:
How did you end up with CMake 3.3 with Ubuntu 18.04??

smunaut, 2020-03-05:
Check you haven't some manual install of cmake in /usr/local or somewhere overriding the system cmake.

pcotret, 2020-03-05:
My bad, it was looking for `cmake` 3.3.2 in my Xilinx 2018.2 SDK directory...
However, it's working with `cmake` 3.16.5 installed through `snap` (just need a proper way to get both working together).

closed 2020-03-05 by pcotret