YosysHQ/nextpnr #411: Building on Alpine Linux - Boost::Python and Eigen3 not found

jchidley, 2020-03-14:
I am trying to run nextpnr on Alpine Linux.  I believe that I have the right packages installed but I am left with the following errors when I run `cmake -DARCH=ice40 .`:

```bash
CMake Warning at /usr/share/cmake/Modules/FindBoost.cmake:2003 (message):
  No header defined for python-py382; skipping header check
Call Stack (most recent call first):
  CMakeLists.txt:157 (find_package)

CMake Warning at /usr/share/cmake/Modules/FindBoost.cmake:2003 (message):
  No header defined for python-py38; skipping header check
Call Stack (most recent call first):
  CMakeLists.txt:164 (find_package)

CMake Warning at /usr/share/cmake/Modules/FindBoost.cmake:2003 (message):
  No header defined for python-py3; skipping header check
Call Stack (most recent call first):
  CMakeLists.txt:164 (find_package)

CMake Error at CMakeLists.txt:199 (find_package):
  Could not find a package configuration file provided by "Eigen3" with any
  of the following names:

    Eigen3Config.cmake
    eigen3-config.cmake

  Add the installation prefix of "Eigen3" to CMAKE_PREFIX_PATH or set
  "Eigen3_DIR" to a directory containing one of the above files.  If "Eigen3"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
See also "/home/jack/nextpnr/CMakeFiles/CMakeOutput.log".
```

From looking at CMakeLists.txt I believe that my problems are caused because Alpine Linux is different from the 'supported' distributions.  See below for the relevant lines (149-194) for boost::python:

```bash
if (BUILD_PYTHON)
    # Find Boost::Python of a suitable version in a cross-platform way
    # Some distributions (Arch) call it libboost_python3, others such as Ubuntu
    # call it libboost_python35. In the latter case we must consider all minor versions
    # Original source: https://github.com/BVLC/caffe/blob/master/cmake/Dependencies.cmake#L148
    set(version ${PYTHONLIBS_VERSION_STRING})

    STRING(REGEX REPLACE "[^0-9]" "" boost_py_version "${version}")
    find_package(Boost QUIET COMPONENTS "python-py${boost_py_version}" ${boost_libs})
    set(Boost_PYTHON_FOUND ${Boost_PYTHON-PY${boost_py_version}_FOUND})

    while (NOT "${version}" STREQUAL "" AND NOT Boost_PYTHON_FOUND)
        STRING(REGEX REPLACE "([0-9.]+).[0-9]+" "\\1" version "${version}")

        STRING(REGEX REPLACE "[^0-9]" "" boost_py_version "${version}")
        find_package(Boost QUIET COMPONENTS "python-py${boost_py_version}" ${boost_libs})
        set(Boost_PYTHON_FOUND ${Boost_PYTHON-PY${boost_py_version}_FOUND})

        STRING(REGEX MATCHALL "([0-9.]+).[0-9]+" has_more_version "${version}")
        if ("${has_more_version}" STREQUAL "")
            break()
        endif ()
    endwhile ()

    if (NOT Boost_PYTHON_FOUND)
        foreach (PyVer 3 36 37 38 39)
            find_package(Boost QUIET COMPONENTS python${PyVer} ${boost_libs})
            if ("${Boost_LIBRARIES}" MATCHES ".*(python|PYTHON).*" )
                set(Boost_PYTHON_FOUND TRUE)
                break()
            endif ()
        endforeach ()
    endif ()

    if (NOT Boost_PYTHON_FOUND)
        STRING(REGEX REPLACE "([0-9]+\\.[0-9]+).*" "\\1" gentoo_version "${PYTHONLIBS_VERSION_STRING}")
        find_package(Boost QUIET COMPONENTS python-${gentoo_version} ${boost_libs})
        if ("${Boost_LIBRARIES}" MATCHES ".*(python|PYTHON).*" )
            set(Boost_PYTHON_FOUND TRUE)
        endif ()
    endif ()

    if (NOT Boost_PYTHON_FOUND )
        message( FATAL_ERROR "No version of Boost::Python 3.x could be found.")
    endif ()
endif()                                  
```

I haven't checked Eigen3 but I assume that it will be the same problem.  I am continuing to investigate the issue and see if I can find a fix.

marzoul, 2020-03-21:
Hi,
I have the same issue with Archlinux to compile the package from source:
[https://aur.archlinux.org/packages/nextpnr-git](https://aur.archlinux.org/packages/nextpnr-git)
I narrowed down the issue to the following line of CMakeLists.txt:
`find_package(PythonLibs 3.5 REQUIRED)`
When I call cmake with additional option:
`-DPYTHON_INCLUDE_DIRS=/usr/include/python3.8`
I can observe that, before the call to `find_package(PythonLibs 3.5 REQUIRED)`, I have:
`PYTHON_INCLUDE_DIRS=/usr/include/python3.8`
and after the call I have:
`PYTHON_INCLUDE_DIRS=/usr/include/python3.7m`
This is a wrong result. I don't know where this comes from, but at least I can manually patch the CMakeLists.txt to override `PYTHON_INCLUDE_DIRS` and compile successfullly.

jchidley, 2020-03-21:
I did this yesterday on Arch Linux.  It is tricky.  You need the AUR packages:
https://aur.archlinux.org/packages/icestorm-git/
https://aur.archlinux.org/packages/trellis-git/
https://aur.archlinux.org/packages/nextpnr-git/
https://aur.archlinux.org/packages/yosys-git/

For some reason, not all the prerequisites installed correctly, so you might want to check that.  And they all have to be up to date.

natevw, 2021-08-17:
I hit this too, it seems to be related to https://gitlab.alpinelinux.org/alpine/aports/-/issues/11233 — which I don't understand the status of. It seemed the consensus was:

> Yes, please put [the files not found] on usr/lib/cmake

and the issue was closed a year ago, but looking at [the alpine v3.14 package](https://pkgs.alpinelinux.org/contents?file=*.cmake&path=&name=eigen-dev&branch=v3.14&repo=community) (as well as the `edge` package) everything is still in "/usr/share/cmake/" instead.

I was able to workaround via:

```
cd nextpnr; Eigen3_DIR=/usr/share/cmake/Modules cmake -DARCH=ice40 -DCMAKE_INSTALL_PREFIX=/usr/local . && make -j$(nproc) && make install
```

i.e. specifically I added `Eigen3_DIR=/usr/share/cmake/Modules` environment variable ahead of the `cmake` call so it can find it in the wrong location. [Still having problems building on alpine but I'll file a separate issue for that.]