YosysHQ/nextpnr #412: building on Kubuntu 19.10 - no header defined - EigneConfig.cmake not found

civicbynature, 2020-03-19:
i have installed all prerequisites via:
sudo apt-get install build-essential clang bison flex libreadline-dev \
                     gawk tcl-dev libffi-dev git mercurial graphviz   \
                     xdot pkg-config python python3 libftdi-dev \
                     qt5-default python3-dev libboost-all-dev cmake

but when ever i try to build nextpnr from git clone. all i get is the below. totally stuck. 

Please help!

sobilinux@nathan-LProgrammer:~/nextpnr$ cmake -DARCH=ice40 -DCMAKE_INSTALL_PREFIX=/usr/local .
-- Boost version: 1.67.0
-- Found the following Boost libraries:
--   filesystem
--   thread
--   program_options
--   iostreams
--   system
--   chrono
--   date_time
--   atomic
--   regex
CMake Warning at /usr/share/cmake-3.13/Modules/FindBoost.cmake:1775 (message):
  No header defined for python-py375; skipping header check
Call Stack (most recent call first):
  CMakeLists.txt:157 (find_package)

CMake Warning at /usr/share/cmake-3.13/Modules/FindBoost.cmake:1775 (message):
  No header defined for python-py37; skipping header check
Call Stack (most recent call first):
  CMakeLists.txt:164 (find_package)

CMake Warning at /usr/share/cmake-3.13/Modules/FindBoost.cmake:1775 (message):
  No header defined for python-py3; skipping header check
Call Stack (most recent call first):
  CMakeLists.txt:164 (find_package)

CMake Error at CMakeLists.txt:199 (find_package):
  Could not find a package configuration file provided by "Eigen3" with any
  of the following names:

    Eigen3Config.cmake
    eigen3-config.cmake

  Add the installation prefix of "Eigen3" to CMAKE_PREFIX_PATH or set
  "Eigen3_DIR" to a directory containing one of the above files.  If "Eigen3"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
See also "/home/sobilinux/nextpnr/CMakeFiles/CMakeOutput.log".
See also "/home/sobilinux/nextpnr/CMakeFiles/CMakeError.log".

daveshah1, 2020-03-19:
You need to install eigen3-dev or equivalent

daveshah1, 2020-04-03:
Did installing the eigen3 dev package work? Can this issue be closed?

wrak, 2020-04-11:
Had same error. It did work for me.

closed 2020-07-05 by daveshah1