YosysHQ/nextpnr #413: macOS Catalina (10.15.3) - build issue with GUI

aklsh, 2020-03-26:
I tried building nextpnr on my Mac with Catalina 10.15.3, but I couldn't build it with GUI support. When I try to build it without GUI, it builds though. I have `qt5` and all the other dependencies installed via HomeBrew, but I don't know what mistake I am making. I am comfortable with both GUI and CLI, but I just wanted to raise this issue as it might help another person who's facing the same thing.

I even tried the solution mentioned [here](https://github.com/YosysHQ/nextpnr/issues/190), but still get the same thing.

When I build with GUI support, I get this:

```
/usr/local/opt/qt/lib/QtGui.framework/Headers/qsurfaceformat.h:154:5: error: unknown type
      name 'ColorSpace'
    ColorSpace colorSpace() const;
    ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
/usr/local/opt/qt/lib/QtGui.framework/Headers/qsurfaceformat.h:154:5: error: unknown type
      name 'ColorSpace'
    ColorSpace colorSpace() const;
    ^
In file included from /Users/akileshkannan/Desktop/nextpnr/gui/fpgaviewwidget.cc:23:
In file included from /usr/local/opt/qt/lib/QtWidgets.framework/Headers/QApplication:1:
In file included from /usr/local/opt/qt/lib/QtWidgets.framework/Headers/qapplication.h:52:
/usr/local/opt/qt/lib/QtGui.framework/Headers/qguiapplication.h:97:12: error: unknown
      type name 'QWindowList'
    static QWindowList allWindows();
           ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
/usr/local/opt/qt/lib/QtGui.framework/Headers/qguiapplication.h:98:12: error: unknown
      type name 'QWindowList'
    static QWindowList topLevelWindows();
           ^
/usr/local/opt/qt/lib/QtCore.framework/Headers/qabstractitemmodel.h:135:5: error:
      function-like macro 'QT_DEPRECATED_SINCE' is not defined
#if QT_DEPRECATED_SINCE(5, 8)
    ^
/usr/local/opt/qt/lib/QtCore.framework/Headers/qabstractitemmodel.h:149:1: error: no
      member named 'isDetached' in 'QPersistentModelIndex'
Q_DECLARE_SHARED(QPersistentModelIndex)
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/local/include/QtCore/qglobal.h:2330:63: note: expanded from macro 'Q_DECLARE_SHARED'
template <> inline bool qIsDetached<TYPE>(TYPE &t) { return t.isDetached(); } \
                                                            ~ ^
In file included from /Users/akileshkannan/Desktop/nextpnr/gui/treemodel.cc:21:
In file included from /Users/akileshkannan/Desktop/nextpnr/gui/treemodel.h:24:
In file included from /usr/local/opt/qt/lib/QtCore.framework/Headers/QAbstractItemModel:1:
/usr/local/opt/qt/lib/QtCore.framework/Headers/qabstractitemmodel.h:149:1: error: no
      member named 'data_ptr' in 'QPersistentModelIndex'
Q_DECLARE_SHARED(QPersistentModelIndex)
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/local/include/QtCore/qglobal.h:2332:16: note: expanded from macro 'Q_DECLARE_SHARED'
{ qSwap(value1.data_ptr(), value2.data_ptr()); } \
        ~~~~~~ ^
In file included from /Users/akileshkannan/Desktop/nextpnr/gui/treemodel.cc:21:
In file included from /Users/akileshkannan/Desktop/nextpnr/gui/treemodel.h:24:
In file included from /usr/local/opt/qt/lib/QtCore.framework/Headers/QAbstractItemModel:1:
/usr/local/opt/qt/lib/QtCore.framework/Headers/qabstractitemmodel.h:149:1: error: no
      member named 'data_ptr' in 'QPersistentModelIndex'
Q_DECLARE_SHARED(QPersistentModelIndex)
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/local/include/QtCore/qglobal.h:2332:35: note: expanded from macro 'Q_DECLARE_SHARED'
{ qSwap(value1.data_ptr(), value2.data_ptr()); } \
                           ~~~~~~ ^
In file included from /Users/akileshkannan/Desktop/nextpnr/gui/treemodel.cc:21:
In file included from /Users/akileshkannan/Desktop/nextpnr/gui/treemodel.h:24:
In file included from /usr/local/opt/qt/lib/QtCore.framework/Headers/QAbstractItemModel:1:
/usr/local/opt/qt/lib/QtCore.framework/Headers/qabstractitemmodel.h:149:1: error:
      'swap<QPersistentModelIndex>' is missing exception specification
      'noexcept(is_nothrow_move_constructible<QPersistentModelIndex>::value &&
      is_nothrow_move_assignable<QPersistentModelIndex>::value)'
/usr/local/include/QtCore/qglobal.h:2332:50: note: expanded from macro 'Q_DECLARE_SHARED'
{ qSwap(value1.data_ptr(), value2.data_ptr()); } \
                                                 ^
/usr/local/include/QtCore/qglobal.h:2323:28: note: expanded from macro '\
[... 931 lines cut ...]
/usr/local/opt/qt/lib/QtWidgets.framework/Headers/qsizepolicy.h:110:21: note: previous
      declaration is here
    Q_DECLARE_FLAGS(ControlTypes, ControlType)
                    ^
/usr/local/opt/qt/lib/QtWidgets.framework/Headers/qsizepolicy.h:111:25: error: expected
      ';' at end of declaration list
    Q_FLAG(ControlTypes)
                        ^
make[2]: *** [generated/gui/ice40/CMakeFiles/gui_ice40.dir/application.cc.o] Error 1
/usr/local/opt/qt/lib/QtWidgets.framework/Headers/qsizepolicy.h:132:5: error: unknown
      type name 'Q_DECL_RELAXED_CONSTEXPR'
    Q_DECL_RELAXED_CONSTEXPR void setHorizontalPolicy(Policy d) noexcept { bits.h...
    ^
/usr/local/opt/qt/lib/QtWidgets.framework/Headers/qsizepolicy.h:132:30: error: expected
      member name or ';' after declaration specifiers
    Q_DECL_RELAXED_CONSTEXPR void setHorizontalPolicy(Policy d) noexcept { bits.h...
    ~~~~~~~~~~~~~~~~~~~~~~~~ ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
make[2]: *** [generated/gui/ice40/CMakeFiles/gui_ice40.dir/ice40/mainwindow.cc.o] Error 1
20 errors generated.
make[2]: *** [generated/gui/ice40/CMakeFiles/gui_ice40.dir/pythontab.cc.o] Error 1
make[1]: *** [generated/gui/ice40/CMakeFiles/gui_ice40.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```

daveshah1, 2020-03-26:
It might be similar to https://github.com/ros2/ros2/issues/441

You could try
```
brew link --overwrite --force qt
```

aklsh, 2020-03-27:
Thanks a lot @daveshah1 ! It worked now. I'll close this issue.

closed 2020-03-27 by aklsh

Patronics, 2024-08-29:
> You could try
> 
> ```
> brew link --overwrite --force qt
> ```
Thanks so much, this fixed it for me! (I adjusted qt to qt@5 since qt6 is the default now) Should probably be added to the official instructions