YosysHQ/nextpnr #429: Error when compiling nextpnr tool

ecp5fpgauser, 2020-04-22:
Here is the result of the following command line on a virtual env (Linux Mint through VirtualBox):
cmake -DCMAKE_INSTALL_PREFIX=/opt/nextpnr -DARCH=ecp5 \
-DTRELLIS_ROOT=/opt/trellis/share/trellis \
-DPYTRELLIS_LIBDIR=/opt/trellis/lib/trellis\
-DTRELLIS_INSTALL_PREFIX=/opt/trellis/database .
and make
I face the following issue: No module named 'database' or even 'pytrellis' or even 'pip_classes'
Do you know what's going wrong ?
Many thanks
___
$ cmake -DCMAKE_INSTALL_PREFIX=/opt/nextpnr -DARCH=ecp5 -DTRELLIS_ROOT=/opt/trellis/share/trellis -DPYTRELLIS_LIBDIR=/opt/trellis/lib/trellis -DTRELLIS_INSTALL_PREFIX=/opt/trellis/database .
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   filesystem
--   thread
--   program_options
--   iostreams
--   system
--   chrono
--   date_time
--   atomic
--   regex
fatal: not a git repository (or any of the parent directories): .git
CMake Warning at /usr/share/cmake-3.10/Modules/FindBoost.cmake:1626 (message):
  No header defined for python-py369; skipping header check
Call Stack (most recent call first):
  CMakeLists.txt:159 (find_package)

CMake Warning at /usr/share/cmake-3.10/Modules/FindBoost.cmake:1626 (message):
  No header defined for python-py36; skipping header check
Call Stack (most recent call first):
  CMakeLists.txt:166 (find_package)

-- Configuring architecture : ecp5
-- Configuring done
-- Generating done
-- Build files have been written to: /home/hilaire/Documents/nextpnr-master
hilaire@hilaire-VirtualBox:~/Documents/nextpnr-master$ make
[  1%] Automatic MOC for target gui_ecp5
[  1%] Built target gui_ecp5_autogen
[  2%] Generating qrc_nextpnr.cpp
[  3%] Generating qrc_base.cpp
Scanning dependencies of target gui_ecp5
[  4%] Building CXX object generated/gui/ecp5/CMakeFiles/gui_ecp5.dir/application.cc.o
[  5%] Building CXX object generated/gui/ecp5/CMakeFiles/gui_ecp5.dir/basewindow.cc.o
[  6%] Building CXX object generated/gui/ecp5/CMakeFiles/gui_ecp5.dir/designwidget.cc.o
[  7%] Building CXX object generated/gui/ecp5/CMakeFiles/gui_ecp5.dir/fpgaviewwidget.cc.o
[  8%] Building CXX object generated/gui/ecp5/CMakeFiles/gui_ecp5.dir/line_editor.cc.o
[ 10%] Building CXX object generated/gui/ecp5/CMakeFiles/gui_ecp5.dir/lineshader.cc.o
[ 11%] Building CXX object generated/gui/ecp5/CMakeFiles/gui_ecp5.dir/pyconsole.cc.o
[ 12%] Building CXX object generated/gui/ecp5/CMakeFiles/gui_ecp5.dir/pythontab.cc.o
[ 13%] Building CXX object generated/gui/ecp5/CMakeFiles/gui_ecp5.dir/treemodel.cc.o
[ 14%] Building CXX object generated/gui/ecp5/CMakeFiles/gui_ecp5.dir/worker.cc.o
[ 15%] Building CXX object generated/gui/ecp5/CMakeFiles/gui_ecp5.dir/ecp5/mainwindow.cc.o
[ 16%] Building CXX object generated/gui/ecp5/CMakeFiles/gui_ecp5.dir/__/3rdparty/python-console/ColumnFormatter.cpp.o
[ 17%] Building CXX object generated/gui/ecp5/CMakeFiles/gui_ecp5.dir/__/3rdparty/python-console/ParseHelper.cpp.o
[ 18%] Building CXX object generated/gui/ecp5/CMakeFiles/gui_ecp5.dir/__/3rdparty/python-console/ParseHelper.BlockParseState.cpp.o
[ 20%] Building CXX object generated/gui/ecp5/CMakeFiles/gui_ecp5.dir/__/3rdparty/python-console/ParseHelper.BracketParseState.cpp.o
[ 21%] Building CXX object generated/gui/ecp5/CMakeFiles/gui_ecp5.dir/__/3rdparty/python-console/ParseHelper.ContinuationParseState.cpp.o
[ 22%] Building CXX object generated/gui/ecp5/CMakeFiles/gui_ecp5.dir/__/3rdparty/python-console/ParseMessage.cpp.o
[ 23%] Building CXX object generated/gui/ecp5/CMakeFiles/gui_ecp5.dir/__/3rdparty/python-console/modified/pyredirector.cc.o
[ 24%] Building CXX object generated/gui/ecp5/CMakeFiles/gui_ecp5.dir/__/3rdparty/python-console/modified/pyinterpreter.cc.o
[ 25%] Building CXX object generated/gui/ecp5/CMakeFiles/gui_ecp5.dir/__/3rdparty/imgui/imgui_widgets.cpp.o
[ 26%] Building CXX object generated/gui/ecp5/CMakeFiles/gui_ecp5.dir/__/3rdparty/imgui/imgui_draw.cpp.o
[ 27%] Building CXX object generated/gui/ecp5/CMakeFiles/gui_ecp5.dir/__/3rdparty/imgui/imgui.cpp.o
[ 28%] Building CXX object generated/gui/ecp5/CMakeFiles/gui_ecp5.dir/__/3rdparty/imgui/imgui_demo.cpp.o
[ 30%] Building CXX object generated/gui/ecp5/CMakeFiles/gui_ecp5.dir/__/3rdparty/qtimgui/ImGuiRenderer.cpp.o
[ 31%] Building CXX object generated/gui/ecp5/CMakeFiles/gui_ecp5.dir/__/3rdparty/qtimgui/QtImGui.cpp.o
[ 32%] Building CXX object generated/gui/ecp5/CMakeFiles/gui_ecp5.dir/qrc_base.cpp.o
[ 33%] Building CXX object generated/gui/ecp5/CMakeFiles/gui_ecp5.dir/qrc_nextpnr.cpp.o
[ 34%] Building CXX object generated/gui/ecp5/CMakeFiles/gui_ecp5.dir/gui_ecp5_autogen/mocs_compilation.cpp.o
[ 35%] Linking CXX static library libgui_ecp5.a
[ 35%] Built target gui_ecp5
[ 36%] Building CXX object CMakeFiles/bbasm.dir/bba/main.cc.o
[ 37%] Linking CXX executable bbasm
[ 37%] Built target bbasm
[ 38%] Generating ecp5/chipdbs/chipdb-25k.bba
Traceback (most recent call last):
  File "/home/hilaire/Documents/nextpnr-master/ecp5/trellis_import.py", line 3, in <module>
    import database
ModuleNotFoundError: No module named 'database'
CMakeFiles/ecp5_chipdb.dir/build.make:81: recipe for target 'ecp5/chipdbs/chipdb-25k.bba' failed
make[2]: *** [ecp5/chipdbs/chipdb-25k.bba] Error 1
CMakeFiles/Makefile2:175: recipe for target 'CMakeFiles/ecp5_chipdb.dir/all' failed
make[1]: *** [CMakeFiles/ecp5_chipdb.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

daveshah1, 2020-04-22:
TRELLIS_INSTALL_PREFIX should just be /opt/trellis

ecp5fpgauser, 2020-04-22:
Thanks daveshah1 for your comment, but I am afraid it does not fix my issue. This time with your advice, a make clean, cmake and make, I face the same kind of issue, here is in attached file the result (No module named 'pip_classes' error).
[compile2.log](https://github.com/YosysHQ/nextpnr/files/4517378/compile2.log)

ecp5fpgauser, 2020-04-23:
It seems I am not alone to face this issue, see here: https://gist.github.com/gojimmypi/f96cd86b2b8595b4cf3be4baf493c5a7 on ligne 527.
I set PYTHONPATH variable with export PYTHONPATH="~/workspace/prjtrellis/database" without any success. Any idea ?

daveshah1, 2020-04-23:
How did you install Project Trellis?

ecp5fpgauser, 2020-04-23:
Here is my method: I've downloaded yosys, prjtrellis and nextpnr source code on github.com, and then I followed instructions from here: https://mindchasers.com/dev/tools-fpga-open-source.

I decided to restart from scratch, this time by cloning the git projects. I faced the out of memory error when building nextpnr project, then I increased RAM of my virtual environment. And now no more compilation issue! I am ready to test the flow on an existing project. Thank you for your support.

sam-falvo, 2020-05-07:
My compilation process fails in a different spot, but with the same underlying cause: Python cannot find the database module.  I'm building on Pop_OS! 20.04 on 64-bit Linux if that helps.  What's left on the screen is posted below for reference as well:

```
[ 98%] Building CXX object CMakeFiles/nextpnr-ice40.dir/ice40/pcf.cc.o
/home/kc5tja/git/build-eda-tools/nextpnr/common/router2.cc: In member function ‘bool nextpnr_ice40::{anonymous}::Router2::is_wire_undriveable(nextpnr_ice40::WireId)’:
/home/kc5tja/git/build-eda-tools/nextpnr/common/router2.cc:370:19: warning: variable ‘p’ set but not used [-Wunused-but-set-variable]
  370 |         for (auto p : ctx->getPipsUphill(wire))
      |                   ^
[100%] Linking CXX executable nextpnr-ice40
make[3]: Leaving directory '/home/kc5tja/git/build-eda-tools/nextpnr'
[100%] Built target nextpnr-ice40
make[2]: Leaving directory '/home/kc5tja/git/build-eda-tools/nextpnr'
make[1]: Leaving directory '/home/kc5tja/git/build-eda-tools/nextpnr'
[  2%] Built target bbasm
[ 18%] Built target ice40_chipdb
[ 19%] Automatic MOC for target gui_ice40
[ 19%] Built target gui_ice40_autogen
[ 52%] Built target gui_ice40
[ 53%] Automatic MOC for target QtPropertyBrowser
[ 53%] Built target QtPropertyBrowser_autogen
[ 65%] Built target QtPropertyBrowser
[100%] Built target nextpnr-ice40
Install the project...
-- Install configuration: "Release"
-- Installing: /usr/local/bin/nextpnr-ice40
(cd nextpnr && cmake -DARCH=ecp5 -DTRELLIS_ROOT=../prjtrellis . && make -j && sudo make install)
CMake Warning at /usr/share/cmake-3.16/Modules/FindBoost.cmake:2020 (message):
  No header defined for python-py382; skipping header check (note:
  header-only libraries have no designated component)
Call Stack (most recent call first):
  CMakeLists.txt:159 (find_package)

CMake Warning at /usr/share/cmake-3.16/Modules/FindBoost.cmake:2020 (message):
  No header defined for python-py38; skipping header check (note: header-only
  libraries have no designated component)
Call Stack (most recent call first):
  CMakeLists.txt:166 (find_package)

CMake Warning at /usr/share/cmake-3.16/Modules/FindBoost.cmake:2020 (message):
  No header defined for python-py3; skipping header check (note: header-only
  libraries have no designated component)
Call Stack (most recent call first):
  CMakeLists.txt:166 (find_package)

-- Configuring architecture : ecp5
-- TRELLIS_INSTALL_PREFIX not defined using -DTRELLIS_INSTALL_PREFIX=/path-prefix/to/prjtrellis-installation. Defaulted to /usr/local
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    TRELLIS_ROOT

-- Build files have been written to: /home/kc5tja/git/build-eda-tools/nextpnr
make[1]: Entering directory '/home/kc5tja/git/build-eda-tools/nextpnr'
make[2]: Entering directory '/home/kc5tja/git/build-eda-tools/nextpnr'
make[3]: Entering directory '/home/kc5tja/git/build-eda-tools/nextpnr'
make[3]: Entering directory '/home/kc5tja/git/build-eda-tools/nextpnr'
make[3]: Entering directory '/home/kc5tja/git/build-eda-tools/nextpnr'
Scanning dependencies of target gui_ecp5_autogen
make[3]: Leaving directory '/home/kc5tja/git/build-eda-tools/nextpnr'
make[3]: Leaving directory '/home/kc5tja/git/build-eda-tools/nextpnr'
make[3]: Leaving directory '/home/kc5tja/git/build-eda-tools/nextpnr'
make[3]: Entering directory '/home/kc5tja/git/build-eda-tools/nextpnr'
make[3]: Entering directory '/home/kc5tja/git/build-eda-tools/nextpnr'
[  1%] Automatic MOC for target QtPropertyBrowser
[  2%] Automatic MOC for target gui_ecp5
[  4%] Built target bbasm
make[3]: Entering directory '/home/kc5tja/git/build-eda-tools/nextpnr'
[  5%] Generating ecp5/chipdbs/chipdb-25k.bba
make[3]: Leaving directory '/home/kc5tja/git/build-eda-tools/nextpnr'
[  5%] Built target QtPropertyBrowser_autogen
make[3]: Entering directory '/home/kc5tja/git/build-eda-tools/nextpnr'
make[3]: Leaving directory '/home/kc5tja/git/build-eda-tools/nextpnr'
[ 18%] Built target QtPropertyBrowser
Traceback (most recent call last):
  File "/home/kc5tja/git/build-eda-tools/nextpnr/ecp5/trellis_import.py", line 3, in <module>
    import database
ModuleNotFoundError: No module named 'database'
make[3]: *** [CMakeFiles/ecp5_chipdb.dir/build.make:82: ecp5/chipdbs/chipdb-25k.bba] Error 1
make[3]: Leaving directory '/home/kc5tja/git/build-eda-tools/nextpnr'
make[2]: *** [CMakeFiles/Makefile2:165: CMakeFiles/ecp5_chipdb.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....
make[3]: Leaving directory '/home/kc5tja/git/build-eda-tools/nextpnr'
[ 18%] Built target gui_ecp5_autogen
make[2]: Leaving directory '/home/kc5tja/git/build-eda-tools/nextpnr'
make[1]: *** [Makefile:130: all] Error 2
make[1]: Leaving directory '/home/kc5tja/git/build-eda-tools/nextpnr'
make: *** [Makefile:77: nextpnr] Error 2
kc5tja@pop-os:~/git/build-eda-tools$ 

```

Also, in case it's useful, my script for building everything can be found at https://github.com/sam-falvo/build-eda-tools ; I first do a `make clean` then `make debian`.

daveshah1, 2020-05-07:
There was a change in the install procedure. Trellis now needs to be installed, TRELLIS_ROOT is not used, and TRELLIS_INSTALL_PREFIX must be set to the prefix where trellis was installed (such as /usr or /usr/local).

sam-falvo, 2020-05-07:
I already have project Trellis built and installed.  I changed the script to use TRELLIS_INSTALL_PREFIX instead of TRELLIS_ROOT, but the outcome is the same.

daveshah1, 2020-05-07:
Can you post the CMake command line and output?

sam-falvo, 2020-05-07:
```
kc5tja@pop-os:~/git/build-eda-tools$ (cd nextpnr && cmake -DARCH=ecp5 -DTRELLIS_INSTALL_PREFIX=../prjtrellis .) 2>&1 | tee /tmp/cmake-output
CMake Warning at /usr/share/cmake-3.16/Modules/FindBoost.cmake:2020 (message):
  No header defined for python-py382; skipping header check (note:
  header-only libraries have no designated component)
Call Stack (most recent call first):
  CMakeLists.txt:159 (find_package)

CMake Warning at /usr/share/cmake-3.16/Modules/FindBoost.cmake:2020 (message):
  No header defined for python-py38; skipping header check (note: header-only
  libraries have no designated component)
Call Stack (most recent call first):
  CMakeLists.txt:166 (find_package)

CMake Warning at /usr/share/cmake-3.16/Modules/FindBoost.cmake:2020 (message):
  No header defined for python-py3; skipping header check (note: header-only
  libraries have no designated component)
Call Stack (most recent call first):
  CMakeLists.txt:166 (find_package)

-- Configuring architecture : ecp5
-- Configuring done
-- Generating done
-- Build files have been written to: /home/kc5tja/git/build-eda-tools/nextpnr
kc5tja@pop-os:~/git/build-eda-tools$ sudo apt install python3-dev
[sudo] password for kc5tja: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
python3-dev is already the newest version (3.8.2-0ubuntu2).
0 upgraded, 0 newly installed, 0 to remove and 88 not upgraded.

```

sam-falvo, 2020-05-07:
Here's the output of the complete build process.  Built with:  `(cd nextpnr && cmake -DARCH=ecp5 -DTRELLIS_INSTALL_PREFIX=../prjtrellis . && make -j$(nproc) && sudo make install) 2>&1 | tee /tmp/nextpnr-build-output.log`

https://pastebin.com/ayNESzQX

daveshah1, 2020-05-07:
Are you sure the install prefix of trellis is ../prjtrellis ? Usually it would be /usr, /usr/local, or somewhere in /opt ?

sam-falvo, 2020-05-07:
It looks as if the ENV_CMD setting in the CMake family file is missing the directory where the database directory resides in (if I'm understanding how Trellis is laid out).

sam-falvo, 2020-05-07:
Yes, it's built in a scripted way, and is explicitly designed to not use system directories, precisely to avoid cases where OS updates destroy the installation (which I've had happen before).

daveshah1, 2020-05-07:
Can you post the output of `tree ../prjtrellis` ? It might also be worth deleting CMakeCache.txt just in case some old config is lingering.

sam-falvo, 2020-05-07:
Ooooh, wait a minute...  I think I goofed.  I just noticed my makefile does use /usr.  Trying to rebuild to double-check with that setting.  I'll be kicking myself if this works.  ;)

sam-falvo, 2020-05-07:
It's at least getting along further now.  Sorry about this; I was completely overlooking the distinction between install and build locations, because that's how I basically set everything else up.

daveshah1, 2020-07-05:
Closing as all issues here seem to be resolved.

closed 2020-07-05 by daveshah1

sam-falvo, 2020-07-05:
They are, thank you!  Sorry, I forgot to update the ticket.