YosysHQ/nextpnr #430: Python binding for IdString

smunaut, 2020-04-24:
Trying to view an IdString in the python shell (like to see the 'Key' value in an IdMap) always results in :

TypeError: No to_python (by-value) converter found for C++ type: nextpnr_ice40::IdString

It would be nice if this was mapped to a python string.

daveshah1, 2020-04-24:
This should happen for all IdStrings, looks like a wrapper is missing here.

smunaut, 2020-04-24:
Just tried the fixes in that PR and although it fixes it for list(ctx.cells)[0][0]  it still happens for others :

>>> list(lc.ports)[0][0]

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: No to_python (by-value) converter found for C++ type: nextpnr_ice40::IdString

>>> list(lc.attrs)[0][0]

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: No to_python (by-value) converter found for C++ type: nextpnr_ice40::IdString

daveshah1, 2020-04-24:
Thanks for spotting, should be fixed for all cases now.

smunaut, 2020-04-24:
Yup, seems to all work now, tx !

PR #433 (2020-04-24): python: Miscellaneous fixes
Fixes #430 
Fixes #431 - autocomplete still isn't very useful but I suspect that is a Python issue rather than anything we can fix, and it doesn't crash any more.
closed 2020-04-24 by daveshah1