YosysHQ/nextpnr #438: Yosys and nextpnr issues with official Lattice serdes eye demo project

ecp5fpgauser, 2020-05-12:
When trying to synthesize Lattice SERDES Eye Demo for Versa 5G demo using yosys/nextpnr/prjtrellis tools, I face several issues:
Error: Module 'DCUA' referenced in module 'test1' in cell 'DCU0_inst' does not have a parameter named 'CH1_TXDEPRE'
and same error message with 'CH1_TXDEPOST'
If I comment out these 2 lines, I get the following error message: Output port top.Serdes_inst.DCU0_inst.CH0_FFS_CC_OVERRUN (DCUA) is connected to constants: 1'z
If I comment out all all lines with this kind of assignments (in test1.v file), I get the following error message: "IO 'ref_clk_refclkn' is unconstrained in LPF"
If I add the following line FREQUENCY NET "FREQUENCY NET "ref_clk_refclkn" 160.00000 MHz ;" in LPF file, I get the following messages:
"Info: constraining clock net 'ref_clk_refclkn' to 160.00 MHz" and
"Error: IO 'ref_clk_refclkn' is unconstrained in LPF"
How is it possible to get these 2 messages at the same time ?
This project is provided here : https://www.latticesemi.com/en/Products/DevelopmentBoardsAndKits/ECP5VersaDevelopmentKit
Do you have an idea of what is going wrong here ?

daveshah1, 2020-05-12:
> Error: Module 'DCUA' referenced in module 'test1' in cell 'DCU0_inst' does not have a parameter named 'CH1_TXDEPRE'

Looks like these parameters are currently missing from cells_bb.v in Yosys

> Output port top.Serdes_inst.DCU0_inst.CH0_FFS_CC_OVERRUN (DCUA) is connected to constants: 1'z

I would need to check the Verilog standard to see if this is a Yosys bug or an error in the example design.

> "Error: IO 'ref_clk_refclkn' is unconstrained in LPF"

This is referring to the lack of a pin constraint rather than a clock constraint issue. Support for the reference clock primitive may not be exactly the same as Diamond. In general SERDES related features have seen limited testing and should be considered experimental.

ecp5fpgauser, 2020-05-12:
"Error: IO 'ref_clk_refclkn' is unconstrained in LPF": it seems that LPF file parser is lost when comment "//" is used. "#" has to be used instead. No more error message like the one above when replacing "//" by "#".
Now I get a new error message: DCU must be constrained to a Bel! Packing design failed. How to go further ?

daveshah1, 2020-05-12:
You need a DCU attribute on the DCUA primitive - you might need to convert the comments that Diamond uses into Verilog-2001 style attributes (that both Yosys and Diamond will accept).

ecp5fpgauser, 2020-05-13:
Now I am using "#" char to comment a line in LPF files.
What do you mean by "you need a DCU attribute on the DUCA primitive" ? Could you please give me an example ? Thanks.

daveshah1, 2020-05-13:
Instead of writing `DCUA dcu_i (...) /* synthesis DCU=DCU0 */;` using a hot comment as Lattice do, use a Verilog-2001 attribute like `(* DCU="DCU0" *) DCUA dcu_i (...);`

ecp5fpgauser, 2020-05-13:
Thanks to your previous advice, I can go further without getting anymore error message "DCU must be constrained to a Bel". I face a new error message: "failed to route HPBX0200 to X42/Y5/SLICEB.DI1". How to investigate this issue ?

daveshah1, 2020-05-14:
It seems to be a problem involving a clock being connected to the data input of a flipflop, which is not a case the clock routing in nextpnr has been tested with (in general I've found a few oddities involving clocks used as logic). I am investigating.

daveshah1, 2020-05-14:
With the fixes in #440 the design at least builds OK, I don't have any SMA cables to hand (and I think the ispCLOCK config on my Versa is not configured right) so I can't do a more thorough test. I had to add `-abc9` to `synth_ecp5` in Yosys to get the design to pass timing with nextpnr. I would be interested to see if it works for you on hardware.

whitequark, 2020-05-14:
I have Versa, SMA cables and necessary test equipment so I can test this if necessary. I'm not sure what to look for though.

ecp5fpgauser, 2020-05-14:
Thank you daveshah1, I'm gonna try this at my end -abc9 option for synth_ecp5 command. I will be in position to test this using Versa board and SMA cables only on next Wednesday. I will let you know.

daveshah1, 2020-05-14:
@whitequark if you want to play with this, the project and bitstreams are here:
[ECP5_5G_Serdes.zip](https://github.com/YosysHQ/nextpnr/files/4628363/ECP5_5G_Serdes.zip)

The demo instructions are at https://www.latticesemi.com/-/media/LatticeSemi/Documents/UserManuals/RZ/SERDESEyeDemoECP5VersaDevKitUG93.ashx?document_id=50998

ecp5fpgauser, 2020-05-14:
I have tested successfully the build of the project with lattice-fixes branch. Thank you daveshah1! One comment: if I omit --lpf-allow-unconstrained option, then 1 constraint is not met: "error, max frequency for clock '$glbnet$test1_rx_pclk': 149.50 MHz (FAIL at 160.00MHz)". This error is not reported by Lattice Diamond toolchain.

daveshah1, 2020-05-14:
I found the -abc9 in Yosys improved the timing enough to fix that issue.

ecp5fpgauser, 2020-05-18:
I've tried on this morning daveshah1 project on Lattice Versa 5G board. Good news! I am able to see an eye diagram on scope, and I measured a total jitter of 82ps (ie. 0.4UI), which is the same value when using Lattice Diamond toolchain.
I used SMA cables for this.
I used openFPGALoader to load  bitstream. I used Lattice ddtcmd program to convert bitstream into SPI Flash program, and then Lattice pgrcmd program to program external SPI flash on the Versa board. Can OpenOCD program able to manage these tasks?
2 suggestions:
- is it appropriate to add CH1_TXDEPRE and CH1_TXDEPOST in cells_bb.v file ?
- is it also appropriate to add this demo project in prjtrellis/examples directory ?
 Thank you one more time davesh1.

daveshah1, 2020-05-18:
ecpprog by Greg Davill can be used for both SRAM and SPI flash programming over JTAG: https://github.com/gregdavill/ecpprog. OpenOCD with a SVF file is only recommended for SRAM programming, there are some Python scripts floating around for flash programming with it but I can't recommend that as it is very slow.

> is it appropriate to add CH1_TXDEPRE and CH1_TXDEPOST in cells_bb.v file ?

This should be done now

> is it also appropriate to add this demo project in prjtrellis/examples directory ?

No, as it is Lattice copyrighted code.

gatecat, 2022-08-18:
I believe the main issues here are now fixed.

closed 2022-08-18 by gatecat