YosysHQ/nextpnr #444: WSL Ubuntu 20.04: cannot find -lpthreads

gojimmypi, 2020-05-19:
This issue is in reference to https://github.com/YosysHQ/nextpnr/pull/443#issuecomment-630538198

I'll see what I can do to come up with how to better demonstrate this. In the meantime: I have a Windows 10 VM that I can repeatedly start from the same fresh install point for this WSL build. Curiously, my most recent build was _successful_. How could that be? I fussed with it for hours yesterday with the ` cannot find -lpthreads` error. So I ran the [build script](https://github.com/gojimmypi/ulx3s-toolchain/blob/master/install_barebones.sh) _again_, that essentially fetches the latest code, and (regardless of change or not) does a `make clean` to rebuild. This rebuild is where it is _failing_ again! Unbelievable. Not the first time, only the second. Here's the full error that I see:

```
[ 97%] Building CXX object CMakeFiles/nextpnr-ecp5.dir/ecp5/pack.cc.o
{standard input}: Assembler messages:
{standard input}:27374108: Internal error (Segmentation fault).
Please report this bug.
as: malloc.c:2379: sysmalloc: Assertion `(old_top == initial_top (av) && old_size == 0) || ((unsigned long) (old_size) >= MINSIZE && prev_inuse (old_top) && ((unsigned long) old_end & (pagesize - 1)) == 0)' failed.
{standard input}:27374108: Internal error (Aborted).
Please report this bug.
as: malloc.c:2379: sysmalloc: Assertion `(old_top == initial_top (av) && old_size == 0) || ((unsigned long) (old_size) >= MINSIZE && prev_inuse (old_top) && ((unsigned long) old_end & (pagesize - 1)) == 0)' failed.
c++: internal compiler error: Aborted signal terminated program as
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-9/README.Bugs> for instructions.
make[2]: *** [CMakeFiles/nextpnr-ecp5.dir/build.make:362: CMakeFiles/nextpnr-ecp5.dir/ecp5/baseconfigs.cc.o] Error 4
make[2]: *** Deleting file 'CMakeFiles/nextpnr-ecp5.dir/ecp5/baseconfigs.cc.o'
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:195: CMakeFiles/nextpnr-ecp5.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
---------------------------
Looking in ./CMakeFiles/CMakeOutput.log for errors...
---------------------------
---------------------------
Looking in ./CMakeFiles/CMakeError.log for errors...
---------------------------
/usr/bin/ld: src.c:(.text+0x52): undefined reference to `pthread_detach'
/usr/bin/ld: src.c:(.text+0x63): undefined reference to `pthread_join'
collect2: error: ld returned 1 exit status
make[1]: *** [CMakeFiles/cmTC_659ac.dir/build.make:87: cmTC_659ac] Error 1
make[1]: Leaving directory '/mnt/c/workspace/nextpnr/CMakeFiles/CMakeTmp'
make: *** [Makefile:121: cmTC_659ac/fast] Error 2
--
/usr/bin/cc  -DCHECK_FUNCTION_EXISTS=pthread_create    -rdynamic CMakeFiles/cmTC_63926.dir/CheckFunctionExists.c.o  -o cmTC_63926  -lpthreads
/usr/bin/ld: cannot find -lpthreads
collect2: error: ld returned 1 exit status
make[1]: *** [CMakeFiles/cmTC_63926.dir/build.make:87: cmTC_63926] Error 1
make[1]: Leaving directory '/mnt/c/workspace/nextpnr/CMakeFiles/CMakeTmp'
make: *** [Makefile:121: cmTC_63926/fast] Error 2
Error detected. Try updating dependencies. Press enter to continue, or Ctrl-C to abort.
``` 
and here's what's in `CMakeError.log`
```
Performing C SOURCE FILE Test CMAKE_HAVE_LIBC_PTHREAD failed with the following output:
Change Dir: /mnt/c/workspace/nextpnr/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/make cmTC_659ac/fast && /usr/bin/make -f CMakeFiles/cmTC_659ac.dir/build.make CMakeFiles/cmTC_659ac.dir/build
make[1]: Entering directory '/mnt/c/workspace/nextpnr/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_659ac.dir/src.c.o
/usr/bin/cc   -DCMAKE_HAVE_LIBC_PTHREAD   -o CMakeFiles/cmTC_659ac.dir/src.c.o   -c /mnt/c/workspace/nextpnr/CMakeFiles/CMakeTmp/src.c
Linking C executable cmTC_659ac
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_659ac.dir/link.txt --verbose=1
/usr/bin/cc  -DCMAKE_HAVE_LIBC_PTHREAD    -rdynamic CMakeFiles/cmTC_659ac.dir/src.c.o  -o cmTC_659ac 
/usr/bin/ld: CMakeFiles/cmTC_659ac.dir/src.c.o: in function `main':
src.c:(.text+0x46): undefined reference to `pthread_create'
/usr/bin/ld: src.c:(.text+0x52): undefined reference to `pthread_detach'
/usr/bin/ld: src.c:(.text+0x63): undefined reference to `pthread_join'
collect2: error: ld returned 1 exit status
make[1]: *** [CMakeFiles/cmTC_659ac.dir/build.make:87: cmTC_659ac] Error 1
make[1]: Leaving directory '/mnt/c/workspace/nextpnr/CMakeFiles/CMakeTmp'
make: *** [Makefile:121: cmTC_659ac/fast] Error 2

Source file was:
#include <pthread.h>

void* test_func(void* data)
{
  return data;
}

int main(void)
{
  pthread_t thread;
  pthread_create(&thread, NULL, test_func, NULL);
  pthread_detach(thread);
  pthread_join(thread, NULL);
  pthread_atfork(NULL, NULL, NULL);
  pthread_exit(NULL);

  return 0;
}

Determining if the function pthread_create exists in the pthreads failed with the following output:
Change Dir: /mnt/c/workspace/nextpnr/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/make cmTC_63926/fast && /usr/bin/make -f CMakeFiles/cmTC_63926.dir/build.make CMakeFiles/cmTC_63926.dir/build
make[1]: Entering directory '/mnt/c/workspace/nextpnr/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_63926.dir/CheckFunctionExists.c.o
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=pthread_create   -o CMakeFiles/cmTC_63926.dir/CheckFunctionExists.c.o   -c /usr/share/cmake-3.16/Modules/CheckFunctionExists.c
Linking C executable cmTC_63926
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_63926.dir/link.txt --verbose=1
/usr/bin/cc  -DCHECK_FUNCTION_EXISTS=pthread_create    -rdynamic CMakeFiles/cmTC_63926.dir/CheckFunctionExists.c.o  -o cmTC_63926  -lpthreads 
/usr/bin/ld: cannot find -lpthreads
collect2: error: ld returned 1 exit status
make[1]: *** [CMakeFiles/cmTC_63926.dir/build.make:87: cmTC_63926] Error 1
make[1]: Leaving directory '/mnt/c/workspace/nextpnr/CMakeFiles/CMakeTmp'
make: *** [Makefile:121: cmTC_63926/fast] Error 2
```
I have a [version.sh]() that reports:
```
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 20.04 LTS
Release:        20.04
Codename:       focal
Python 2.7.18rc1
Python 3.8.2
cmake version 3.16.3

CMake suite maintained and supported by Kitware (kitware.com/cmake).
clang version 10.0.0-4ubuntu1
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/bin
qtf_default Version: 5.12.8+dfsg-0ubuntu1
libboost-all-dev Version: 1.71.0.0ubuntu2
build-essential Version: 12.8ubuntu1
clang Version: 1:10.0-50~exp1
bison Version: 2:3.5.1+dfsg-1
flex Version: 2.6.4-6.2
libreadline-dev Version: 8.0-4
gawk Version: 1:5.0.1+dfsg-1
tcl-dev Version: 8.6.9+1
libffi-dev Version: 3.3-4
git Version: 1:2.25.1-1ubuntu3
mercurial Version: 5.3.1-1ubuntu1
graphviz Version: 2.42.2-3build2
xdot Version: 1.1-2
pkg-config Version: 0.29.1-0ubuntu4
python
python3 Version: 3.8.2-0ubuntu2
libftdi-dev Version: 0.20-4build8
qt5-default Version: 5.12.8+dfsg-0ubuntu1
python3-dev Version: 3.8.2-0ubuntu2
libboost-dev Version: 1.71.0.0ubuntu2
nextpnr-ecp5: error while loading shared libraries: libQt5Core.so.5: cannot open shared object file: No such file or directory
```
Perhaps that last line is the most insightful? 
```
which nextpnr-ecp5
/usr/local/bin/nextpnr-ecp5
$ ls -al /usr/local/bin/nextpnr-ecp5
-rwxr-xr-x 1 root root 351722168 May 18 15:05 /usr/local/bin/nextpnr-ecp5
```
That's the `nextpnr-ecp5` that was built successfully this afternoon. Yet when running it, gives that error:
```
error while loading shared libraries: libQt5Core.so.5: cannot open shared object file: No such file or directory
```
This is just too weird.

daveshah1, 2020-05-19:
Can you run `which cmake`?

daveshah1, 2020-05-19:
Also, as an addendum, as I've mentioned previously, the ` undefined reference to `pthread_detach'` etc in CMakeError.log occurs from CMake trying different possibilties for `pthread`. It is **not** an error that you need to worry about - as working `pthread` is found later on - and I see it here too with a fully working build.

The two genuine issues are the segmentation fault in `as`, which is either a compiler bug or possibly running out of memory; and the inability to find `libQt5Core.so.5`.

gojimmypi, 2020-05-19:
oof. I'm such a newbie. Indeed it was all apparently a low-memory issue. For some reason I thought WSL on Windows would swap more to the disk and perhaps be doggedly slow... but not actually have such serious problems. I bumped the VM memory from the default 2 GB to the 5GB (that I even warn about - and ignored - in the script I wrote for "real" Linux!)... and all is well.

I wonder if it would be worthwhile to have some sort of low memory message in the Makefile?

```
$ which cmake
/usr/bin/cmake
```

closed 2020-05-19 by gojimmypi

daveshah1, 2020-05-19:
The real bug is in gcc, which should fail more gracefully if it runs out of memory, imo

smunaut, 2020-05-19:
At least on linux gcc can hardly do better because of overcommit it always gets a valid pointer on malloc and when it tries to access it, if the kernel can't actually allocate memory, it will send a signal to terminate the process ... and gcc will have no clue where that is coming from.

gojimmypi, 2020-05-19:
I just realized that the `apt-get install cmake` on WSL Ubuntu 20.04 gave me cmake version 3.16.3, the [prerequisites](https://github.com/YosysHQ/nextpnr#prerequisites) indicate CMake 3.3 or later. Although a successful compile, `nextpnr-ecp5` is still not working for me. Is this because of the wrong `cmake`?

```
$ nextpnr-ecp5 --version
nextpnr-ecp5: error while loading shared libraries: libQt5Core.so.5: cannot open shared object file: No such file or directory

$ sudo apt-get install libqt5gui5
Reading package lists... Done
Building dependency tree
Reading state information... Done
libqt5gui5 is already the newest version (5.12.8+dfsg-0ubuntu1).
libqt5gui5 set to manually installed.
The following package was automatically installed and is no longer required:
  guile-2.0-libs
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```

daveshah1, 2020-05-19:
3.16 is much newer than 3.3 so that is fine. It is possible that the Qt configuration is broken though. What does `ldd nextpnr-ecp5` say?

gojimmypi, 2020-05-19:
```
$ which nextpnr-ecp5
/usr/local/bin/nextpnr-ecp5

$ cd /usr/local/bin/
$ ldd nextpnr-ecp5
        linux-vdso.so.1 (0x00007fffdd1bd000)
        libboost_python38.so.1.71.0 => /lib/x86_64-linux-gnu/libboost_python38.so.1.71.0 (0x00007fb9b8eb0000)
        libboost_filesystem.so.1.71.0 => /lib/x86_64-linux-gnu/libboost_filesystem.so.1.71.0 (0x00007fb9b8e90000)
        libboost_thread.so.1.71.0 => /lib/x86_64-linux-gnu/libboost_thread.so.1.71.0 (0x00007fb9b8e60000)
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fb9b8e3d000)
        libboost_program_options.so.1.71.0 => /lib/x86_64-linux-gnu/libboost_program_options.so.1.71.0 (0x00007fb9b8da0000)
        libpython3.8.so.1.0 => /lib/x86_64-linux-gnu/libpython3.8.so.1.0 (0x00007fb9b8850000)
        libQt5Widgets.so.5 => /lib/x86_64-linux-gnu/libQt5Widgets.so.5 (0x00007fb9b81b0000)
        libQt5Gui.so.5 => /lib/x86_64-linux-gnu/libQt5Gui.so.5 (0x00007fb9b7bc0000)
        libQt5Core.so.5 => not found
        libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fb9b79d0000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fb9b7881000)
        libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fb9b7860000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fb9b7650000)
        /lib64/ld-linux-x86-64.so.2 (0x00007fb9c0237000)
        libexpat.so.1 => /lib/x86_64-linux-gnu/libexpat.so.1 (0x00007fb9b7620000)
        libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007fb9b7600000)
        libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fb9b75f0000)
        libutil.so.1 => /lib/x86_64-linux-gnu/libutil.so.1 (0x00007fb9b75e0000)
        libQt5Core.so.5 => not found
        libQt5Core.so.5 => not found
        libGL.so.1 => /lib/x86_64-linux-gnu/libGL.so.1 (0x00007fb9b7540000)
        libpng16.so.16 => /lib/x86_64-linux-gnu/libpng16.so.16 (0x00007fb9b7500000)
        libharfbuzz.so.0 => /lib/x86_64-linux-gnu/libharfbuzz.so.0 (0x00007fb9b73f0000)
        libGLdispatch.so.0 => /lib/x86_64-linux-gnu/libGLdispatch.so.0 (0x00007fb9b7320000)
        libGLX.so.0 => /lib/x86_64-linux-gnu/libGLX.so.0 (0x00007fb9b72e0000)
        libglib-2.0.so.0 => /lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007fb9b71b0000)
        libfreetype.so.6 => /lib/x86_64-linux-gnu/libfreetype.so.6 (0x00007fb9b70f0000)
        libgraphite2.so.3 => /lib/x86_64-linux-gnu/libgraphite2.so.3 (0x00007fb9b70c0000)
        libX11.so.6 => /lib/x86_64-linux-gnu/libX11.so.6 (0x00007fb9b6f80000)
        libpcre.so.3 => /lib/x86_64-linux-gnu/libpcre.so.3 (0x00007fb9b6ef0000)
        libxcb.so.1 => /lib/x86_64-linux-gnu/libxcb.so.1 (0x00007fb9b6ec0000)
        libXau.so.6 => /lib/x86_64-linux-gnu/libXau.so.6 (0x00007fb9b6eb0000)
        libXdmcp.so.6 => /lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007fb9b6ea0000)
        libbsd.so.0 => /lib/x86_64-linux-gnu/libbsd.so.0 (0x00007fb9b6e80000)
```

daveshah1, 2020-05-19:
Is libqt5core5a installed? Does /lib/x86_64-linux-gnu/libQt5Core.so.5 exist?

gojimmypi, 2020-05-19:
I think so...
```
$ ls /lib/x86_64-linux-gnu/libQt5Core.so.5 -al
lrwxrwxrwx 1 root root 20 Apr  9 00:57 /lib/x86_64-linux-gnu/libQt5Core.so.5 -> libQt5Core.so.5.12.8
```

daveshah1, 2020-05-19:
And does /lib/x86_64-linux-gnu/libQt5Core.so.5.12.8 exist?

daveshah1, 2020-05-19:
Looking a bit further I think this could also be a kernel/ABI issue. Some more things to try:

    $ file /lib/x86_64-linux-gnu/libQt5Core.so.5.12.8
    $ uname -a

gojimmypi, 2020-05-19:
```
gojimmypi@DESKTOP-AHQNIJN:/mnt/c/workspace/ulx3s-toolchain$ file /lib/x86_64-linux-gnu/libQt5Core.so.5.12.8
/lib/x86_64-linux-gnu/libQt5Core.so.5.12.8: ELF 64-bit LSB shared object, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, BuildID[sha1]=e54a6cea97c3942fbcec1973574545fb3ad74015, for GNU/Linux 3.17.0, stripped
gojimmypi@DESKTOP-AHQNIJN:/mnt/c/workspace/ulx3s-toolchain$ uname -a
Linux DESKTOP-AHQNIJN 4.4.0-18362-Microsoft #836-Microsoft Mon May 05 16:04:00 PST 2020 x86_64 x86_64 x86_64 GNU/Linux
gojimmypi@DESKTOP-AHQNIJN:/mnt/c/workspace/ulx3s-toolchain$
```

daveshah1, 2020-05-19:
I believe this is an issue in WSL, see microsoft/WSL#3023.

A possible workaround is:

    $ sudo /usr/bin/strip --remove-section=.note.ABI-tag /lib/x86_64-linux-gnu/libQt5Core.so.5.12.8

(but no liability accepted if this breaks your system or causes other issues!)

daveshah1, 2020-05-19:
See also #375

gojimmypi, 2020-05-19:
well, that's certainly discouraging. I know, I know: why am I not using a _real_ Linux box, lol 

I did see [something similar](https://superuser.com/questions/1347723/arch-on-wsl-libqt5core-so-5-not-found-despite-being-installed) but it looked so crazy that surely it wasn't the answer... not to mention that I have this all working just fine on my WSL Ubuntu 18.04. 

Should I open an issue here on this?

I added [some comments on that open WSL#3023](https://github.com/microsoft/WSL/issues/3023#issuecomment-631033201).

Seems odd that issue would be open so long. Even weirder that I was able to get this working in WSL Ubuntu 18.04. I certainly didn't do the `remove tag` thing.  

I think I'll try the 18.04 fresh install in that VM...

daveshah1, 2020-05-19:
Downloading the [18.04 package](https://packages.ubuntu.com/bionic/amd64/libqt5core5a/download) and looking at it:

```
$ file libQt5Core.so.5.9.5 
libQt5Core.so.5.9.5: ELF 64-bit LSB shared object, x86-64, version 1 (GNU/Linux), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, BuildID[sha1]=382bfacc4e104df505a0eb409a5a225145b3aae3, stripped
```

It doesn't have an ABI tag (note the lack of `for GNU/Linux 3.17.0`) so it will work fine regardless of what kernel version WSL claims to be.

Xiretza, 2020-05-20:
I don't use WSL so I have no idea if this will work, but from what I can see in microsoft/WSL#3023 I think it should - try using the `LD_ASSUME_KERNEL` environment variable to override the kernel version the linker compares the library to (see `ld.so(8)`). Something like this:

```
$ env LD_ASSUME_KERNEL=4.4.0 nextpnr-ecp5
```

gojimmypi, 2020-05-20:
although that sounded promising.... no joy. same error.  perhaps a different "assume" value? I tried several:
```
$ uname -a
Linux DESKTOP-AHQNIJN 4.4.0-18362-Microsoft #836-Microsoft Mon May 05 16:04:00 PST 2020 x86_64 x86_64 x86_64 GNU/Linux

$ env LD_ASSUME_KERNEL=4.4.0 nextpnr-ecp5
nextpnr-ecp5: error while loading shared libraries: libQt5Core.so.5: cannot open shared object file: No such file or directory

$ env LD_ASSUME_KERNEL=4.4.0-18362-Microsoft nextpnr-ecp5
nextpnr-ecp5: error while loading shared libraries: libQt5Core.so.5: cannot open shared object file: No such file or directory

$ env LD_ASSUME_KERNEL=4.4.0-18362 nextpnr-ecp5
nextpnr-ecp5: error while loading shared libraries: libQt5Core.so.5: cannot open shared object file: No such file or directory
```
on a working "real" Ubuntu VM, I could not get it to fail with any values of `LD_ASSUME_KERNEL`.

gojimmypi, 2020-05-20:
well, after a [suggestion from a relatively authoritative source](https://twitter.com/unixterminal/status/1262986117596446720?s=20) I was able to hack this to get it working with that `strip` suggestion from @daveshah1 :

```
$ find /usr/. -name "libQt5Core.so.5"
/usr/./lib/x86_64-linux-gnu/libQt5Core.so.5

$ sudo strip --remove-section=.note.ABI-tag /usr/./lib/x86_64-linux-gnu/libQt5Core.so.5

$ nextpnr-ecp5
nextpnr-ecp5 -- Next Generation Place and Route (Version 2d406f3)
```
but OMG what a hack.