YosysHQ/nextpnr #445: Main OS would kill chipdb.py script

kxynos, 2020-05-20:
On my Linux system I had issues completing the compilation since the system would kill nextpnr/ice40/chipdb.py before it completed the task. 

Output of the issue : 
```
$ make -j$(nproc)
Scanning dependencies of target bbasm
[  1%] Building CXX object CMakeFiles/bbasm.dir/bba/main.cc.o
[  2%] Linking CXX executable bbasm
[  2%] Built target bbasm
[  3%] Generating ice40/chipdbs/chipdb-384.bba
[  4%] Generating ice40/chipdbs/chipdb-1k.bba
[  5%] Generating ice40/chipdbs/chipdb-5k.bba
[  6%] Generating ice40/chipdbs/chipdb-u4k.bba
[  7%] Generating ice40/chipdbs/chipdb-8k.bba
Killed
make[2]: *** [CMakeFiles/ice40_chipdb.dir/build.make:131: ice40/chipdbs/chipdb-8k.bba] Error 137
make[1]: *** [CMakeFiles/Makefile2:165: CMakeFiles/ice40_chipdb.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```
The problem is that the Python script output is piped to a file. This creates a number of issues. 

I quick fix is to redirect STDOUT to a file instead. I have implemented the required updates and tested them.

whitequark, 2020-05-21:
> This creates a number of issues.

Could you detail which issues? Your fix seems reasonable to me but I don't understand why it's needed.

smunaut, 2020-05-21:
This seems more like a random running out of memory and the patch either just steers the OOM to some other process or some memory got freed in the mean time or something that's not really the root cause of the issue.

kxynos, 2020-05-21:
> > This creates a number of issues.
> 
> Could you detail which issues? Your fix seems reasonable to me but I don't understand why it's needed.

I had an error in dmesg of the process running out of memory and it being killed by the OS. The system really suffered as nextpnr/ice40/chipdbs/chipdb-8k.bba was assembled. I tried it a few times and it got killed every time. With the modification I had no issues.

The speed up of saving to a file seems marginal on tests on run again just now. I did not run enough tests to say if the speedup is significant overall.

You are correct, something did go wrong. I think piping such a large file, with the processing that occurs, is not ideal. Hence the PR. 

Ideally the Python script should be writing the output to a file and not printing to the console. I used the if statement to maintain the ability to still print to the console when needed. So it is backwards compatible.

whitequark, 2020-05-21:
> This seems more like a random running out of memory and the patch either just steers the OOM to some other process or some memory got freed in the mean time or something that's not really the root cause of the issue.

I believe the patch genuinely reduces memory pressure by avoiding buffering of the output on Python side of things, but I didn't verify that this is actually what's happening.

smunaut, 2020-05-21:
It will buffer a bit more, (because in addition to the python buffer you'd have the pipe), but that should be negligible. If the write to the pipe isn't fast, it will apply back pressure on the writing app (python here).

For instance an infinite generator :

```python
#!/usr/bin/env python3
import sys
d = 1024 * ' ' + '\n';
while True:
	sys.stdout.write(d)
```

Then pipe it to `less` which won't read anything :

```
./gen.py | less
```

And the python is blocked very quickly, there is almost no output buffered. (I mean chipdb is in the order of Gbytes of mem usage, the pipe buffers should be in the 100's of Kbytes ...)

(note that I'm not saying writing to file directly isn't better, I'd just like to understand what's going on ...)

whitequark, 2020-05-21:
> (note that I'm not saying writing to file directly isn't better, I'd just like to understand what's going on ...)

Same here. So my guess was incorrect. I'm not sure what's happening here then.

kxynos, 2020-05-21:
Not sure if this will help. 

I did a simple test to track CPU and memory usage of the two commands, using psrecord. It might be that my system was low on memory and outputting it to a file didn't push it to the limits compared to piping to a file. 

It is clear that there is a higher memory usage for the one using file pipe. 

Time test shell script:
```
/usr/bin/python3 nextpnr/ice40/chipdb.py -p nextpnr/ice40/constids.inc -g nextpnr/ice40/gfx.h --fast /usr/local/share/icebox/timings_hx8k.txt --slow /usr/local/share/icebox/timings_lp8k.txt /usr/local/share/icebox/chipdb-8k.txt -o nextpnr/ice40/chipdbs/chipdb-8k.bba.new & psrecord $(pgrep python3) --interval 1 --plot 0x.output.png --log 0x.output.log

/usr/bin/python3 nextpnr/ice40/chipdb.py -p nextpnr/ice40/constids.inc -g nextpnr/ice40/gfx.h --fast /usr/local/share/icebox/timings_hx8k.txt --slow /usr/local/share/icebox/timings_lp8k.txt /usr/local/share/icebox/chipdb-8k.txt >> nextpnr/ice40/chipdbs/chipdb-8k.bba.new & psrecord $(pgrep python3) --interval 1 --plot 0x.pipe.png --log 0x.pipe.log 
```

File pipe timings(0x.pipe.log) snippet:
```
# Elapsed time   CPU (%)     Real (MB)   Virtual (MB)
       0.003        0.000       29.715       35.703
       1.006       98.700       93.113       99.078
       2.009       98.700      163.598      169.578
       3.014       99.500      215.340      221.328
       4.015       98.900      322.320      328.328
       5.018       97.700      410.457      416.410
       6.019       99.800      507.621      513.375
       7.023       98.700      593.625      599.504
       8.024       97.900      701.844      707.895
       9.026       98.800      746.777      752.688
      10.027       97.900      876.551      882.742
...
```

File output timings(0x.output.log) snippet:
```
# Elapsed time   CPU (%)     Real (MB)   Virtual (MB)
       0.002        0.000       26.074       31.953
       1.004       98.800       91.449       97.328
       2.006       97.800      143.469      149.328
       3.007       97.900      215.453      221.328
       4.009       97.700      306.219      312.078
       5.011       99.900      393.758      399.660
       6.013       98.800      498.051      503.977
       7.015       98.800      571.129      576.973
       8.017       97.800      679.824      685.762
       9.018       97.900      746.945      752.691
      10.025       98.400      856.250      901.434
...
```

Full file listings:

File pipe timings(0x.pipe.log):
```
# Elapsed time   CPU (%)     Real (MB)   Virtual (MB)
       0.000        0.000       29.223       35.203
       1.002       97.900       93.637       99.578
       2.004       99.800      164.836      170.828
       3.005       97.800      215.340      221.328
       4.007       98.800      287.336      293.328
       5.009       98.800      352.859      358.887
       6.012       98.700      470.727      476.773
       7.015       98.700      541.191      547.258
       8.017       97.800      649.996      655.785
       9.019       97.800      746.898      752.688
      10.021       98.800      778.672      784.707
      11.023       96.800      909.680      915.770
      12.025       99.800      999.977     1006.023
      13.027       97.800     1019.293     1025.273
      14.034       99.300     1093.285     1099.578
      15.037       98.700     1145.406     1151.578
      16.040       98.700     1196.941     1203.078
      17.042       99.800     1238.691     1244.828
[... 93 lines cut ...]
      41.094       96.800     2109.133     2115.082
      42.096       99.800     2109.133     2115.082
      43.099       98.700     2109.133     2115.082
      44.100       96.800     2109.133     2115.082
      45.108       99.300     2109.133     2115.082
      46.114       98.400     2109.133     2115.082
      47.117       99.700     2109.133     2115.082
      48.121       96.700     2109.133     2115.082
      49.123       98.700     2109.133     2115.082
      50.125       98.800     2109.133     2115.082
      51.127       95.900     2109.133     2115.082
      52.132       98.500     2109.133     2115.082
      53.134       96.800     2109.133     2115.082
      54.135       98.900     2109.133     2115.082
      55.139       96.600     2160.355     2166.332
      56.142       98.700     2160.355     2166.332
      57.144       96.800     2252.738     2258.832
      58.146       96.800     2252.738     2258.832
      59.150       97.600     2252.734     2258.832
      60.154       97.700     2252.734     2258.832
      61.157       98.700     2252.734     2258.832
      62.162       98.500     2252.734     2258.832
      63.167       98.500      248.168      254.070
[EOF]
```

smunaut, 2020-05-21:
What do you mean by "It is clear that there is a higher memory usage for the one using file pipe."

Both show the exact same peak memory usage of 2252 Mb ...

kxynos, 2020-05-21:
> What do you mean by "It is clear that there is a higher memory usage for the one using file pipe."
> 
> Both show the exact same peak memory usage of 2252 Mb ...
Correct. 
I want to highlight that the progression shows a higher usage from the start.

daveshah1, 2020-06-24:
Given this doesn't actually seem to help much, I am closing the issue and PR.

closed 2020-06-24 by daveshah1

zzdts, 2021-02-15:
I meet this issue when i allocate 2GiB memory to my virtual machine, I solved when  allocate 5.8GiB memory to vm.