YosysHQ/nextpnr #448: ice40up5k pin 35 and 'no suitable BEL found'

hightowera, 2020-05-24:
There is something non-intuitive I'm missing regarding a PLL constraint.  I've reduced my reproduction issue down to this code:

`
	wire sys_HFOSC;

	SB_HFOSC OSCInst0
	(
		.CLKHFEN   (1'b1),
		.CLKHFPU   (1'b1),
		.CLKHF     (sys_HFOSC)   // 48 MHz
	);

	wire pll_LOCK;
	wire pll_CLK;

	SB_PLL40_CORE #(

		.FEEDBACK_PATH ("SIMPLE"),
		.DIVR          (4'b0000),     // DIVR =  0
		.DIVF          (7'b0010011),  // DIVF = 19 / F_VCO = 960 MHz
		.DIVQ          (3'b101),      // DIVQ =  5
		.FILTER_RANGE  (3'b100)       // FILTER_RANGE = 4

	) uut (

		.LOCK          (pll_LOCK),
		.RESETB        (1'b1),
		.BYPASS        (1'b0),
		.REFERENCECLK  (sys_HFOSC),   // 48 MHz
		.PLLOUTCORE    (pll_CLK)      // 30 MHz
	);

	wire ide_CLK = pll_CLK;

	reg [22:0] pwm_COUNT = 23'd0;
 
	always @(posedge ide_CLK)
		pwm_COUNT <= pwm_COUNT + 1;
 
	assign ide_DASP = pwm_COUNT[22];
`
This works fine.  As soon as I add a top level signal (completely un-referenced) and constrain it to pin 35, I get this output

`ERROR: PLL 'uut' couldn't be placed anywhere, no suitable BEL found.
ERROR: Packing design failed.
          - 0 warnings, 2 errors
`

What is special about pin 35 (or a subset including it) and why couldn't it place the PLL considering it uses the internal RC as an input reference and only drives an internal counter?

hightowera, 2020-05-24:
Forgot to add:

$ /usr/local/bin/nextpnr-ice40  --version
nextpnr-ice40 -- Next Generation Place and Route (git sha1 cea27dc)

smunaut, 2020-05-24:
Pin 35 and the PLL live in the same IO tile.

The output of the PLL and the input path of pin 35 are shared and cannot be used at the same time.
So if you use the PLL, you can only use pin 35 as an ouptu.

hightowera, 2020-05-24:
Changed that assignment from inout to output and it cleared the problem.  Sorry for the ticket.  I've reviewed most of the docs for the UP and never ran across the specific cell assignment for the PLL.  The sysCLOCK PLL Design and Usage Guide talks about it in vague terms.  I'm surprised the pin muxing Excel doc on the Lattice web-site doesn't annotate that row.  Thanks for the clarification!

closed 2020-05-24 by hightowera

osresearch, 2020-06-25:
I just ran into this same issue of using both gpio 35 as `inout` and the PLL. Eventually debugged it by finding [a random irc chat log](https://freenode.irclog.whitequark.org/yosys/2019-03-13) where the IO tiles were discussed.  Is it possible to improve the error in some way so that it is clearer what is going on and the limitations around pin 35?

smunaut, 2020-06-25:
It's actually pretty hard to make a proper error message :/  There is quite a few reasons why a PLL BEL might end up being unsuitable for placement. When I rewrote the PLL code I tried making it more explicit but I stopped trying when there was more code to try and print a correct error message than there was to actually place the damn PLL.

Writing a proper doc page about the PLL and just linking to that might be the most viable option.

osresearch, 2020-06-25:
yeah, good errors are really difficult. Having a top-level symbiflow page for the major chip families would be very helpful -- claire's http://www.clifford.at/icestorm/ has a chart of supported parts, and some deep dive pages like http://www.clifford.at/icestorm/ultraplus.html that have quite a bit of detail, although this quirk isn't mentioned.

Heh, I just noticed that my own https://github.com/osresearch/up5k has a pinout image that marks 35 as `DPIO(?)`:

[image]

mithro, 2020-06-25:
@osresearch - Couple of things;
 * I agree that we could do a better job of describing the support status of both parts and boards. I do have http://j.mp/symbiflow-status-doc and there is also some stuff at https://symbiflow.github.io/index.html but it could all be a lot better. Help wanted? :-P
 * SymbiFlow doesn't "own" nor "control" nextpnr in any way -- SymbiFlow is just a project trying to bring together a collection of tools owned by groups like @YosysHQ and @verilog-to-routing. All praise for the hard work on making nextpnr awesome should be directed to people like @daveshah1 and @cliffordwolf. You could think of SymbiFlow as being  like Debian and nextpnr is like the Linux kernel or Gnome desktop environment.