YosysHQ/nextpnr #449: Yosys and nextpnr-ice40 issues with ROM and $readmemh

WillBux, 2020-05-24:
nextpnr-ice40 hangs indefinitely when instantiating a ROM like so. With the verbose flag the command does not print anything. 
```
reg [7:0] ZigZagROM [127:0];
initial $readmemh("ZigZagROM.data", ZigZagROM);
```
When the initial line is commented out the verilog compiles nicely and nextpnr-ice40 works as intended. Also I am never writing to the memory after the initial statement so it should be treating it as ROM.
Does anyone know what I'm doing wrong or Is there another way I should be loading in the memory?

smunaut, 2020-05-24:
This belongs in yosys, not nextpnr.

WillBux, 2020-05-24:
Because I can't share the complete project I wrote up a simple module in an attempt to reproduce the error, but of course, it worked. Just for future reference here is what I did wrong. When reading from the ROM it was in an if statement with a few conditions and I think yosys could not synthesize this into a ROM with a simple enable signal. I got around this by creating a simple ROM module (although it could easily be embedded in my other module this adds clarity).
```
module Rom (
    input clk,
    input [6:0] addr,
    output reg [7:0] data);
    
    reg [7:0] Rom [127:0];
    initial $readmemh("Rom.data", Rom);

    always @(posedge clk)
        data <= Rom[addr];

endmodule
```

closed 2020-05-24 by WillBux