YosysHQ/nextpnr #451: Built nextpnr on Ubuntu 20.04

blakesmith, 2020-06-01:
Not sure if there's something unique about my setup, but for posterity's sake, I wanted to show how I was able to build nextpnr after upgrading to Ubuntu 20.04.

With Ubuntu 20.04 installed, nextpnr was having trouble finding Python during the build (both the include dir, as well as during linking).
After installing libboost-python-dev and python3-dev, I had to configure cmake with (ECP5 arch):

```
cmake -DARCH=ecp5 -DPYTHONLIBS_VERSION_STRING=3.8 -DPYTHON_INCLUDE_DIR=/usr/include/python3.8 -DTRELLIS_INSTALL_PREFIX=/usr .
make -j4
sudo make install
```

I hope this helps someone!

SingularitySurfer, 2020-06-05:
Hi, thank you very much for the heads up! Had the same problem on 20.04 with the build failing with " undefined reference to `_Py_tracemalloc_config' ". 
Configuring cmake to python 3.8 fixed it. :)

blakesmith, 2020-06-05:
@SingularitySurfer Glad it could help! I might dig into the Python 3.8 problem a little bit deeper, so others don't get caught up on this.

For the trellis issue: I was able to make it build using the trellis README recommended prefix:

In prjtrellis:

```
cmake -DCMAKE_INSTALL_PREFIX=/usr .
```

... And then installing nextpnr. What error are you getting? After installing trellis, for ECP5 builds in /usr, here's what the directory structure looks like:

```
blake@blake-XPS-13-9360:~$ find /usr | grep -i trellis
/usr/share/trellis
/usr/share/trellis/util
/usr/share/trellis/util/common
/usr/share/trellis/util/common/nets.py
/usr/share/trellis/util/common/devices.py
/usr/share/trellis/util/common/isptcl.py
/usr/share/trellis/util/common/database.py
/usr/share/trellis/util/common/diamond.py
/usr/share/trellis/util/common/__init__.py
/usr/share/trellis/util/common/tiles.py
/usr/share/trellis/misc
/usr/share/trellis/misc/basecfgs
/usr/share/trellis/misc/basecfgs/empty_lfe5u-85f.config
/usr/share/trellis/misc/basecfgs/README.md
/usr/share/trellis/misc/basecfgs/empty_lfe5um5g-25f.config
/usr/share/trellis/misc/basecfgs/empty_lfe5um-85f.config
/usr/share/trellis/misc/basecfgs/empty_lfe5um5g-85f.config
/usr/share/trellis/misc/basecfgs/empty_lfe5um-25f.config
/usr/share/trellis/misc/basecfgs/empty_lfe5um-45f.config
/usr/share/trellis/misc/basecfgs/empty_lfe5um5g-45f.config
/usr/share/trellis/misc/basecfgs/empty_lfe5u-45f.config
/usr/share/trellis/misc/basecfgs/empty_lfe5u-25f.config
/usr/share/trellis/misc/openocd
/usr/share/trellis/misc/openocd/ecp5-versa5g.cfg
/usr/share/trellis/misc/openocd/ulx3s.cfg
/usr/share/trellis/misc/openocd/ecp5-versa.cfg
/usr/share/trellis/misc/openocd/ulx3s_85k.cfg
/usr/share/trellis/misc/openocd/trellisboard.cfg
/usr/share/trellis/misc/openocd/ecp5-evn.cfg
/usr/share/trellis/timing
/usr/share/trellis/timing/util
/usr/share/trellis/timing/util/pip_classes.py
/usr/share/trellis/timing/util/parse_sdf.py
/usr/share/trellis/timing/util/.gitignore
/usr/share/trellis/timing/util/extract_ncl_routing.py
/usr/share/trellis/timing/util/cell_fuzzers.py
/usr/share/trellis/timing/util/cell_timings.py
/usr/share/trellis/timing/util/design_pip_classes.py
/usr/share/trellis/timing/util/__init__.py
/usr/share/trellis/timing/util/timing_dbs.py
/usr/share/trellis/timing/util/interconnect_html.py
/usr/share/trellis/timing/util/timing_solver.py
/usr/share/trellis/timing/util/cell_html.py
/usr/share/trellis/database
/usr/share/trellis/database/ECP5
/usr/share/trellis/database/ECP5/LFE5UM-25F
/usr/share/trellis/database/ECP5/LFE5UM-25F/globals.json
/usr/share/trellis/database/ECP5/LFE5UM-25F/tilegrid.json
/usr/share/trellis/database/ECP5/LFE5UM-25F/iodb.json
/usr/share/trellis/database/ECP5/tiledata
/usr/share/trellis/database/ECP5/tiledata/DUMMY_TILE_S
/usr/share/trellis/database/ECP5/tiledata/DUMMY_TILE_S/bits.db
/usr/share/trellis/database/ECP5/tiledata/EBR_SPINE_LR0
/usr/share/trellis/database/ECP5/tiledata/EBR_SPINE_LR0/bits.db
/usr/share/trellis/database/ECP5/tiledata/DSP_SPINE_UR0
/usr/share/trellis/database/ECP5/tiledata/DSP_SPINE_UR0/bits.db
[... 397 lines cut ...]
/usr/share/trellis/database/ECP5/LFE5U-45F
/usr/share/trellis/database/ECP5/LFE5U-45F/globals.json
/usr/share/trellis/database/ECP5/LFE5U-45F/tilegrid.json
/usr/share/trellis/database/ECP5/LFE5U-45F/iodb.json
/usr/share/trellis/database/ECP5/LFE5UM-45F
/usr/share/trellis/database/ECP5/LFE5UM-45F/globals.json
/usr/share/trellis/database/ECP5/LFE5UM-45F/tilegrid.json
/usr/share/trellis/database/ECP5/LFE5UM-45F/iodb.json
/usr/share/trellis/database/ECP5/LFE5UM-85F
/usr/share/trellis/database/ECP5/LFE5UM-85F/globals.json
/usr/share/trellis/database/ECP5/LFE5UM-85F/tilegrid.json
/usr/share/trellis/database/ECP5/LFE5UM-85F/iodb.json
/usr/share/trellis/database/ECP5/LFE5UM5G-45F
/usr/share/trellis/database/ECP5/LFE5UM5G-45F/globals.json
/usr/share/trellis/database/ECP5/LFE5UM5G-45F/tilegrid.json
/usr/share/trellis/database/ECP5/LFE5UM5G-45F/iodb.json
/usr/share/trellis/database/README.md
/usr/share/trellis/database/.gitignore
/usr/share/trellis/database/devices.json
/usr/share/trellis/database/COPYING
find: ‘/usr/share/doc/google-chrome-stable’: Permission denied
/usr/lib/x86_64-linux-gnu/trellis
/usr/lib/x86_64-linux-gnu/trellis/libtrellis.so
/usr/lib/x86_64-linux-gnu/trellis/pytrellis.so
```

SingularitySurfer, 2020-06-05:
Hi, yes I used the (default) `cmake -DCMAKE_INSTALL_PREFIX=/usr .` to build Trellis.
And I also get:
`find /usr | grep -i trellis
/usr/lib/x86_64-linux-gnu/trellis
/usr/lib/x86_64-linux-gnu/trellis/pytrellis.so
/usr/lib/x86_64-linux-gnu/trellis/libtrellis.so
/usr/share/trellis
/usr/share/trellis/timing
/usr/share/trellis/timing/util
/usr/share/trellis/timing/util/interconnect_html.py
/usr/share/trellis/timing/util/parse_sdf.py
/usr/share/trellis/timing/util/extract_ncl_routing.py
/usr/share/trellis/timing/util/timing_dbs.py
/usr/share/trellis/timing/util/cell_timings.py
/usr/share/trellis/timing/util/.gitignore
/usr/share/trellis/timing/util/cell_html.py
/usr/share/trellis/timing/util/timing_solver.py
/usr/share/trellis/timing/util/pip_classes.py
/usr/share/trellis/timing/util/design_pip_classes.py
/usr/share/trellis/timing/util/cell_fuzzers.py
/usr/share/trellis/timing/util/__init__.py
/usr/share/trellis/database
/usr/share/trellis/database/devices.json
/usr/share/trellis/database/README.md
/usr/share/trellis/database/.gitignore
/usr/share/trellis/database/COPYING
/usr/share/trellis/database/ECP5
/usr/share/trellis/database/ECP5/LFE5UM-45F
/usr/share/trellis/database/ECP5/LFE5UM-45F/globals.json
/usr/share/trellis/database/ECP5/LFE5UM-45F/iodb.json
/usr/share/trellis/database/ECP5/LFE5UM-45F/tilegrid.json
/usr/share/trellis/database/ECP5/timing
/usr/share/trellis/database/ECP5/timing/speed_6
/usr/share/trellis/database/ECP5/timing/speed_6/interconnect.json
/usr/share/trellis/database/ECP5/timing/speed_6/cells.json
/usr/share/trellis/database/ECP5/timing/speed_8
/usr/share/trellis/database/ECP5/timing/speed_8/interconnect.json
/usr/share/trellis/database/ECP5/timing/speed_8/cells.json
/usr/share/trellis/database/ECP5/timing/speed_7
/usr/share/trellis/database/ECP5/timing/speed_7/interconnect.json
/usr/share/trellis/database/ECP5/timing/speed_7/cells.json
/usr/share/trellis/database/ECP5/timing/speed_8_5G
/usr/share/trellis/database/ECP5/timing/speed_8_5G/interconnect.json
/usr/share/trellis/database/ECP5/timing/speed_8_5G/cells.json
/usr/share/trellis/database/ECP5/LFE5UM5G-25F
/usr/share/trellis/database/ECP5/LFE5UM5G-25F/globals.json
/usr/share/trellis/database/ECP5/LFE5UM5G-25F/iodb.json
/usr/share/trellis/database/ECP5/LFE5UM5G-25F/tilegrid.json
/usr/share/trellis/database/ECP5/LFE5UM-25F
/usr/share/trellis/database/ECP5/LFE5UM-25F/globals.json
/usr/share/trellis/database/ECP5/LFE5UM-25F/iodb.json
/usr/share/trellis/database/ECP5/LFE5UM-25F/tilegrid.json
/usr/share/trellis/database/ECP5/LFE5U-85F
/usr/share/trellis/database/ECP5/LFE5U-85F/globals.json
/usr/share/trellis/database/ECP5/LFE5U-85F/iodb.json
/usr/share/trellis/database/ECP5/LFE5U-85F/tilegrid.json
/usr/share/trellis/database/ECP5/LFE5UM5G-85F
/usr/share/trellis/database/ECP5/LFE5UM5G-85F/globals.json
/usr/share/trellis/database/ECP5/LFE5UM5G-85F/iodb.json
/usr/share/trellis/database/ECP5/LFE5UM5G-85F/tilegrid.json
/usr/share/trellis/database/ECP5/LFE5UM5G-45F
/usr/share/trellis/database/ECP5/LFE5UM5G-45F/globals.json
/usr/share/trellis/database/ECP5/LFE5UM5G-45F/iodb.json
/usr/share/trellis/database/ECP5/LFE5UM5G-45F/tilegrid.json
/usr/share/trellis/database/ECP5/LFE5UM-85F
/usr/share/trellis/database/ECP5/LFE5UM-85F/globals.json
/usr/share/trellis/database/ECP5/LFE5UM-85F/iodb.json
/usr/share/trellis/database/ECP5/LFE5UM-85F/tilegrid.json
/usr/share/trellis/database/ECP5/LFE5U-25F
/usr/share/trellis/database/ECP5/LFE5U-25F/globals.json
[... 393 lines cut ...]
/usr/share/trellis/util/common/nets/__init__.py
/usr/share/trellis/util/common/__init__.py
/usr/share/trellis/util/common/database.py
/usr/share/trellis/misc
/usr/share/trellis/misc/openocd
/usr/share/trellis/misc/openocd/ulx3s_85k.cfg
/usr/share/trellis/misc/openocd/ecp5-evn.cfg
/usr/share/trellis/misc/openocd/ecp5-versa5g.cfg
/usr/share/trellis/misc/openocd/ecp5-versa.cfg
/usr/share/trellis/misc/openocd/trellisboard.cfg
/usr/share/trellis/misc/openocd/ulx3s.cfg
/usr/share/trellis/misc/basecfgs
/usr/share/trellis/misc/basecfgs/empty_lfe5um5g-25f.config
/usr/share/trellis/misc/basecfgs/empty_lfe5u-85f.config
/usr/share/trellis/misc/basecfgs/README.md
/usr/share/trellis/misc/basecfgs/empty_lfe5um5g-45f.config
/usr/share/trellis/misc/basecfgs/empty_lfe5u-25f.config
/usr/share/trellis/misc/basecfgs/empty_lfe5um-45f.config
/usr/share/trellis/misc/basecfgs/empty_lfe5um-85f.config
/usr/share/trellis/misc/basecfgs/empty_lfe5um-25f.config
/usr/share/trellis/misc/basecfgs/empty_lfe5um5g-85f.config
/usr/share/trellis/misc/basecfgs/empty_lfe5u-45f.config
`

However, nextpnr seems to not find it during the build.. (same `-DCMAKE_INSTALL_PREFIX=/usr .`)

SingularitySurfer, 2020-06-05:
I get the same errors as in #429

SingularitySurfer, 2020-06-10:
Just FIY, I was able to install it using the litex-conda-packages https://anaconda.org/LiteX-Hub/repo :+1:

lrenaud, 2020-10-11:
Looks like one of the build scripts still has some explicit ties to python3.7. Even running with the cmake command provided I get a few hits for 3.7.

```
$ make -DARCH="ice40;ecp5" -DPYTHONLIBS_VERSION_STRING=3.8  -DPYTHON_INCLUDE_DIR=/usr/include/python3.8 .
$ grep -R python3.7 -l .
./CMakeCache.txt
./CMakeFiles/nextpnr-ice40.dir/link.txt
./CMakeFiles/nextpnr-ice40.dir/build.make
./CMakeFiles/nextpnr-ecp5.dir/link.txt
./CMakeFiles/nextpnr-ecp5.dir/build.make
./generated/gui/ice40/CMakeFiles/gui_ice40_autogen.dir/AutogenInfo.cmake
```

One more flag to clear it out.
`make -DARCH="ice40;ecp5" -DPYTHONLIBS_VERSION_STRING=3.8  -DPYTHON_INCLUDE_DIR=/usr/include/python3.8 -DPYTHON_LIBRARY=/usr/lib/x86_64-linux-gnu/libpython3.8.so .`

I'm not much of a cmake guru, but it's clearly getting snagged on some sort of legacy reference. Might be related to people doing a 19.xx -> 20.04 upgrade and a system cache not being migrated properly? If I knew more about how to debug this I'd provide a patch.

arblake, 2020-12-04:
Following your steps I get to this point:
[ 51%] Building CXX object CMakeFiles/nextpnr-ecp5.dir/ecp5/baseconfigs.cc.o
/home/adrian/nextpnr/ecp5/baseconfigs.cc: In function ‘void nextpnr_ecp5::BaseConfigs::config_empty_lfe5u_45f(nextpnr_ecp5::ChipConfig&)’:
/home/adrian/nextpnr/ecp5/baseconfigs.cc:190:6: note: variable tracking size limit exceeded with ‘-fvar-tracking-assignments’, retrying without
  190 | void config_empty_lfe5u_45f(ChipConfig &cc)
      |      ^~~~~~~~~~~~~~~~~~~~~~
/home/adrian/nextpnr/ecp5/baseconfigs.cc: In function ‘void nextpnr_ecp5::BaseConfigs::config_empty_lfe5u_85f(nextpnr_ecp5::ChipConfig&)’:
/home/adrian/nextpnr/ecp5/baseconfigs.cc:523:6: note: variable tracking size limit exceeded with ‘-fvar-tracking-assignments’, retrying without
  523 | void config_empty_lfe5u_85f(ChipConfig &cc)
      |      ^~~~~~~~~~~~~~~~~~~~~~
{standard input}: Assembler messages:
{standard input}:24803774: Warning: end of file not at end of a line; newline inserted
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[2]: *** [CMakeFiles/nextpnr-ecp5.dir/build.make:375: CMakeFiles/nextpnr-ecp5.dir/ecp5/baseconfigs.cc.o] Error 1
make[2]: *** Deleting file 'CMakeFiles/nextpnr-ecp5.dir/ecp5/baseconfigs.cc.o'
make[1]: *** [CMakeFiles/Makefile2:265: CMakeFiles/nextpnr-ecp5.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
adrian@adrian-HP-Spectre-Notebook:~/nextpnr$