YosysHQ/nextpnr #453: Assert triggers for lp384 devices

olofk, 2020-06-08:
I'm hitting

    nextpnr-ice40 -q --lp384 --package cm36 --pcf tinyfpga_bx.pcf --json servant_1.0.2.json --asc servant_1.0.2_next.asc
     terminate called after throwing an instance of 'nextpnr_ice40::assertion_failure'
      what():  Assertion failure: has_clktoq (/home/olof/code/icestorm/nextpnr/ice40/arch.cc:1132)
    make: *** [Makefile:16: servant_1.0.2_next.asc] Aborted (core dumped)

when targeting an lp384 device. Changing to lp1k gets rid of the assert

Attaching a minimal testcase
[lp384bug.tar.gz](https://github.com/YosysHQ/nextpnr/files/4747576/lp384bug.tar.gz)

daveshah1, 2020-06-08:
Definitely could do with a nicer error, but the problem here is that the design uses a RAM block which don't exist on the lp384, so there is no timing data for these

olofk, 2020-06-08:
Aha, I see. Now I know. Feel free to close or keep open as you see fit