YosysHQ/nextpnr #464: Failing to find trellis library directory on Ubuntu 20

miek, 2020-07-01:
Versions:
prjtrellis: yosyshq/prjtrellis@f93243b000c52b755c70829768d2ae6bcf7bb91a
nextpnr: c0901fb972e34d2d16b3ac0074a18542ba1c448d
cmake: cmake version 3.16.3

I'm doing a fresh install of Ubuntu 20.04 & I'm hitting a couple of cmake issues finding the trellis library directory.

I installed trellis in the usual way:
```
cmake -DCMAKE_INSTALL_PREFIX=/usr/local .
make
sudo make install
```
This ends up with the following library directory:
```
-- Up-to-date: /usr/lib/x86_64-linux-gnu/trellis/libtrellis.so
[...]
-- Up-to-date: /usr/lib/x86_64-linux-gnu/trellis/pytrellis.so
```

and the nextpnr build fails to find it:
```
cmake .. -DARCH=ecp5 -DTRELLIS_INSTALL_PREFIX=/usr
[...]
-- Configuring architecture: ecp5
-- Enabled ECP5 devices: 25k;45k;85k
-- Trellis install prefix: /usr
CMake Error at ecp5/CMakeLists.txt:40 (message):
  Failed to locate the pytrellis library

-- Configuring incomplete, errors occurred!
See also "/home/mike/projects/nextpnr/build/CMakeFiles/CMakeOutput.log".
See also "/home/mike/projects/nextpnr/build/CMakeFiles/CMakeError.log".
```

I assume this is because it's looking for `/usr/lib/trellis` rather than `/usr/lib/<arch>/trellis`.

---

The other issue I'm seeing is that a second run of cmake results in this:
```
$ cmake .. -DARCH=ecp5 -DTRELLIS_INSTALL_PREFIX=/usr                                              
[...]
-- Configuring architecture: ecp5
-- Enabled ECP5 devices: 25k;45k;85k
-- Trellis install prefix: /usr
-- Trellis library directory: TRELLIS_LIBDIR-NOTFOUND
-- Trellis data directory: /usr/share/trellis
-- Using ECP5 chipdb: /home/mike/projects/nextpnr/build/ecp5/chipdb
-- Configuring done
-- Generating done
-- Build files have been written to: /home/mike/projects/nextpnr/build
```

My guess is that cmake loads the variable from its cache and, since the cmake config checks `if (NOT DEFINED TRELLIS_LIBDIR)`, it sees that it's already defined and skips over the checks/errors. Dropping `DEFINED` helps, but there could probably still be some weirdness with the cached path?

whitequark, 2020-07-01:
Yeah I hoped this (the former part) won't come up, I know how to fix it.

whitequark, 2020-07-01:
Please check if https://github.com/YosysHQ/nextpnr/pull/465 works for you.

PR #465 (2020-07-01): Improve Trellis discovery logic
Fixes #464.
closed 2020-07-01 by whitequark