YosysHQ/nextpnr #466: Error in timing analysis when building fx68k 68000 cpu, converted to Verilog

lawrie, 2020-07-04:
I am running the [fx68k](https://github.com/ijor/fx68k) 68000 CPU, converted  from SystemVerilog to Verilog by zachjs sv2v.

All my uses of it fail with this error message:

`ERROR: timing analysis failed due to presence of combinatorial loops, incomplete specification of timing ports, etc.`

This is a simple test case - https://github.com/lawrie/ulx3s_examples/blob/master/test68/test68.v

A slightly modified version of that runs in iVerilog.

It builds if the pwrReset is set to 1''b0, but then the CPU does not run.

I have attached the nextpnr --debug log.

[nextpnr.log](https://github.com/YosysHQ/nextpnr/files/4873483/nextpnr.log)

smunaut, 2020-07-04:
That design contains at least one latch. (in ccrTable for ccrMask signal).
That's not really valid for a FPGA implementation targeting the ECP5.

lawrie, 2020-07-04:
Thanks for that. I am trying a fix at the moment. There is discussion on this on the Ulx3s gitter - https://gitter.im/ulx3s/Lobby

lawrie, 2020-07-04:
Adding a default to the case statement that was missing seems to fix the problem - https://github.com/lawrie/ulx3s_68k/commit/3f5610ebd73396d3b11b7f8f25c10dfeef674b08

daveshah1, 2020-07-05:
Closing as was a genuine combinational loop.

closed 2020-07-05 by daveshah1